moveit/moveit_task_constructor #5: wip-refactor

rhaschke, 2017-09-30:
Dear Michael,
When thinking about the ROS debug interface, I got a whole bunch of new ideas how to generalize and further improve your implementation. Hence, I started to implement some of them here: 
https://github.com/ubi-agni/moveit_task_constructor/tree/wip-refactor

Currently, this branch implements some more specialized classes derived from SubTask, namely 
- PropagatingAnyWay
- PropagatingForward
- PropagatingBackward
- Connecting
- Generator

I also worked on hiding the implementation details. I think in a next step, these specialized classes can be used for task validation at runtime.

However, another generalization I want to work on next, is generic `SubTaskContainers`. The idea is to group subtasks in serial or parallel fashion. However, parallel subtasks require a more generalized handling of the `InterfaceStateList`. I have some rough ideas on that, but I need to sleep about them...
The proposed new API is open for discussion and not yet intended for merging. So, stay tuned ;-)

rhaschke, 2017-10-03:
I already implemented different compute() methods for the different stage types. However, instead of using template overloading, I used virtual methods. I think that's required, because we want to maintain a list of stage pointers in the task...
The next bunch of commits, refactored the `SubTask` classes to allow for hierarchical grouping using the `ContainerBase` class. Right now, only a `SerialContainer` is implemented, but a parallel will follow asap.
For now, I'm happy that the previous behaviour is restored. I also started with some unit tests...

Before you dive into the source changes too deeply, please take a look at the slides for an overview and then let us discuss a bit over phone...
[MoveIt Pipeline.pdf](https://github.com/TAMS-Group/moveit_task_constructor/files/1354161/MoveIt.Pipeline.pdf)

v4hn, 2017-10-04:
Wow. Big 👍 for the slides. I like most of the ideas.
You are aiming really high there.
I do not think we want (well, I don't want) to replace PlanRequestAdapters.
Yes, it is possible to implement `fix_start_state_*` as forward "stages", but the "modify solution"-component of the adapters is not reflected in the task concept.
`add_time_parameterization` and things like resampling-filters cannot be added there in an intuitive way.

I took a quick look into the code by now and have to say I'm not a big fan of `friend` and `impl` classes.
Let's talk about this on the phone after I had some more time to look through this new heap of code.
I'll send you a mail.

rhaschke, 2017-10-04:
Dear Michael,
sorry, I was very busy today and didn't even visited my office. Nevertheless, I managed some further simplification of the code. Gripper, using `PropagatingAnyWay` scheme, can now plan in both directions. I think this class could serve as a template also for `Move` and `CartensianPath`. See [here](https://github.com/ubi-agni/moveit_task_constructor/blob/187f8c4e1563081b1eb6be946e3cd8e544ccaeb9/src/subtasks/gripper.cpp#L85).

> Yes, it is possible to implement fix_start_state_* as forward "stages", but the "modify solution"-component of the adapters is not reflected in the task concept.

Indeed, modifying the solutions is not easily possible yet.

> I'm not a big fan of friend and impl classes.

This comes as a surprise. Wasn't you to one who advertised ABI compatibility?
Using a hidden implementation is the only way to achieve that, isn't it?
I indeed explicitly tried to restrict the API as far as possible to avoid anybody messing up with the data structures. Using all-public structs as you had, doesn't look like a good solution for me ;-)
Let's discuss that on phone.

rhaschke, 2017-10-12:
I realized the renamings we agreed on. Current state compiles and runs again.
`IMPL` changes and `friends` pending. I updated the API [slides](https://github.com/TAMS-Group/moveit_task_constructor/files/1378500/MoveIt.Pipeline.pdf).

closed 2017-11-24 by rhaschke