moveit/moveit_task_constructor #21: MonitoringGenerator: use solutions of a monitored stage as basis for generation

rhaschke, 2018-02-21:
Sometimes its necessary to reuse a previously planned solution, e.g. to traverse it in reverse order or to access the state of another generator. The MonitoringGenerator stage hooks into the `onNewSolution()`.
- [x] For now: reference via raw pointers
- [ ] Later: make `Stage::pointer` a `shared_ptr` (instead of a `unique_ptr`)
- [x] Rewrite `CurrentScene` stage such that it fetches the current planning scene from `get_planning_scene` service (Luca)
- [ ] Rewrite grasp generators to hook up on initial `CurrentScene` stage.
- [ ] Rewrite `Stage::init()` to receive a `RobotModel` argument instead of the weird `PlanningScene` argument. All but generator stages, receive their planning scene through the `Interface` API.

v4hn, 2018-02-26:
I believe it would be a good idea to add a mechanism to retrieve a specific stage pointer from a container/Task, after we moved to `shared_ptr`.
Possibly recursively. Although I'm unsure whether want to use the descriptive name string as id here.

This would help with setting the monitored stage for monitoring generators.

rhaschke, 2019-02-10:
`MonitoringGenerator` is implemented. For `shared_ptr` stuff I opened the new issue #74.

closed 2019-02-10 by rhaschke