moveit/moveit_task_constructor #26: non-empty solution in `onNewSolution`

v4hn, 2018-02-26:
After longer execution (2-3 minutes) of my 10+ stages serial container and possibly close to the end of exhaustive planning I currently see the following quite often:

```
moveit_task_constructor/core/src/container.cpp:347: virtual void moveit::task_constructor::SerialContainer::onNewSolution(const moveit::task_constructor::SolutionBase&): Assertion `solution.empty()' failed.
```

I did not investigate further yet, because I guess you have some ideas what the problem might be.

rhaschke, 2018-02-26:
For me the assertion never failed: As every solution should be moved into the sorted list, I cannot really see, why the assertion fails. However, I provided a quick fix in #27.
Maybe, you should provide your stage pipeline for further investigation.

v4hn, 2018-05-30:
hopefully obsolete, I haven't seen the problem in a long time now.

closed 2018-05-30 by v4hn