moveit/moveit_task_constructor #50: Memory leak in RViz display

v4hn, 2018-05-30:
I currently see a huge memory leak in RViz when I plan and visualize a task with our display.
I didn't have a chance to look into it yet though and it might be a problem of my local ROS setup.

@rhaschke you fixed memory leaks in there before, do you still observe them here?

rhaschke, 2018-05-30:
Could you please be more concrete?
Of course, the display caches the scenes and solutions of all submitted solutions.

Which kind of display options are you using?
This might be related to https://github.com/ros-visualization/rviz/issues/1222 if you have traces enabled.

Is memory freed when you remove tasks from the display?
Is the memory freed when removing the display?

rhaschke, 2018-05-31:
Looks like PlanningScenes are never destructed. Maybe there is a circular reference of PlanningScenePtrs.
No classic leak that could be found with asan address sanitizer.

rhaschke, 2018-05-31:
No, its not the PlanningScenes. They are nicely destroyed.
I now also see a continuous increase of memory when only a single solution is displayed...

rhaschke, 2018-05-31:
However, this is plain rviz receiving tf messages (+1MB per ~15s). I think that's normal.

rhaschke, 2018-05-31:
I have no more ideas for now. Definitely, the memory allocated by new display solutions is not freed, although the solutions themselves are.
@v4hn Please provide more insight from your side.

rhaschke, 2018-10-21:
The underlying reason was rviz: https://github.com/ros-visualization/rviz/pull/1294.

rhaschke, 2019-07-11:
I released rviz with the corresponding fix a while ago :wink:

closed 2019-07-11 by rhaschke