moveit/moveit_task_constructor #63: Add to MoveIt! Melodic Release

davetcoleman, 2018-10-24:
Can we add this to the moveit.rosinstall melodic release of MoveIt? 

Is it time to deprecate the old pick place functionality in Melodic?

rhaschke, 2018-10-24:
We don't yet have a drop-in replacement for the old pick-and-place.
That's a nice WMD candidate though.

> Can we add this to the moveit.rosinstall melodic release of MoveIt?

Why do you think, this is useful? MTC is a package on its own and MoveIt doesn't depend on it.

v4hn, 2018-10-24:
> Is it time to deprecate the old pick place functionality in Melodic?

Not yet, but we are getting closer.

> We don't yet have a drop-in replacement for the old pick-and-place.
That's a nice WMD candidate though.

Sadly this is too much work for WMD, unless someone with a strong background in the old pick pipeline and some insight into the Task Constructor project works on Pick (not Place yet, that's more work).
This should be implemented as a MoveGroupCapability.

Also we still change API and fix core problems, so I'm not in favor of a MoveIt Melodic release yet.

davetcoleman, 2018-10-24:
Can you at least announce it and add tutorials / documentation for it? The more users the better, right?

btw I finally read your paper on the MTC. Nice paper! I'm curious how the task state machine handles error states but that's a discussion for a different thread.

rhaschke, 2018-10-24:
> I'm curious how the task state machine handles error states but that's a discussion for a different thread.

Not yet at all. This is an idea for future work...

v4hn, 2018-10-24:
> Can you at least announce it

It's somewhat hard to announce it louder than via a ROSCon presentation.

> and add tutorials / documentation for it?

Yes. It's on my radar and gets more and more urgent.

You are welcome to fix bugs in MoveIt in the meanwhile.

davetcoleman, 2018-10-26:
> It's somewhat hard to announce it louder than via a ROSCon presentation.

I think there are people in the community that have not yet heard of it actually. There was discussion someone about that exact this week (I forget where). How about a blog post? If you don't want to fully create it certainly @rcc0002 can help. I'm sure we can copy the ROSCon abstract as a good start

> You are welcome to fix bugs in MoveIt in the meanwhile.

Touche ;-)

davetcoleman, 2020-06-02:
2 years ago you said this was not ready to be released, but surely not it is. Can we release it to Melodic and Noetic? We can still call it beta and make breaking changes to it...

rhaschke, 2020-06-02:
I'm fine with doing so.

davetcoleman, 2020-06-02:
heads up @tylerjw 
Note also MTC is now on the /moveit README badge table:
https://github.com/ros-planning/moveit/blob/master/README.md

v4hn, 2020-06-03:
I agree, we will definitely release for Noetic and at the same time might release the same version to Melodic as well.

I would like to have a few interface changes in there before then (prominently finally moving away from `unique_ptr`), but let's see once everything else is ready for noetic.