moveit/moveit_task_constructor #64: catkin build error: ‘class collision_detection::World’ has no member named ‘moveObject’

youtalk, 2018-10-24:
I was so impressed the "MoveIt! Task Planning" presentation at ROSCon 2018.
I'm trying to use it but I found a `catkin build` error as follow.

```
Starting  >>> moveit_task_constructor_core                                                                                 
Finished  <<< moveit_task_constructor_capabilities                 [ 7.4 seconds ]                                         
___________________________________________________________________________________________________________________________
Errors     << moveit_task_constructor_core:make /home/youtalk/ros/kinetic/logs/moveit_task_constructor_core/build.make.003.log
/home/youtalk/ros/kinetic/src/ros-planning/moveit_task_constructor/core/src/stages/fix_collision_objects.cpp: In member function ‘moveit::task_constructor::SubTrajectory moveit::task_constructor::stages::FixCollisionObjects::fixCollisions(planning_scene::PlanningScene&) const’:
/home/youtalk/ros/kinetic/src/ros-planning/moveit_task_constructor/core/src/stages/fix_collision_objects.cpp:150:30: error: ‘class collision_detection::World’ has no member named ‘moveObject’
    scene.getWorldNonConst()->moveObject(name, Eigen::Affine3d(Eigen::Translation3d(correction)));
                              ^
make[2]: *** [src/stages/CMakeFiles/moveit_task_constructor_core_stages.dir/fix_collision_objects.cpp.o] Error 1
make[1]: *** [src/stages/CMakeFiles/moveit_task_constructor_core_stages.dir/all] Error 2
make: *** [all] Error 2
```

Actually the `World` class has not `moveObject` method. Do you have any idea?

youtalk, 2018-10-24:
I've solved by myself.
When I use the latest `moveit` package from source instead of binary, it can be built successfully.

closed 2018-10-24 by youtalk

rhaschke, 2018-10-24:
Hm. Somehow my comments get lost on github. That's exactly what I suggested ;-)
We added moveObject to MoveIt core only recently and the new release is just still in the pipeline...