moveit/moveit_task_constructor #67: Unhelpful error message during task construction

mxm0, 2018-11-01:
We had the following error when using the task constructor. The error was due to a stage "connector" type which was missing information from the stage below. The error message was not really helpful and redrawing the task structure in the command line would be more helpful.

```
[ERROR] [1541083635.557176600]: Error initializing stage:
task pipeline: output interface of 'move to pre-pour pose' doesn't match mine
```

@v4hn

rhaschke, 2018-11-01:
> redrawing the task structure in the command line would be more helpful

This can be easily achieved like this:
```
Task t;
std::cerr << t << std::endl;
```

v4hn, 2018-11-02:
> This can be easily achieved like this

Yes, maybe, I didn't check yet.
For newcomers the error message is rather more confusing than necessary and I proposed filing this issue so that we can improve the reported error later.

rhaschke, 2018-11-02:
What kind of msg do you propose?

v4hn, 2018-11-02:
Ideally, we might print the structure of the task and add a highlight on the conflicting stage interfaces.

rhaschke, 2018-11-02:
> Ideally, we might print the structure of the task and add a highlight on the conflicting stage interfaces.

That's what `std::cerr << t << std::endl;` does. But I think, this should be added at a higher level.

rhaschke, 2019-02-10:
Addressed by #71.

closed 2019-02-10 by rhaschke