moveit/moveit_task_constructor #86: compilation problem

ipa-mjp, 2019-04-04:
Hello,

I clone moveit task constructor successfully. I am going to compile it and give me error about 

moveit_task_constructor/core/src/solvers/cartesian_path.cpp:72:52:   required from here
/usr/include/eigen3/Eigen/src/Core/util/StaticAssert.h:32:40: error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);

Please let me know, what am I doing wrong.

rhaschke, 2019-04-04:
MTC is currently limited to ROS Melodic / Ubuntu Bionic. Looks like you are building against the older Kinetic, do you?

ipa-mjp, 2019-04-04:
Thank you for your quick replay. Yup, I am using ROS kinetic with ROS version 1. Than probably gives this error. Is there any plan to release with ROS Kinetic?

rhaschke, 2019-04-04:
No. MTC, naturally relies on MoveIt and MoveIt has many API changes in Melodic, we rely on. However, you can build MoveIt's melodic branch from source and then should be able to build MTC as well.

ipa-mjp, 2019-04-04:
Thank you so much for information.

closed 2019-06-03 by rhaschke