moveit/moveit_task_constructor #90: Add custom action between 2 stages

cpt-yoshi, 2019-05-30:
Is there a way to add a custom blocking action between 2 stages that doesn't rely on a trajectory? If not, could you give a few pointers for where I could implement this behaviour?

For example if I wanted to dispense some liquid (through a ros action server) after a certain stage.

I have tried adding my action between 2 tasks instead of stages as a workaround without success:
- Create task 1 to insert a tip (for holding the liquid) and place above a container.
- Execute task 1
- Dispense liquid through a custom ros action server
- Create task 2 to eject the tip in the trash (**NOT WORKING**)

Task 2 stages can been seen in RViz (Motion Planning Tasks) but dosent compute anything. An error is returned:
``` bash
[ERROR] [1559245916.561719187]: Tried to advertise a service that is already advertised in this node [/move_group/ompl/set_parameters]
```

The code below implements a modified pick and place from TAMS-Group _mtc_pour_ with 2 tasks instead of one (To validate the above workaround). 

<details><summary>Code that fails</summary>
<p>

``` c++

/** 
 *
 * TASK 1: Pick 
 * 
 */
  Task t1("pick");
  t1.loadRobotModel();

  auto sampling_planner = std::make_shared<solvers::PipelinePlanner>();
  sampling_planner->setProperty("goal_joint_tolerance", 1e-5); // 1e-5
  sampling_planner->setProperty("max_velocity_scaling_factor", 0.5);
  sampling_planner->setProperty("max_acceleration_scaling_factor", 0.5);

  // Cartesian planner
  auto cartesian_planner = std::make_shared<solvers::CartesianPath>();
  cartesian_planner->setMaxVelocityScaling(.5);
  cartesian_planner->setMaxAccelerationScaling(.5);
  cartesian_planner->setStepSize(.002);

  t1.setProperty("group", "ur3_arm");
  t1.setProperty("eef", "robotiq_2f_140");
  t1.setProperty("gripper", "robotiq_2f_140"); // TODO: use this

  Stage* current_state= nullptr;
  {
    auto _current_state = std::make_unique<stages::CurrentState>("current state");
    current_state= _current_state.get();
    t1.add(std::move(_current_state));
  }

  // open gripper
  {
    auto stage = std::make_unique<stages::MoveTo>("open gripper", sampling_planner);
    stage->setGoal("open"); 
    stage->setGroup("robotiq_2f_140");
    t1.add(std::move(stage));
  }

  {
    // TODO: overload for single planner case
    auto stage = std::make_unique<stages::Connect>("move to pre-grasp pose", 
    stages::Connect::GroupPlannerVector {{"ur3_bio_ik", sampling_planner}});
    stage->properties().configureInitFrom(Stage::PARENT); // TODO: convenience-wrapper
    t1.add(std::move(stage));
  }

  {
    auto stage = std::make_unique<stages::MoveRelative>("approach object", cartesian_planner);
[... 236 lines cut ...]
    // TODO: optionally also plan stages if incoming states have infinite cost. This facilitates debugging
    t2.plan(nh.param<int>("solutions", 3));
  }
  catch(InitStageException& e){
    ROS_ERROR_STREAM(e);
  }

  if(!execute || t2.numSolutions() == 0){
    std::cout << "waiting for <enter>" << std::endl;
    std::cin.get();
  }
  else {
    moveit_task_constructor_msgs::Solution solution;
    t2.solutions().front()->fillMessage(solution);
    std::cout << "executing solution" << std::endl;
    executeSolution(solution);
    std::cout << "done" << std::endl;
    std::cout << "waiting for <enter>" << std::endl;
    std::cin.get();
  }

  return 0;
```
</p>
</details>

rhaschke, 2019-06-02:
I'm not sure what exactly you want to achieve. Do you want to have a pause (e.g. waiting for a key press or some other event) between the execution of 2 stages?
This behaviour is indeed not yet possible, because execution relies on trajectory execution only.
In the future, I envision to allow configuring different action controllers for individual stages, which then could handle such external events.
For now, your approach using two tasks for this seems reasonable. Looks like creating two tasks from the same node is not really supported. My guess is: You cannot create two (OMPL) pipelines in the same node. Try to reuse the pipeline planner, or clear the task1 instance and reuse it for task2:
```c++
Task t ("pick")
/* configure and execute task1 */

t = Task("place");
/* configure and execute task1 */
```
As a result, you cannot inspect task1 anymore after resetting it.

cpt-yoshi, 2019-06-03:
> I'm not sure what exactly you want to achieve. Do you want to have a pause (e.g. waiting for a key press or some other event) between the execution of 2 stages?

Thx for the quick response! My project uses a custom pipette attached to the robot so that it can aspirate and dispense liquids. The sequence is to attach the tip to the pipette tool (move to pose + attach mesh of tip). Then go in a container containing liquid (move to pose). After these stages, I would have liked to be able to aspirate water with my pipette tool (action server using topics). Then go above a well to dispense the water...

> For now, your approach using two tasks for this seems reasonable.

Reading #17, it dosen't seem that I can get the finished stage state from Task 1 to be used as the initial stage state for Task 2 yet. Is this correct? This would force me to always execute the previous task?

> Try to reuse the pipeline planner, or clear the task1 instance and reuse it for task2:

No matter how I try to add a second task, resetting the task, reusing the same pipeline, it fails when trying to plan for Task 2 (Note that the previous OMPL error is now gone :smiley:):

``` bash
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc

Thread 1 "bob_planner_mtc" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) backtrace 
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff4736801 in __GI_abort () at abort.c:79
#2  0x00007ffff5129957 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff512fab6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff512faf1 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff512fd24 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff513029c in operator new(unsigned long) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff7a29489 in void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char*>(char*, char*, std::forward_iterator_tag) ()
   from /home/bob/ws/ros/catkin_ws/devel/lib/libmoveit_task_constructor_core_stages.so
#8  0x00007ffff576cb8d in moveit::task_constructor::Task::id[abi:cxx11]() const () from /home/bob/ws/ros/catkin_ws/devel/lib/libmoveit_task_constructor_core.so
#9  0x00007ffff57127c1 in moveit::task_constructor::Introspection::reset() () from /home/bob/ws/ros/catkin_ws/devel/lib/libmoveit_task_constructor_core.so
#10 0x00007ffff576c24a in moveit::task_constructor::Task::reset() () from /home/bob/ws/ros/catkin_ws/devel/lib/libmoveit_task_constructor_core.so
#11 0x00007ffff576c5cb in moveit::task_constructor::Task::plan(unsigned long) () from /home/bob/ws/ros/catkin_ws/devel/lib/libmoveit_task_constructor_core.so
#12 0x00005555556cf726 in main ()
```
You can reproduce the above error with the code below :

``` c++
Task t("pick");
t.loadRobotModel();
t.setProperty("group", "ur3_arm");
t.setProperty("eef", "robotiq_2f_140");
robotModel = t.getRobotModel();

// Create stages for Task: pick
{
}

// Run Task place
t.enableIntrospection();
t.plan(nh.param<int>("solution", 1));
executeSolution ...

// Task place
// t.clear();    // Does not change runtime error at the end
t = Task("place");
// t.loadRobotModel();    // SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.
t.setRobotModel(robotModel);    // Removes warning

// Create stages for Task: place
{
}

// t.enableIntrospection();    
t.plan(nh.param<int>("solution", 1));     // RUNTIME ERROR
```

Another hackish solution for now would be to add a GripperCommand controller for my pipette tool in MoveIt config and create a Gripper action server (that would decode a position value to aspirate or dispense a liquid). With this approach, a pipette action can be added as a stage.

rhaschke, 2019-06-03:
I can reproduce your issue / segfault. Problem is that `Task::operator=(Task&&)` is buggy.
The solution is not trivial though. I'm working on it...

rhaschke, 2019-07-11:
Should be fixed with #94.

closed 2019-07-11 by rhaschke