moveit/moveit_task_constructor #92: How to open gripper in Place stage

yepw, 2019-07-05:
Hi, 
Thank you for providing such a good tool!

I am trying to implement a Place stage after a Pick stage. Currently, my pipeline seems stuck in the substage "open gripper" under "ungrasp". In Pick stage, we use `grasp_generator->setPreGraspPose("open")` and `grasp_generator->setGraspPose("closed")` to define the grasp pose, but I couldn't find a corresponding command in Place stage. I think we need to tell the ungrasp generator to set the gripper pose to "open" after detaching the object. Could you tell me how to do that?

It would be super helpful if you could post some demo that uses the Place stage. Thank you in advance!

rhaschke, 2019-07-05:
Please have a look into https://github.com/ubi-agni/mtc_demos/blob/iros19/stages/grasping_tasks.cpp and the corresponding "demo" program https://github.com/ubi-agni/mtc_demos/blob/iros19/src/iros_grasp.cpp.
Unfortunately, I didn't have much time recently to continue on this project and provide nice tutorials/examples for all use cases.

May I ask, where you learned about this (still unreleased) package?

yepw, 2019-07-08:
Thank you for your reply! I started with the old pick/place pipeline and found it had some limitation. I found this package from one issue (I don't remember which) under the moveit repo.

closed 2019-07-08 by yepw