moveit/moveit_task_constructor #93: Error when executing a solution

yepw, 2019-07-10:
Hello,

I am trying to use MTC to do a pick/place task on a UR5 with a Robotiq 2-finger gripper. `task->plan()` works well for me. I can visualize the movement in Rviz after doing `task->publishAllSolutions()`.

However, when I try to execute the robot using `task->execute(* (task->solutions().front()))`, I got the following error:

`Could not find JointModelGroup that actuates {shoulder_pan_joint, shoulder_lift_joint, elbow_joint, wrist_1_joint, wrist_2_joint, wrist_3_joint, finger_joint} `

A similar issue was reported [here](https://github.com/ubi-agni/moveit_task_constructor/issues/14).

---

After reading some source code, I found the error is from the `findJointModelGroup` function in `execute_task_solution_capability.cpp`. My `solution` has two `sub_trajectory`. If I understand it correctly, one `sub_trajectory` is corresponding to the arm (6 DoF), so the `sub_trajectory` should have **6** `joint_names`. The other 'sub_trajectory' is corresponding to the gripper (1 DoF), so the second `sub_trajectory` should have only **one** `joint_names`. However, currently, both the `sub_trajectory` have **7** `joint_names`. Neither of my JMGs has the 7 joints, so the error occurs.

I noticed there is a property called `group`, which seems like to indicate which JMG should be used. I printed the `group` property for each stage:

```
/ my                                                        manipulator
-- / current                                                None
-- / approach_my_pick                                       manipulator
-- / my_pick                                                None
----- / approach object                                     manipulator   
----- / grasp                                               gripper
-------- / compute ik                                                
----------- / return pre-defined grasp poses                gripper 
-------- / allow object collision                           None 
-------- / close gripper                                    gripper   
-------- / attach object                                    None    
----- /lift object                                          manipulator
```

The `group` properties seem right for me. Another thing I think might be relevant is how I declare the JMG:

```ruby
    <group name="manipulator">
        <chain base_link="base_link" tip_link="fts_toolside" />
    </group>
    <group name="gripper">
        <joint name="finger_joint" />
        <joint name="left_outer_finger_joint" />
        <joint name="left_inner_finger_joint" />
        <joint name="left_inner_knuckle_joint" />
        <joint name="right_inner_knuckle_joint" />
        <joint name="right_outer_knuckle_joint" />
        <joint name="right_outer_finger_joint" />
        <joint name="right_inner_finger_joint" />
        <joint name="right_finger_tip_joint" />
        <joint name="left_finger_tip_joint" />
        <joint name="ur5_ee_link-gripper_base" />
    </group>
```

I got stuck here for a long time, so any suggestion or example would be greatly appreciated! 

Many thanks!

yepw, 2019-07-10:
Ok, I think I might know the reason.

When setting up `approach_my_pick (stages::Connect)`, I used to do this:

```ruby
stages::Connect::GroupPlannerVector planners = {{eef, interpolate}, {arm, pipeline}};
auto connect = new stages::Connect("approach_" + name, planners);
connect->properties().configureInitFrom(Stage::PARENT);
```

Here I combined the arm planner and eff planner as a new planner. When running `plan()`, everything is fine, but an error will occur during `execute()` because no JointModelGroup satisfies the new planner.

To correct it, I first open the eff ("gripper") and then use `stages::Connect` to connect the current arm status to the pre-grasp arm status.

```ruby
auto open_gripper = new stages::MoveTo("open gripper", pipeline);
open_gripper->setGroup("gripper");
open_gripper->setGoal("open");
container.insert(Stage::pointer(open_gripper));

stages::Connect::GroupPlannerVector planners = {{arm, pipeline}};
auto connect = new stages::Connect("approach_" + name, planners);
connect->properties().configureInitFrom(Stage::PARENT);
```

rhaschke, 2019-07-11:
Thanks for reporting and tracking this issue down. Unfortunately, I didn't have much time recently to look into issues. For me, the `Connect` stage also works with multiple planners. But, indeed, I do have combined groups defined as well. Could you try to define such a combined group in your srdf as well (using the subgroup feature) and try again?

v4hn, 2019-07-11:
I believe we discussed this issue before and it requires changes in moveit_core datastructures.

The problem boils down to `robot_trajectory::RobotTrajectory` [requiring a JointModelGroup for construction](https://github.com/ros-planning/moveit_task_constructor/blob/master/capabilities/src/execute_task_solution_capability.cpp#L170). If the JMG is not specified, MoveIt implicitly assumes it relates to [all joints](https://github.com/ros-planning/moveit/blob/master/moveit_core/robot_trajectory/src/robot_trajectory.cpp#L209) of the robot.
To solve this, the class could also rely on a list of actuated joints (or JMGs) instead of this one pointer to convey the information.

As the easy workaround simply adds a `<group>` with the two subgroups to the srdf, we did not push this change in MoveIt yet. @yepw your workaround to add the additional `MoveTo` is inferior to this, so you probably want to apply the proposed change instead.

Let's leave this issue open as a reminder until the problem is fully fixed.

yepw, 2019-07-12:
Hi, thanks for the detailed response!

I try the following two ways to add subgroups, but neither of them works.
```ruby
        <group name = "manipulator_gripper">
		<subgroup name = "manipulator">
			<chain base_link="base_link" tip_link="fts_toolside" />
		</subgroup>
		<subgroup name = "gripper">
		     <joint name="finger_joint" />
		    <joint name="left_outer_finger_joint" />
		    <joint name="left_inner_finger_joint" />
		    <joint name="left_inner_knuckle_joint" />
		    <joint name="right_inner_knuckle_joint" />
		    <joint name="right_outer_knuckle_joint" />
		    <joint name="right_outer_finger_joint" />
		    <joint name="right_inner_finger_joint" />
		    <joint name="right_finger_tip_joint" />
		    <joint name="left_finger_tip_joint" />
		    <joint name="ur5_ee_link-gripper_base" />
		</subgroup>
	</group>    
```
```ruby
	<group name = "manipulator_gripper">
		<subgroup name = "manipulator"/>
		<subgroup name = "gripper"/>
	</group>   
	<group name="manipulator">
		<chain base_link="base_link" tip_link="fts_toolside" />
	</group>
	<group name="gripper">
		<joint name="finger_joint" />
		<joint name="left_outer_finger_joint" />
		<joint name="left_inner_finger_joint" />
		<joint name="left_inner_knuckle_joint" />
		<joint name="right_inner_knuckle_joint" />
		<joint name="right_outer_knuckle_joint" />
		<joint name="right_outer_finger_joint" />
		<joint name="right_inner_finger_joint" />
		<joint name="right_finger_tip_joint" />
		<joint name="left_finger_tip_joint" />
		<joint name="ur5_ee_link-gripper_base" />
	</group>
```
In both way, I will get warning:

[ WARN] [1562896590.434899647, 3.601000000]: Group 'manipulator_gripper' must have at least one valid joint
[ WARN] [1562896590.434952535, 3.601000000]: Failed to add group 'manipulator_gripper'

Maybe I understand you in the wrong way.

rhaschke, 2019-07-12:
This should be the correct syntax:
```xml
	<group name = "manipulator_gripper">
		<group name = "manipulator"/>
		<group name = "gripper"/>
	</group>   
```
Maybe the order is relevant as well.

yepw, 2019-07-13:
@rhaschke Thank you! The syntax works!

closed 2020-02-18 by rhaschke