moveit/moveit_task_constructor #103: Crash when OMPL_ERROR

yepw, 2019-07-17:
Hi, just found MTC might crash if OMPL raises an OMPL_ERROR. Here is the error:

---

[ INFO] [1563330382.250656588]: Planner configuration 'manipulator' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1563330382.250938641]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1563330382.301680845]: Constraint satisfied:: Joint name: 'shoulder_pan_joint', actual value: -0.862016, desired value: -0.861956, tolerance_above: 0.000100, tolerance_below: 0.000100
[ INFO] [1563330382.301706924]: Constraint satisfied:: Joint name: 'shoulder_lift_joint', actual value: -1.225333, desired value: -1.225381, tolerance_above: 0.000100, tolerance_below: 0.000100
[ INFO] [1563330382.301715709]: Constraint satisfied:: Joint name: 'elbow_joint', actual value: 1.545479, desired value: 1.545482, tolerance_above: 0.000100, tolerance_below: 0.000100
[ INFO] [1563330382.301723375]: Constraint satisfied:: Joint name: 'wrist_1_joint', actual value: 1.653272, desired value: 1.653322, tolerance_above: 0.000100, tolerance_below: 0.000100
[ INFO] [1563330382.301732369]: Constraint satisfied:: Joint name: 'wrist_2_joint', actual value: 0.272998, desired value: 0.272977, tolerance_above: 0.000100, tolerance_below: 0.000100
[ INFO] [1563330382.301740413]: Constraint satisfied:: Joint name: 'wrist_3_joint', actual value: 1.413279, desired value: 1.413316, tolerance_above: 0.000100, tolerance_below: 0.000100
[ INFO] [1563330382.301897636]: Found a contact between 'wrist_2_link' (type 'Robot link') and 'forearm_link' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1563330382.301911324]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ERROR] [1563330382.351721638]: RRTConnect: Unable to sample any valid states for goal tree
[ INFO] [1563330382.351751785]: RRTConnect: Created 1 states (1 start + 0 goal)
[ INFO] [1563330382.351790976]: No solution found after 0.101040 seconds
[ INFO] [1563330382.351816087]: Unable to solve the planning problem
[ur5_gripper-2] process has died

---

I guess `RRTConnect` could not find a valid goal states (collision was found). However, the `compute_ik` returned 4 potential IK solutions. It seems MTC only tried the first one, found collision and crashed.

[image]

rhaschke, 2019-07-17:
Thanks for reporting this issue. Would be great, if you could provide a stacktrace as well.
What is strange that ComputeIK returns an invalid solution (Found a contact between 'wrist_2_link' (type 'Robot link') and 'forearm_link' (type 'Robot link')). Can you confirm this?

v4hn, 2019-07-17:
To clear up some potential confusion:

MTC never plans to Cartesian targets.
The ComputeIK stage factorizes such targets and forwards individual corresponding joint targets to the Connect stage which plans, e.g. using RRTConnect, joint trajectories for them.

As Robert pointed out, it looks like ComputeIK forwarded an invalid solution in your case.
Did you by any chance set the `ignore_collisions` flag (which should be used mostly for debugging)?
Still, even with the flag, `Connect` should simply reject the solution.

Please provide a stacktrace and possibly a minimal example to resolve this.
The segfault might not be related to OMPL at all.

yepw, 2019-07-17:
Hi, thanks for the quick response. 

I tried to manually move the robot to the target pose. There is a collision between  'wrist_2_link' (type 'Robot link') and 'forearm_link' (type 'Robot link').

[image]

I am not familiar with GDB or stack trace. Hope this is what you want:

---

[ INFO] [1563375030.361168682]: Found a contact between 'wrist_2_link' (type 'Robot link') and 'forearm_link' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1563375030.361184966]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[Thread 0x7fff9bb62700 (LWP 16765) exited]
[ERROR] [1563375030.410282380]: RRTConnect: Unable to sample any valid states for goal tree
[ INFO] [1563375030.410314371]: RRTConnect: Created 1 states (1 start + 0 goal)
[ INFO] [1563375030.410350897]: No solution found after 0.121263 seconds
[ INFO] [1563375030.410388303]: Unable to solve the planning problem

Thread 16 "robotic_assembl" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffb17fa700 (LWP 15743)]
0x00007ffff66ccda4 in std::_Deque_iterator<double, double const&, double const*>::_Deque_iterator(std::_Deque_iterator<double, double&, double*> const&) ()
   from /home/yeping/moveit_melodic_ws/devel/.private/moveit_task_constructor_core/lib/libmo---Type <return> to continue, or q <return> to quit---veit_task_constructor_core_stages.so
(gdb) #0  0x00007ffff66ccda4 in std::_Deque_iterator<double, double const&, double const*>::_Deque_iterator(std::_Deque_iterator<double, double&, double*> const&) ()
   from /home/yeping/moveit_melodic_ws/devel/.private/moveit_task_constructor_core/lib/libmoveit_task_constructor_core_stages.so
#1  0x00007ffff66ce649 in std::deque<double, std::allocator<double> >::begin() const ()
   from /home/yeping/moveit_melodic_ws/devel/.private/moveit_task_constructor_core/lib/libmoveit_task_constructor_core_stages.so
#2  0x00007ffff66cb499 in moveit::task_constructor::stages::Connect::makeSequential(std::vector<std::shared_ptr<robot_trajectory::RobotTrajectory const>, std::allocator<std::shared_ptr<robot_trajectory::RobotTrajectory const> > > const&, std::vector<std::shared_ptr<planning_scene::PlanningScene const>, std::allocator<std::shared_ptr<planning_scene::PlanningScene const> > > const&, moveit::task_constructor::InterfaceState const&, moveit::task_constructor::I---Type <return> to continue, or q <return> to quit---

rhaschke, 2019-07-17:
Thanks for providing the stacktrace. It clearly shows that the issue is in MTC's Connect stage, calling makeSequential. However, to analyze this error, I would need to reproduce it. Could you possibly provide a (minimal) example of your MTC program?

yepw, 2019-07-17:
Hi, below is one example. 

---
``` ruby
#include <moveit/task_constructor/solvers/pipeline_planner.h>
#include <moveit/task_constructor/solvers/cartesian_path.h>
#include <moveit/task_constructor/solvers/joint_interpolation.h>
#include <moveit/task_constructor/stages/current_state.h>
#include <moveit/task_constructor/stages/modify_planning_scene.h>
#include <moveit/task_constructor/stages/fix_collision_objects.h>
#include <moveit/task_constructor/stages/fixed_cartesian_poses.h>
#include <moveit/task_constructor/stages/connect.h>
#include <moveit/task_constructor/stages/compute_ik.h>
#include <moveit/task_constructor/stages/move_to.h>
#include <moveit/task_constructor/stages/move_relative.h>
#include <moveit/task_constructor/stages/simple_grasp.h>
#include <moveit/task_constructor/stages/generate_place_pose.h>
#include <moveit/task_constructor/stages/pick.h>
#include <moveit/task_constructor/task.h>

#include <geometry_msgs/PoseStamped.h>
#include <eigen_conversions/eigen_msg.h>
#include <tf/tf.h>

using namespace moveit::task_constructor;

void addMove(ContainerBase& container,Stage* grasped,
                        const geometry_msgs::PoseStamped& p,  const geometry_msgs::Pose& tf_eef_to_object, 
						const std::string& name, const std::string & object_name) {
    const std::string tool_frame = "fts_toolside";
    const std::string eef = "gripper";
    const std::string arm = "manipulator";

    auto pipeline = std::make_shared<solvers::PipelinePlanner>();
    // pipeline->setPlannerId("RRTstarkConfigDefault");

    stages::Connect::GroupPlannerVector planners = { {arm, pipeline}};
    auto connect = new stages::Connect("approach " + name, planners);
    container.insert(Stage::pointer(connect));

    auto move_generator = std::make_unique<stages::GeneratePose>("pose");
    move_generator->properties().configureInitFrom(Stage::PARENT);
    move_generator->setPose(p);
    move_generator->setMonitoredStage(grasped);

    auto ik = std::make_unique<stages::ComputeIK>("compute", std::move(move_generator) );
    ik->setMaxIKSolutions(8);
    ik->properties().configureInitFrom(Stage::INTERFACE, {"target_pose"});

    Eigen::Isometry3d eigen;
    tf::poseMsgToEigen(tf_eef_to_object, eigen);
    ik->setIKFrame(eigen, tool_frame);
    ik->properties().set("eef", eef);
    ik->properties().set("object", object_name);

    container.insert(std::move(ik));

    return ;
}

geometry_msgs::Pose tftoMsg(tf::Transform transform)
{
    geometry_msgs::Pose pose;
    pose.position.x = transform.getOrigin().getX();
    pose.position.y = transform.getOrigin().getY();
    pose.position.z = transform.getOrigin().getZ();
    pose.orientation.x = transform.getRotation().getX();
    pose.orientation.y = transform.getRotation().getY();
    pose.orientation.z = transform.getRotation().getZ();
    pose.orientation.w = transform.getRotation().getW();
    return pose;
}

void executeTask (Task* task) {
    task->plan(30);
    ROS_INFO("planning");

    task->publishAllSolutions(false);
    task->enableIntrospection(); 
    ros::WallDuration(1.0).sleep();
    if (task->numSolutions() > 0) {
    task->execute(*task->solutions().front());
    }
    return;
}

int main(int argc, char **argv)
{
    ros::init(argc, argv, "ur5");
    ros::NodeHandle node;

    geometry_msgs::PoseStamped p;
    p.header.frame_id = "world";
    p.pose.position.x = 0.579956;
    p.pose.position.y = -0.022243;
    p.pose.position.z = 0.321161;
    p.pose.orientation.x = 0.759275;
    p.pose.orientation.y =  0.217995; 
    p.pose.orientation.z = 0.52639;
    p.pose.orientation.w =  -0.314472;

    geometry_msgs::Pose ik_frame;
    ik_frame = tftoMsg( tf::Transform(tf::createQuaternionFromRPY(0, 0, -M_PI / 2), tf::Vector3(0.0, 0.0, 0.2)) );

    Task t ("first_task");

    // t = Task ("second_task");
    t.setProperty("group", "manipulator");
    t.setProperty("eef", "gripper");
    t.setProperty("gripper", "gripper");
    t.loadRobotModel();

    Task * task = &t;
    task->add(std::move( std::make_unique<stages::CurrentState>("current")));
    std::string monitoed_stage = "current";
    addMove(*task->stages(), task->stages()->findChild(monitoed_stage), p , ik_frame, "my_task", "test");

    executeTask(task);

    ros::WallDuration(1.0).sleep();

    ros::AsyncSpinner spinner(0);
    spinner.start();
    ros::waitForShutdown();
    return 0;
}

```
---
I was running this on an ur5 hardware. We might have different setup on the end-effector offset. Hope you can reproduce this.

rhaschke, 2019-07-18:
Thanks for providing the MTC code. However, without your specific robot setup, I'm not yet able to reproduce the problem. I do have the `ur_description` package. Could you, please, send your (generated) urdf and the moveit_config package (zipped). Use a [gist](https://gist.github.com/) to upload those files. The main information missing right now: what is the base transform of your robot in the world and what is the transform from `ur5_tool0` to `fts_toolside`?
Even better, just zip your whole catkin workspace-src, comprising the robot description packages (arm and gripper) and the moveit_config package.

yepw, 2019-07-18:
Very sorry for the incomplete information! 

All the robot description and moveit_config info are in [this repo](https://github.com/intuitivecomputing/ur5_with_robotiq_gripper/tree/mtc) (**MTC** branch). I found the error is also reproducible in Gazebo simulation, so you can:
```
roslaunch icl_ur5_setup_gazebo icl_ur5_gripper.launch
```
```
roslaunch icl_ur5_setup_moveit_config ur5_gripper_moveit_planning_execution.launch sim:=true
```
```
roslaunch icl_ur5_setup_moveit_config moveit_rviz.launch
```
```
rosrun the-code-above
```
If you meet any problem, feel free to let me know. Thank you very much for your help!

rhaschke, 2019-07-19:
Thanks for providing more details. I was able to run your code, but I couldn't reproduce your issues.
MTC's ComputeIK returns with 3 IK solutions only. Key question, is why you have another one, which might be the culprit.
```
  1  -   <-  3   ->  -  3 / first_task
    1  -   <-  1   ->  -  1 / current
    -  1   ->  3   <-  3  - / approach my_task
    3  -   <-  3   ->  -  3 / compute
      1  -   <-  1   ->  -  1 / pose
```
Which version / commit of MTC are you using?

rhaschke, 2019-07-19:
The culprit was a recent PR (#97), which introduced this bug. I committed an appropriate fix to master: 6ff1a5162f99fb08dbf78311f0ebc0212d0b0506.

closed 2019-07-19 by rhaschke