moveit/moveit_task_constructor #113: Demo fails for modular executable

cpt-yoshi, 2019-11-21:
The **default** modular demo works. There are 2 _SerialContainer_ added to the _Task_. But when adding 8 _SerialContainer_ it fails:
```
Task createTask() {
	Task t;
	t.stages()->setName("Reusable Containers");
	t.add(std::make_unique<stages::CurrentState>("current"));

	const std::string group = "panda_arm";
	t.add(createModule(group));
	t.add(createModule(group));
	t.add(createModule(group));
	t.add(createModule(group));
	t.add(createModule(group));
	t.add(createModule(group));
	t.add(createModule(group));
	t.add(createModule(group));

	return t;
}
```
The planning works for the first 4 _SerialContainer_ but fails at _SerialContainer_ 5:

[image]

rhaschke, 2019-11-21:
The explanation is simple: Moving five times in a row to the left by 0.2m runs the robot into its limits. Thus the 5th attempt fails, which should be obvious from the visualization of the RobotState in rviz.

closed 2019-11-21 by rhaschke

cpt-yoshi, 2019-11-21:
At the end of each subtasks the robot returns to `moveTo ready` pose. Correct me if i'm wrong but shouldn't the robot be able to do this subtask any number of times because it returns to that pose after each subtask?

rhaschke, 2019-11-21:
You are right. I missed that point. However, looking at the different solutions, you probably will observe that they are drifting into joint limits? This is because the arm is 7DoF and there is no mechanism to explicitly resolve the redundancy. I'm not sure though w/o looking into it.

commit a0705248 (2019-11-28): Enable infinite number of repetitions for "modular" demo

Using a joint-space interpolator to move back to start pose,
we avoid pose drift in the Cartesian loop, which pushes the robot
into its joint limits over time. Fixes #113.