moveit/moveit_task_constructor #122: object collision with itself before and after it is being attached to robot

iamrajee, 2020-01-15:
I'm getting below error while running this [`demo`](https://ros-planning.github.io/moveit_tutorials/doc/moveit_task_constructor/moveit_task_constructor_tutorial.html):
_**Terminal 1 : roslaunch moveit_task_constructor_demo demo.launch**_
Fake execution of trajectory
**Found a contact between 'object' (type 'Object') and 'object' (type 'Robot attached'), which constitutes a collision.** Contact information is not stored.
Collision checking is considered complete (collision was found and 0 contacts are stored)
Trajectory component '10/19 - subsolution 0 of stage 0' is invalid after scene update
Stopping execution because the path to execute became invalid(probably the environment changed)
Fake trajectory execution cancelled
Stopped trajectory execution.
Completed trajectory execution with status PREEMPTED ...

_**Terminal 2 : roslaunch moveit_task_constructor_demo pickplace.launch**_
[ INFO] [1579090525.982145137]: Planning succeded
[ INFO] [1579090525.982788967]: Executing solution trajectory
[ERROR] [1579090529.816725764]: **Task execution failed and returned: ABORTED**
[ INFO] [1579090529.817463899]: Execution complete

Attaching below screenshot of the same here.

[image]

My system information are as:
Ubuntu 18.04
ros melodic

rhaschke, 2020-01-15:
The key problem is that the object seems to exist twice: once as a collision object and once attached to the robot. That shouldn't be the case. Looks like a recent change in MoveIt introduced this bug.
Need to have a deeper look.

rhaschke, 2020-01-15:
Do you use MoveIt binary packages or your own build from master?

iamrajee, 2020-01-15:
> Do you use MoveIt binary packages or your own build from master?

Appreciate your prompt response. I built both Moveit and MTC from a source from the master branch. 

Later, I also tried to install the binary version of Moveit(see below image) and tried the same demo in a new workspace where no source built moveit was there. It still gave the same error.
[image]

iamrajee, 2020-01-15:
Q. While running the demo can I check If I'm using binary or source built Moveit?

iamrajee, 2020-01-16:
> I can reproduce the problem in master, but not with MoveIt's binary releases. So the issue was introduced somewhere in the MoveIt master branch. Hence, I filed an issue there: [ros-planning/moveit#1835](https://github.com/ros-planning/moveit/issues/1835).
> I suggest that you work on the binary releases for now, because I don't have time to debug this issue in more detail now.

Sure. Thank you.

felixvd, 2020-01-21:
> Q. While running the demo can I check If I'm using binary or source built Moveit?

FYI, you can use `rospack list` to check if the package points to your workspace or not.

iamrajee, 2020-02-05:
> > Q. While running the demo can I check If I'm using binary or source built Moveit?
> 
> FYI, you can use `rospack list` to check if the package points to your workspace or not.

Thanks, this helped. I have two related queries as below. I would really appreciate your help.

Q.1) After sourcing devel/setup.bash of a workspace, then running "rospack list" I was able to see the packages from other workspaces as well listed(in addition to the binary package at /opt/ros/melodic/share/). Just wondering if this is normal or something might be wrong?

Q.2) After cloning any git package in the workspace when and why should I use wstool merge? Does this make a package available to use from other workspaces as well?

felixvd, 2020-02-05:
1) Packages from other workspaces which your workspace extends (check by running `catkin config` in the workspace) may also appear in the list of all packages. It might also happen if you work in multiple separate workspaces in general, but I am not confident in this answer.

Either way, I believe there should only be one location for each package. If you mean that the entry for `moveit_core` for example references both `/opt/ros/melodic/share/moveit_core` and `your_workspace/src/moveit/moveit_core`, then I don't know if it's a problem, but on my system I don't get the reference to the installed binary after building from source. After sourcing the workspace, `rospack list` only returns the reference to my that workspace.

2) You only need to use `wstool` if you want to add the entries from a `.rosinstall` file to your workspace (also see [the documentation](http://wiki.ros.org/wstool)). The `.rosinstall` file gives the locations of your package's dependencies so you can install them with the [magic rosdep command](http://wiki.ros.org/rosdep#Install_dependency_of_all_packages_in_the_workspace) mentioned in the tutorial. If you already have the dependencies of a package you downloaded (cloned), you don't strictly need to use `wstool merge` to add the `.rosinstall` file, I suppose.

iamrajee, 2020-02-05:
Thank you very much!

rhaschke, 2020-02-18:
I added a temporary workaround for this in c44d0cac19ae96c042e262350419f91c77a80eb5.

j-petit, 2020-02-27:
I am failing to reproduce the issue on a fresh build with master.

Terminal output from `roslaunch moveit_task_constructor_demo demo.launch`
```
	[ INFO] [1582844110.250936430]: Executing TaskSolution
[ INFO] [1582844110.251858438]: Disabling trajectory recording
[ INFO] [1582844110.359601000]: Fake execution of trajectory
[ INFO] [1582844110.962650807]: Fake execution of trajectory
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
[ INFO] [1582844112.764528702]: Fake execution of trajectory
[ INFO] [1582844113.868055420]: Fake execution of trajectory
[ INFO] [1582844114.671044862]: Fake execution of trajectory
[ INFO] [1582844115.672741223]: Fake execution of trajectory
[ INFO] [1582844116.375126952]: Fake execution of trajectory
[ INFO] [1582844117.478484525]: Fake execution of trajectory
[ INFO] [1582844118.281926114]: Fake execution of trajectory
[ INFO] [1582844119.384487044]: Fake execution of trajectory
[ INFO] [1582844121.235328458]: Completed trajectory execution with status SUCCEEDED ...
```

Terminal output from `roslaunch moveit_task_constructor_demo pickplace.launch`:
```
[ INFO] [1582844540.403111492]: Init moveit_task_constructor_demo
[ INFO] [1582844541.456077510]: Loading task parameters
[ INFO] [1582844541.495915387]: Initializing task pipeline
[ INFO] [1582844541.505183226]: Loading robot model 'panda'...
[ WARN] [1582844541.535273567]: Group state 'ready' doesn't specify all group joints.
[ WARN] [1582844541.535310383]: Group state 'extended' doesn't specify all group joints.
[ INFO] [1582844541.581418793]: Start searching for task solutions
[ INFO] [1582844541.643424375]: Using planning interface 'OMPL'
[ INFO] [1582844541.646641116]: Param 'default_workspace_bounds' was not set. Using default value: 10
[ INFO] [1582844541.647249623]: Param 'start_state_max_bounds_error' was set to 0.1
[ INFO] [1582844541.647762106]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1582844541.648309799]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1582844541.648744284]: Param 'jiggle_fraction' was set to 0.05
[ INFO] [1582844541.649209524]: Param 'max_sampling_attempts' was not set. Using default value: 100
[ INFO] [1582844541.649312026]: Using planning request adapter 'Fix Workspace Bounds'
[ INFO] [1582844541.649332107]: Using planning request adapter 'Fix Start State Bounds'
[ INFO] [1582844541.649344984]: Using planning request adapter 'Fix Start State In Collision'
[ INFO] [1582844541.649358406]: Using planning request adapter 'Fix Start State Path Constraints'
[ INFO] [1582844541.649369861]: Using planning request adapter 'Resolve constraint frames to robot links'
[ INFO] [1582844541.649383469]: Using planning request adapter 'Add Time Parameterization'
[ INFO] [1582844541.663895333]: Planner configuration 'hand' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1582844541.664273305]: hand/hand: Starting planning with 1 states already in datastructure
[ INFO] [1582844541.675255673]: hand/hand: Created 4 states (2 start + 2 goal)
[ INFO] [1582844541.675299342]: Solution found in 0.011130 seconds
[ INFO] [1582844541.677154928]: SimpleSetup: Path simplification took 0.001776 seconds and changed from 3 to 2 states
[ INFO] [1582844541.686288544]: Planner configuration 'hand' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1582844541.686450116]: hand/hand: Starting planning with 1 states already in datastructure
[ INFO] [1582844541.697930202]: hand/hand: Created 8 states (2 start + 6 goal)
[ INFO] [1582844541.697967080]: Solution found in 0.011591 seconds
[ INFO] [1582844541.713152964]: SimpleSetup: Path simplification took 0.015145 seconds and changed from 7 to 2 states
[ INFO] [1582844541.717845472]: Planner configuration 'hand' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1582844541.717977607]: hand/hand: Starting planning with 1 states already in datastructure
[ INFO] [1582844541.729638343]: hand/hand: Created 7 states (2 start + 5 goal)
[ INFO] [1582844541.729695155]: Solution found in 0.011768 seconds
[ INFO] [1582844541.745943089]: SimpleSetup: Path simplification took 0.016210 seconds and changed from 6 to 2 states
[ INFO] [1582844541.757399755]: Planner configuration 'hand' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1582844541.757533267]: hand/hand: Starting planning with 1 states already in datastructure
[ INFO] [1582844541.769058612]: hand/hand: Created 7 states (2 start + 5 goal)
[ INFO] [1582844541.769086570]: Solution found in 0.011604 seconds
[ INFO] [1582844541.785989840]: SimpleSetup: Path simplification took 0.016869 seconds and changed from 6 to 2 states
[ INFO] [1582844541.791653185]: Planner configuration 'hand' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1582844541.791796038]: hand/hand: Starting planning with 1 states already in datastructure
[... cut ...]
      240  -   <-240   ->  -240 / generate place pose
      -  4   -> 12   ->  -  0 / open hand
      -  0   -> 12   ->  -  0 / forbid collision (hand,object)
      -  0   -> 12   ->  -  0 / detach object
      -  0   -> 10   ->  - 10 / retreat after place
    -  0   ->  8   ->  -  8 / move home
[ INFO] [1582844543.755381456]: Planning succeded
[ INFO] [1582844543.756089177]: Executing solution trajectory
[ INFO] [1582844554.565476600]: Execution complete
```
Am I missing something to reproduce the error?

rhaschke, 2020-02-28:
@j-petit, yes please revert https://github.com/ros-planning/moveit_task_constructor/commit/c44d0cac19ae96c042e262350419f91c77a80eb5. We added that as a minimally-invasive workaround in MTC.

j-petit, 2020-02-28:
Thanks for the hint, I missed that the MTC was patched.

iamrajee, 2020-03-14:
Closing as it is fixed via [#1899](https://github.com/ros-planning/moveit/pull/1899).

closed 2020-03-14 by iamrajee