moveit/moveit_task_constructor #128: Property's type doesn't match it's declared type

iamrajee, 2020-02-11:
[**ERROR**] [1581401630.026326889]: Initialization failed: Error initializing stage:
generate grasp pose2: Property: type (N13geometry_msgs12PoseStamped_ISaIvEEE) doesn't match property's declared type (NSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE)

**Getting above error while in below piece of code. Unable to understand its reason? I am trying to perform a task similar to pick place demo for two different panda arm(modified the panda arm urdf and srdf). I got this error after introducing below line1 and line2 in place of line3. Full PickPlace code can be found [here](https://github.com/iamrajee/ws_moveit/blob/master/src/moveit_task_constructor/demo/src/moveit_task_constructor_demo6.cpp) and here is the [srdf](https://github.com/iamrajee/ws_moveit/blob/master/src/two_panda_moveit_config/config/panda.srdf) & [urdf](https://github.com/iamrajee/ws_moveit/blob/master/src/panda/robots/two_panda_arm_hand.urdf.xacro) on two_panda_arm**. I'm using melodic, moveit master branch on ubuntu 18.04.

**CODE:**
`        // ---------------------- Generate Grasp Pose ---------------------- //
		{
			// Sample grasp pose
			auto stage = std::make_unique<stages::GenerateGraspPose>("generate grasp pose2");

			t.properties().exposeTo(stage->properties(), { "eef", "hand", "group", "ik_frame","eef2", "hand2", "group2", "ik_frame2" }); //line1
			stage->properties().configureInitFrom(Stage::PARENT, { "eef", "hand", "group", "ik_frame","eef2", "hand2", "group2", "ik_frame2" }); //line2
                        //stage->properties().configureInitFrom(Stage::PARENT,{ "eef", "hand", "group", "ik_frame","eef2", "hand2", "group2", "ik_frame2" }); //line3

			stage->properties().set("marker_ns", "grasp_pose");
			stage->setPreGraspPose(hand2_open_pose_);
			stage->setObject(object2);
			stage->setAngleDelta(M_PI / 12);
			stage->setMonitoredStage(current_state);  // Hook into current state
			
                      // Compute IK
			auto wrapper = std::make_unique<stages::ComputeIK>("grasp pose IK2", std::move(stage));

			wrapper->setMaxIKSolutions(8);
			wrapper->setMinSolutionDistance(1.0);
			wrapper->setIKFrame(grasp_frame_transform_, hand2_frame_);
			wrapper->properties().configureInitFrom(Stage::PARENT, { "eef", "hand", "group", "ik_frame","eef2", "hand2", "group2", "ik_frame2" });
			wrapper->properties().property("eef").configureInitFrom(Stage::PARENT, "eef2");
			wrapper->properties().configureInitFrom(Stage::INTERFACE, { "target_pose" });
			grasp->insert(std::move(wrapper));
		}
`
[image]

rhaschke, 2020-02-11:
This looks like you are trying to assign the wrong data type (PoseStamped) to a property expecting a std::string. Sorry, I should improve the error reporting and provide the property name and demangled type names.

iamrajee, 2020-02-11:
> This looks like you are trying to assign the wrong data type (PoseStamped) to a property expecting a std::string. Sorry, I should improve the error reporting and provide the property name and demangled type names.

Thank you, I would look into it. Would update you soon.

iamrajee, 2020-02-11:
**Update:** I got the same error while introducing the below line
`stage->properties().set("ik_frame", "panda2_link8");`
I understood the error and could resolve it by directly inheriting property from PARENT stage then reconfiguring it as below
`stage->properties().configureInitFrom(Stage::PARENT, { "ik_frame", "ik_frame2" });
stage->properties().property("ik_frame").configureInitFrom(Stage::PARENT, "ik_frame");`
But I didn't understood why I got it. Shouldn't "ik_frame" property expect the string type insead of PoseStamp? Point me if I'm wrong.

rhaschke, 2020-02-11:
`ik_frame` expects a `PoseStamped` to refer to an arbitrary frame. The convenience method `setIKFrame()` combines the identity transform with the given parent frame name.

iamrajee, 2020-02-11:
> `ik_frame` expects a `PoseStamped` to refer to an arbitrary frame. The convenience method `setIKFrame()` combines the identity transform with the given parent frame name.

Thank you. I understood.

rhaschke, 2020-02-18:
I think, this can be closed. If not, feel free to reopen.
Unfortunately, demangling names programmatically failed for an unknown reason.

closed 2020-02-18 by rhaschke

iamrajee, 2020-02-18:
> I think, this can be closed. If not, feel free to reopen.
> Unfortunately, demangling names programmatically failed for an unknown reason.

Sure

danem, 2020-07-15:
I'm a bit confused by this behavior. For instance, the MoveTo stage provides a method:

    void setGoal(const geometry_msgs::PoseStamped& pose) 

Yet, I'm getting this error with the following code:

    geometry_msgs::PoseStamped ps;
    ps.header.frame_id ="world";
    ps.pose.position.x = 0.4;
    ps.pose.position.y = 0.4;
    ps.pose.position.z = 0.4;

    auto stage = std::make_unique<stages::MoveTo>("to pick", sampling_planner);
    // Fails regardless of whether this line is included
    stage->properties().configureInitFrom(Stage::PARENT);
    stage->setGoal(ps);
    c->add(std::move(stage)); 

Under what conditions can I safely use this function?

rhaschke, 2020-07-15:
@danem, please open a new issue exactly describing your problem. What is the erroneous behaviour you observe?