moveit/moveit_task_constructor #131: How to use parallel container?

iamrajee, 2020-02-17:
I'm want to use a parallel container to execute a very simple task(only MoveTo) for two different panda arm. I had modified urdf and srdf accordingly and I'm able to perform the same task using the serial container(see below) but while doing it with ParallelContainerBase it gave me the compile-time error and while using another wrappers of Parallel Container like Alternatives, Fallbacks & Merger it gave me a runtime error. I'm not sure how to proceed ahead. Any help in this regard will be greatly appreciated.
**Code of simple task is as below:**
`    // ====================== Current State ====================== //
	{
		auto _current_state = std::make_unique<stages::CurrentState>("current state");
		t.add(std::move(_current_state));
	}
	// ====================== test container ====================== //
	{ 
		// auto test_container = std::make_unique<SerialContainer>("test container");  //worked fine !!
		// auto test_container = std::make_unique<ParallelContainerBase>("test container"); //gave compiletime error ??
		auto test_container = std::make_unique<Alternatives>("test container"); //gave runtime error ??
		// auto test_container = std::make_unique<Fallbacks>("test container"); //gave runtime error ??
		// auto test_container = std::make_unique<Merger>("test container");//gave runtime error ??

		// ====================== Move to Home ====================== //
		{
			auto stage = std::make_unique<stages::MoveTo>("move home", sampling_planner);
			stage->setGroup(arm_group_name_);
			stage->setGoal(arm_home_pose_);
			test_container->insert(std::move(stage));
		}
		// ====================== Move to Home 2 ====================== //
		{
			auto stage = std::make_unique<stages::MoveTo>("move home2", sampling_planner);
			stage->setGroup(arm2_group_name_);
			stage->setGoal(arm2_home_pose_);
			test_container->insert(std::move(stage));
		}
		t.add(std::move(test_container));
	}`

**Compile time error is as below:**
**Image version:**
[image]

**Text version:**
`usr/include/c++/7/bits/unique_ptr.h: In instantiation of ‘typename std::_MakeUniq<_Tp>::__single_object std::make_unique(_Args&& ...) [with _Tp = moveit::task_constructor::ParallelContainerBase; _Args = {const char (&)[15]}; typename std::_MakeUniq<_Tp>::__single_object = std::unique_ptr<moveit::task_constructor::ParallelContainerBase, std::default_delete<moveit::task_constructor::ParallelContainerBase> >]’:
/home/rajendra/ws_moveit/src/moveit_task_constructor/demo/src/pick_place_task8.cpp:141:81:   required from here
/usr/include/c++/7/bits/unique_ptr.h:825:30: **error**: invalid new-expression of abstract class type ‘moveit::task_constructor::ParallelContainerBase’
     { return unique_ptr<_Tp>(new _Tp(std::forward<_Args>(__args)...)); }
                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/rajendra/ws_moveit/src/moveit_task_constructor/core/include/moveit/task_constructor/task.h:42:0,
                 from /home/rajendra/ws_moveit/src/moveit_task_constructor/demo/include/moveit_task_constructor_demo/pick_place_task8.h:8,
                 from /home/rajendra/ws_moveit/src/moveit_task_constructor/demo/src/pick_place_task8.cpp:1:
/home/rajendra/ws_moveit/src/moveit_task_constructor/core/include/moveit/task_constructor/container.h:123:7: note:   because the following virtual functions are pure within ‘moveit::task_constructor::ParallelContainerBase’:
 class ParallelContainerBase : public ContainerBase
       ^~~~~~~~~~~~~~~~~~~~~
/home/rajendra/ws_moveit/src/moveit_task_constructor/core/include/moveit/task_constructor/container.h:71:15: note: 	virtual bool moveit::task_constructor::ContainerBase::canCompute() const
  virtual bool canCompute() const = 0;
               ^~~~~~~~~~
/home/rajendra/ws_moveit/src/moveit_task_constructor/core/include/moveit/task_constructor/container.h:72:15: note: 	virtual void moveit::task_constructor::ContainerBase::compute()
  virtual void compute() = 0;
               ^~~~~~~
/home/rajendra/ws_moveit/src/moveit_task_constructor/core/include/moveit/task_constructor/container.h:75:15: note: 	virtual void moveit::task_constructor::ContainerBase::onNewSolution(const moveit::task_constructor::SolutionBase&)
  virtual void onNewSolution(const SolutionBase& s) = 0;
`

**Runtime error is as below:**
**Image version:**
[image]

**Text version:**
`[ERROR] [1581928246.963531562]: Initialization failed: Error initializing stages:
current state: required interface is not satisfied
test container: start interface of 'move home' (???) doesn't match mine (?)
test container: start interface of 'move home2' (???) doesn't match mine (?)
test container: no child provides expected push interface
`

**Other Information is as below**
-> I'm using master branch of MTC and moveit in ros melodic on ubuntu 18.04.
Below is the two arm setup of moveit:
[image]

rhaschke, 2020-02-17:
`ParallelContainerBase` is an abstract base class and thus cannot be instantiated -> comiler error
That the others don't work seems to be a bug. Could you provide your C++ code in a gist?

iamrajee, 2020-02-17:
Thank you for your prompt response. Please find the gist [here](https://gist.github.com/iamrajee/5b11811e8eecd28f390c3c01cd668697).

rhaschke, 2020-02-17:
Looks like the container's interface resolution isn't perfect yet. The culprit are the `MoveTo` stages, which - in principle - can propagate in both directions, forward and backward.
You need to explicitly provide a hint to help interface resolution like this:
```C
auto stage = std::make_unique<stages::MoveTo>("move home", sampling_planner);
stage->restrictDirection(stages::MoveTo::FORWARD);
```
I'm leaving this issue open as a reminder to continue working on it later.
The culprit is a ParallelContainer as the single child within a SerialContainer.

iamrajee, 2020-02-17:
Thank you very much. I can confirm that by restricting the direction as you mentioned above, its working.

iamrajee, 2020-02-18:
Q. **How to work with the nested container?**
I want to run four MoveTo stages. I tried to put two serial containers each one containing two MoveTo stages in a parallel container(Merger), but it generates compile error that merger class has no member named 'add'. When I tried to replace this add with insert then also it gave the same error. Code of stages can be found [here](https://gist.github.com/iamrajee/6a36972f70be5e62143785e96b37224f).

rhaschke, 2020-02-18:
`merger->insert(std::move(stage))` should work. What's the error message you get?

iamrajee, 2020-02-18:
> `merger->insert(std::move(stage))` should work. What's the error message you get?
**Error**
error: ‘class std::unique_ptr<moveit::task_constructor::Merger, std::default_delete<moveit::task_constructor::Merger> >’ has no member named ‘insert’; did you mean ‘reset’?
    test_container.insert(std::move(test_container2));

rhaschke, 2020-02-18:
Please, read your error messages more carefully. It's exactly written that the pointer doesn't have an insert method. So, you should replace
`test_container.insert()` → `test_container->insert()`.

iamrajee, 2020-02-18:
I'm sorry for not paying close attention to the error. I would make sure I don't repeat the same in future. 
However, this solved the compile-time error but planning failed with error
**[ERROR] [1582010021.064243764]: Only simple trajectories are supported**
I'm not sure as to what trajectory are considered to be simple and what not. It would help, If you can give me a hint on the same.
[image]

rhaschke, 2020-02-18:
It's not possible to merge trajectories generated by a serial container.
Consider some substages in the serial container that just modify the planning scene (e.g. attaching/detaching an object, or modifying the ACM). How should we merge such a modification into another motion trajectory?
If you want to operate two independent arms in parallel, you should just consider the robots as independent and create two separate tasks for them.

iamrajee, 2020-02-18:
I really understood what you trying to convey. I would create two separate tasks(not sure how?) for both arms and try to perform the above task with it. Thanks!

closed 2021-03-22 by v4hn