moveit/moveit_task_constructor #132: Joint '' not found in model 'my_scene'

karolyartur, 2020-02-17:
I got the following error wile experimenting with the Python API's customized version of `cartesian.py`:

`Joint '' not found in model 'my_scene'`

when trying to use moverelative with joint offsets and a Cartesian planner like this:

```python
move = stages.MoveRelative("joint offset", cartesian)
move.group = group
move.setDirection({'b_bot_shoulder_lift_joint': pi/6, 'b_bot_wrist_1_joint': -pi/6})
task.add(move)
```
In the custom 'my_scene' scene I would like to use a UR5 robot with the Python API of MTC. Currently it is working with the C++ API, but using the Python API I always get the above error.

One thing I realized, is that if the specified joint name is longer than 15 characters it is not displayed properly in the error message, so for example using

```python
move.setDirection({'b_bot_shoulder_': pi/6})
```

results in an error message:

`Joint 'b_bot_shoulder_' not found in model 'my_scene'`

and using

```python
move.setDirection({'b_bot_shoulder_l': pi/6})
```

results in an error message:

`Joint '??[ZUV' not found in model 'my_scene'`

rhaschke, 2020-02-17:
Which version (commit id) of MTC are you using?
I recently fixed a bug regarding this issue. Did you update? Be aware of force-pushes.

karolyartur, 2020-02-18:
Thank you for the fast reply.
The latest commit is 569836e1 on branch `wip-python-api` in the version that I have.

rhaschke, 2020-02-18:
It turned out that you found an actual bug. I have fixed this issue in d10565cabfae07f466143c02830444b01c958f93 and pushed to the branch wip-python-api.

commit d10565ca (2020-02-18): fix memory access issue

Fixes #132.
closed 2020-02-18 by rhaschke