moveit/moveit_task_constructor #134: Unregistered type error while running cartesian.cpp

iamrajee, 2020-02-19:
I'm getting unregistered type error while running cartesian demo as

**Terminal1:** roslaunch moveit_task_constructor_demo demo.launch
**Terminal2:** rosrun moveit_task_constructor_demo cartesian
`Unregistered type: St3mapINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEdSt4lessIS5_ESaISt4pairIKS5_dEEE`
[image]

Although I can still run the demo but just wondering why I got this error and which object have this weird looking type?

rhaschke, 2020-02-19:
c++filt reveals the demangled type (its `std::map<std::string, double>`)
```
c++filt -t St3mapINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEdSt4lessIS5_ESaISt4pairIKS5_dEEE
std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, double, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, double> > >
```
In a later version, I already downgraded the error to a warning. You can safely ignore it.
It's just the information that a property cannot be transmitted to rviz for inspection.

closed 2020-02-19 by rhaschke

iamrajee, 2020-02-19:
> c++filt reveals the demangled type (its `std::map<std::string, double>`)
>
> ```
> c++filt -t St3mapINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEdSt4lessIS5_ESaISt4pairIKS5_dEEE [...]

Sure, Thank you.