moveit/moveit_task_constructor #135: How to create multiple Task?

iamrajee, 2020-02-20:
I'm trying to create two pick & place tasks for two different panda arm(same srdf, different planning group) as suggested in [#131 comment](https://github.com/ros-planning/moveit_task_constructor/issues/131#issuecomment-587322319). 

Firstly, I tried by calling this(similar to what is being done in [cartesian.cpp](https://github.com/ros-planning/moveit_task_constructor/blob/c44d0cac19ae96c042e262350419f91c77a80eb5/demo/src/cartesian.cpp#L130))
`task = createTask();
task.plan();
task.introspection().publishSolution(*task.solutions().front());
` twice in my code, but with this last task has overwritten the /node_name/solutions/ topic, and hence I was only able to visualise the last called task only. 

Then, I also tried this
Terminal 1: `roslaunch moveit_task_constructor_demo demo.launch`
Terminal 2: `rosrun moveit_task_constructor_demo cartesian`
Terminal 3: `rosrun moveit_task_constructor_demo cartesian2` (this one had different node name)
this way I was able to visualise planning of both arm simultaneously(by adding one more MTC viewer & clicking reset in rviz), but here how can I take collision between two arms in consideration. Also, I feel this might be not the right way to work with multiple arms. **What is the best way to work with multiple arms with MTC for both parallel and sequential task?**

**Is there a way to call the two different task from the same node and publish on two different solution topic in rviz topics?**

Also, how can we execute two different tasks (both involving different planning group) in particular manner e.g parallelly, serially etc same as what with do with stages with help of a container?

rhaschke, 2020-02-20:
The `Task` class has an optional `id` parameter, which allows differentiating the two tasks at the ROS topic level. This `id` is actually a ROS namespace.
How to handle multiple arms, strongly depends on their independence. If they cannot collide, they can be handled independently. If they can, you actually need to consider the combined configuration space if they should move simultaneously. Using a `Merger` is not feasible in this case, because potential collisions will not be resolved.
Alternatively, you can plan for the arms sequentially.

iamrajee, 2020-02-20:
Thank you. I tried something like this 
```
Task createTask(std::string s) {
	Task t;
	 t.id() = s;
         t.stages()->setName(s);
         //... remaining code ...//
         return t;
}

int main(){
        Task t;
        s = "Cartesian Path";
	t = createTask(s);
	// std::cout<<"t.id()="<<t.id()<<"\n";        // it doesn't print id here??
        t.plan();
        t.introspection().publishSolution(*t.solutions().front();

       Task t2;
        s = "Cartesian Path2";
	t2 = createTask(s);
	// std::cout<<"t2.id()="<<t2.id()<<"\n";        // it doesn't print id here??
        t2.plan();
        t2.introspection().publishSolution(*t2.solutions().front();
}
```
But still `Task t2` overwrite `Task t`. 

> If they can, you actually need to consider the combined configuration space if they should move simultaneously.
 How can we do that?

v4hn, 2020-02-20:
```
	Task t;
	 t.id() = s;
```

This should be 
```
       Task t(s);
```

iamrajee, 2020-02-20:
> ```
> 	Task t;
> 	 t.id() = s;
> ```
> 
> This should be
> 
> ```
>        Task t(s);
> ```

Thank you. This helped.

iamrajee, 2020-02-25:
I'm closing this issue as it is solved now. Thank you, Michael and Robert, for your help.

closed 2020-02-25 by iamrajee

YannickRiou, 2020-07-17:
When you create multiple task, don't you have problem with the planning scene being updated to last executed task ?  
I made a simple example of PR2 that execute a pick of "obj_0" with left hand, then a pick of "obj_1" with right hand. After the end of the second pick, only obj_1 is shown in the planning scene. obj_0 just disappear. 

Here is the example : https://gist.github.com/YannickRiou/8b47e304ef9e267f5f2de77c3ee4bb2f 

And the result of execution : 
[image]

The fact that obj_0 disappeared is confirmed by trying to do a place after the two picks. Moveit just answer that attached "obj_0" isn't found.

rhaschke, 2020-07-17:
Of course, you need to be careful regarding the order of planning these multiple tasks. If you plan both in parallel on a common PlanningScene, but then execute one of them, thus modifying the PlanningScene, the second plan obviously becomes invalid.

YannickRiou, 2020-07-17:
Ok I see, I understood that for a pick and place on the same object, but didn't thought this was the same even with two different planning group and object but it now makes sense (when the right pick is executed it didn't know about the left object was picked). Thanks for this enlightenment.