moveit/moveit_task_constructor #137: Uable to execute task

iamrajee, 2020-02-24:
I'm running cartesian task using for **two panda arm** in a similar way done here in the demo. Two though I can plan the task but **cannot execute** it as it doesn't return(I think stuck at waitForResult). Yes, I have manually included the changes made in PR #136 and confirm that task execution **works perfectly** and also return back **for singe arm**. 
**Could it because I forgot to add some controller/planner in setup assistance while creating the srdf of these two panda arms?**
I'm creating, planning and executing task as below
```
        s = "Task1";
        Task t(s);
        t.stages()->setName(s);

        //defined the task here

	try {
		if (t.plan()){
                        t.enableIntrospection();
			// t.introspection().publishSolution(*t.solutions().front()); //automatic display plan

			moveit_msgs::MoveItErrorCodes execute_result = t.execute(*t.solutions().front()); //stuck at this line and dont proceed further

			if (execute_result.val != moveit_msgs::MoveItErrorCodes::SUCCESS) {
				ROS_INFO_NAMED(LOGNAME, "Task t execution Failed!");
			}else{
				ROS_INFO_NAMED(LOGNAME, "Task t execution completed!");
			}
        }
	} catch (const InitStageException& ex) {
		std::cerr << "planning failed with exception" << std::endl << ex << t;
	}
```

rhaschke, 2020-02-24:
Please check the output of the move_group node for hints regarding the stuck execution.

iamrajee, 2020-02-24:
Thank you. On a closer look at move_group output, I realised that it is missing "**ExecuteTaskSolution**" capability. I found out that by default setup assistance has introduced this four-line in newly generated move_group.launch for my custom robot as below
```
<arg name="capabilities" default=""/>
<arg name="disable_capabilities" default=""/>
<param name="capabilities" value="$(arg capabilities)"/>
<param name="disable_capabilities" value="$(arg disable_capabilities)"/>
```
These line were overwriting the move group capabilities by non-move group capabilities.
Although in my main demo.launch I had define `capabilities` as
`<param name="move_group/capabilities" value="move_group/ExecuteTaskSolutionCapability" />`
but it does seem to overwrite the `capabilities` parameter defined in move_group.launch

I'm above four-line in move_group.launch.

v4hn, 2020-02-24:
>  but it does seem to overwrite the capabilities parameter defined in move_group.launch

The current take of the `move_group.launch` is that you pass additional capabilities **as arguments** to the launch file and do not define the parameters on your own.
You can do that too, but I believe you have to ensure to define the parameter *after* you include the `move_group` launch.

closed 2020-02-24 by v4hn

iamrajee, 2020-02-24:
> The current take of the `move_group.launch` is that you pass additional capabilities **as arguments** to the launch file and do not define the parameters on your own.
> You can do that too, but I [...]

Sorry, there was a typo in the previous reply. I meant "but it doesn't..." instead of "it does....".
I understood what you trying to tell me. Thank you.