moveit/moveit_task_constructor #138: How to execute multiple task simonteneously?

iamrajee, 2020-02-25:
I can execute two independent tasks on two separate panda arm serially. Now I want to execute them simonteneously. I have an blocking execute_helper function of type
```
moveit_msgs::MoveItErrorCodes execute_helper(moveit::task_constructor::Task& t) {
 //execute task here and return only after task execution complete
}
```
How can I call it twice with different inputs? I tried something as below but I dont know how can I push my costum callback function on `GlobalCallbackQueue` to be considered by `spinner.start()`
```
ros::AsyncSpinner spinner(2);
spinner.start();
execute_helper(task1);
execute_helper(task2);
ros::waitForShutdown();
```
Alternatively, how to push costum callback function in my constum queue to be used as
```
#include <ros/callback_queue.h>
ros::CallbackQueue my_callback_queue;
//how to push execute_helper in my_callback_queue
ros::AsyncSpinner spinner(0, &my_callback_queue);
spinner.start();
ros::waitForShutdown();
```
How can I decide which spinner to be used here, i.e MultiThreadedSpinner or AsyncSpinner?

rhaschke, 2020-02-25:
To call your two execute_helpers independently, you can just use two threads directly.
But, even if you manage this, I don't think, a single move_group node can handle two execution requests in parallel. As the corresponding capability relies on a `SimpleActionServer` the following doc applies:

> only one goal can have an active status at a time, new goals preempt previous goals based on the
> stamp in their GoalID field (later goals preempt earlier ones)

iamrajee, 2020-02-25:
> To call your two execute_helpers ...
Thank you I understood. Can we change this preempt behaviour of action goal? 
If not in ROS1, is it possible to do multiple executions at the same time in ROS2 move_group?

v4hn, 2020-02-26:
On Tue, Feb 25, 2020 at 12:27:02AM -0800, Rajendra Singh wrote:
> > To call your two execute_helpers ...
> Thank you I understood. Can we change this preempt behaviour of action goal? 

This is a matter of changing the ExecuteTaskSolution capability, at least, to a general `ActionServer`.
This requires additional bookkeeping, probably a similar transition in general plan execution in MoveIt
and would basically "only" add support for your current use-case where you want to execute independent controllers.

Of course, you're welcome to provide a pull-request that achieves this behavior, but the more reasonable
solution for yourself might be to run two independent `PlanExecution` classes locally, or even execute the subtrajectories of the solutions yourself
by sending them to the correct `FollowJointTrajectory` actions. This is of course not very elegant, though...

> If not in ROS1, is it possible to do multiple executions at the same time in ROS2 move_group?

The matter is independent of ROS1/2.
Of course, there is probably interested parties who would implement it there, given dedicated funding. :)

iamrajee, 2020-02-26:
> This is a matter of changing ...
Thank you Michael for such a helpful explanation. I would work on the possible solutions suggested here.

closed 2021-03-22 by v4hn

felixvd, 2021-03-22:
Just want to mention that executing multiple trajectories is something I'd love to see implemented, it's on the [future projects page](https://moveit.ros.org/documentation/contributing/future_projects/#simultaneous-trajectory-execution) and there's been discussion in https://github.com/ros-planning/moveit/issues/2287.

iamrajee, 2021-03-22:
> Just want to mention that executing multiple trajectories is something I'd love to see implemented, it's on the [future projects page](https://moveit.ros.org/documentation/contributing/future_projects/#simultaneous-trajectory-execution) and there's been discussion in [ros-planning/moveit#2287](https://github.com/ros-planning/moveit/issues/2287).

Sure, thanks for the update @felixvd

v4hn, 2021-03-22:
> Just want to mention that executing multiple trajectories is something I'd love to see implemented, it's on the [future projects [...]

Indeed, and I'm not saying it will never be there.
But in most cases described by users who request it, there are already better ways to handle this outside one move_group instance.