moveit/moveit_task_constructor #139: How to use GenerateGraspPose state in Python API?

karolyartur, 2020-02-28:
I've been trying to recreate [pick_place_task](https://github.com/ros-planning/moveit_task_constructor/blob/wip-python-api/demo/src/pick_place_task.cpp) using the Python API and I have the following problem, when I try to set the monitored stage for the GenerateGraspPose state:

The PredicateFilter stage is not implemented in the python wrapper, so I cannot wrap the CurrentState stage like in the cpp example. However, if I add it to the task without wrapping it, and then try to set it as monitored stage for the GenerateGraspPose state I get the following error:

```bash
Boost.Python.ArgumentError: Python argument types in
    MonitoringGenerator.setMonitoredStage(GenerateGraspPose, CurrentState)
did not match C++ signature:
    setMonitoredStage(moveit::task_constructor::MonitoringGenerator {lvalue}, moveit::task_constructor::Stage*)
```

What is the correct way to set a monitored stage for the GenerateGraspPose state in the Python API?

Here is the brief task definition so far:

```python
    task = core.Task()

    task.properties.update({'group': robot_name, 'eef': group.get_end_effector_link(), 'hand': hand_name, 'hand_grasping_frame': hand_name, 'ik_frame': hand_name})

    currstate = stages.CurrentState('current state')
    task.add(currstate)  # Adding it to the task results in error for argument types in setMonitoredStage in GenerateGraspPose

    open_hand = stages.MoveTo("open hand", sampling_planner)
    open_hand.group = hand_name
    open_hand.setGoal('open')
    task.add(open_hand)

    connect = stages.Connect('move to pick', [(robot_name, sampling_planner)])
    connect.timeout = 5
    connect.properties.configureInitFrom(core.PARENT)
    task.add(connect)

    grasp = core.SerialContainer('pick object')
    task.properties.exposeTo(grasp.properties, ['eef', 'hand', 'group', 'ik_frame'])
    grasp.properties.configureInitFrom(core.PARENT, ['eef', 'hand', 'group', 'ik_frame'])

    approach_object = stages.MoveRelative("approach_object", cartesian_planner)
    approach_object.properties.update({'marker_ns': 'approach_object', 'link': hand_name})
    approach_object.properties.configureInitFrom(core.PARENT, ['group'])
    approach_object.min_distance = 0.01
    approach_object.max_distance = 0.1
    print(approach_object.properties.__getitem__('group'))   # Why is this None? how to get properties from within SerialContainer?
    approach_object.setDirection(geometry_msgs.msg.TwistStamped(header=std_msgs.msg.Header(frame_id = hand_name), twist=geometry_msgs.msg.Twist(linear=geometry_msgs.msg.Vector3(0,0,0.1))))
    grasp.insert(approach_object)

    generatepose = stages.GenerateGraspPose('generate grasp pose')
    generatepose.properties.configureInitFrom(core.PARENT)
    generatepose.properties.update({'marker_ns': 'grasp_pose'})
    generatepose.pregrasp = 'open'
    generatepose.object = 'base'
    generatepose.angle_delta = pi/12
    generatepose.setMonitoredStage(currstate)

    # To be continued
    task.add(grasp)
```
I am using moveit task constructor built from source at a97d860d commit of the wip-python-api branch. Any help would be appreciated.

rhaschke, 2020-02-28:
Unfortunately, `MonitoringGenerators` are not yet fully supported in Python, because it is not possible to pass a stage instance from Python to C++ _twice_ as it is required here:
1. the stage is passed to the task and
2. the same stage is passed to the MonitoringGenerator.

The reasons for this limitation are:
- our design decision to use unique_ptrs for stages in C++ (to indicate that ownership is transferred when passing a stage into a container)
- This already in C++ requires to keep a _raw_ pointer as a reference to a stage to pass it into the MonitoringGenerator as well.
- The python wrapper cannot handle raw pointers by design. Essentially, it needs to use a reference-counted pointer to know when to release it.

We discussed in the past to switch to shared pointers. However, aside from being less efficient, this will not reflect the fact anymore that a single stage can be owned by a single container only.
For this reason, I think a better solution is to introduce the concept of a stage reference. While containers, taking ownership of a passed (sub)stage, will still expect `unique_ptr<Stage>` arguments, MonitoringGenerators would take a reference instead. This concept of stage references could be mapped to python as well then.
As I would like to transition from boost::python to pybind11 (actually, [I already prepared this already](https://github.com/ubi-agni/moveit_task_constructor/tree/wip-pybind11)), I didn't put effort into the current boost::python wrapper anymore. However, the transition requires to overcome some limitations in pybind11. I have opened corresponding PRs against pybind11 already, but there are in discussion...
So to summarize, there will be no solution to this problem available in the very next future. Sorry.

karolyartur, 2020-03-02:
Thank you for the detailed response. Fortunately it is not strictly necessary for me to use the Python API, but I thought I will give it a try since I am more confident in Python than in C++.

closed 2020-03-02 by karolyartur