moveit/moveit_task_constructor #145: Linker error "undefined reference"

iamrajee, 2020-03-16:
### Environment
ROS Distro: [eloquent]
OS Version: e.g. Ubuntu 18.04
Binary build
distribution status: active
moveit2: cloned from ros-planning
/moveit2(master branch) at this [state](https://github.com/ros-planning/moveit2/tree/6a23d3284d6d1e1d02347612da348b01519e8745).

### Description
I'm trying to create MTC for ros2 eloquent. I'm currently at this [stage](https://github.com/ros-planning/moveit_task_constructor/commit/0e5d39a94d820254d50ea6e98a5ee723a44c8e0a). After introducing `rclcpp::init(argc, argv);` in [test_plan_current_state.cpp](https://github.com/iamrajee/ros2eloquent_moveit_ws/blob/master/src/moveit_task_constructor/src/test/test_plan_current_state.cpp#L15) I start getting below linker error:

```
CMakeFiles/test_plan_current_state.dir/src/test/test_plan_current_state.cpp.o: In function `main':
test_plan_current_state.cpp:(.text+0xd6): undefined reference to `rcutils_get_default_allocator'
test_plan_current_state.cpp:(.text+0xf8): undefined reference to `rclcpp::InitOptions::InitOptions(rcutils_allocator_t)'
test_plan_current_state.cpp:(.text+0x11a): undefined reference to `rclcpp::init(int, char const* const*, rclcpp::InitOptions const&)'
test_plan_current_state.cpp:(.text+0x129): undefined reference to `rclcpp::InitOptions::~InitOptions()'
test_plan_current_state.cpp:(.text+0x200): undefined reference to `rclcpp::shutdown(std::shared_ptr<rclcpp::Context>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
test_plan_current_state.cpp:(.text+0x26a): undefined reference to `rclcpp::InitOptions::~InitOptions()'
collect2: error: ld returned 1 exit status
```

I tried a few ways to resolve it, but couldn't. Any help would be greatly appreciated.

rhaschke, 2020-03-16:
Both core developers of MTC are not yet using ROS2. Thus we cannot really help yet.
The linker error reported indicates that you are missing to link the rclcpp library. Did you adapted the cmake files accordingly?

iamrajee, 2020-03-16:
> Both core developers of MTC are not yet using ROS2. Thus we cannot really help yet.
> The linker error reported indicates that you are missing to link the rclcpp library. Did you adapted the cmake files accordingly?

Alright. Yes, I did. Could you please have look at my [CMakeLists.txt](https://github.com/iamrajee/ros2eloquent_moveit_ws/blob/master/src/moveit_task_constructor/CMakeLists.txt), tell me you spot something wrong in it?

iamrajee, 2020-03-17:
I realised that I had forgotten to add target dependencies in `CMakeLists.txt` as
`ament_target_dependencies(executable_name rclcpp)`. After adding the same I could resolve the issue.

closed 2020-03-17 by iamrajee

rhaschke, 2020-03-17:
Great.