moveit/moveit_task_constructor #150: Setting path constraints for stages

karolyartur, 2020-03-23:
While setting up a custom scene that I made out of the MTC pickplace demo I encountered the following problem:

When I set path constraints for the MoveRelative stages, the resulted trajectory in the solutions seems to jump at some joints. A very dirty working example of the issue can be found at my [fork](https://github.com/karolyartur/moveit_task_constructor) in the benchmarking branch.
The problematic part is introduced in the latest commit (c05fa72) in the `move_relative.cpp` file.
In the trajectory printed by `compute_brenchmark.py` it can be seen that at the approach, lift etc. stages the fifth joint of the robot is skipping.

This behavior does not appear in the visualization.
Where is the actual solution published? Where does the visualization plugin get it from?
Why is there much less waypoints in the trajectory if a path constraint is set?

Thank you for your response and once again sorry for the very dirty example, I just need an urgent solution even if it is dirty.

v4hn, 2020-03-23:
The *specification* of the path constraints is not the crucial point here, neither is the `MoveRelative` stage. Its *solver* has to succeed with the constraint.

MoveIt's pipeline planner for OMPL does not cope with path constraints very well and can generate paths with joint discontinuities. Patches are welcome.

Previously, I worked around this [by rejection sampling in joint space](https://github.com/ros-planning/moveit/blob/master/moveit_planners/ompl/ompl_interface/src/planning_context_manager.cpp#L478) for coarse constraints, [joint space approximations](https://github.com/ros-planning/moveit/blob/master/moveit_planners/ompl/ompl_interface/launch/generate_state_database.launch), or custom planners where feasible.

v4hn, 2020-03-23:
Closing as the observed problem is in MoveIt's OMPL plugin.

closed 2020-03-23 by v4hn