moveit/moveit_task_constructor #158: demo->cartesian

Reaper151, 2020-04-12:
I tried to integrate moveit_task_controller to our robot and when I ran cartesian demo it provide solution on +x task because there are waypoints in "MoveIt Planning Tasks-Slider" and it can animate but the states fell under "fail solution attempts" and console held at bottom line of this logs 

`rosrun moveit_task_constructor_demo cartesian
[ INFO] [1586695240.930185694]: Loading robot model 'sia5'...
[ INFO] [1586695241.504806751]: cache file /home/reaper/sia5arm_base_linktool_ee_link_5000_1.000000_1.000000.ikcache initialized!
  0  -   <-  0   ->  -  0 / Cartesian Path
    1  -   <-  1   ->  -  0 / initial state
    -  0   ->  0   ->  -  0 / x +0.2
    -  0   ->  0   ->  -  0 / y -0.3
    -  0   ->  0   ->  -  0 / rz +45°
    -  0   ->  0   ->  -  0 / open gripper
    -  0   ->  0   <-  1  - / connect
    1  -   <-  1   ->  -  1 / final state`

iamrajee, 2020-04-12:
> have no idea why cartesian interpolation and pickplace demo not working.

Can you provide more information with error message?

Reaper151, 2020-04-13:
> > have no idea why cartesian interpolation and pickplace demo not working.
> 
> Can you provide more information with error msg?

That's the problem I don't see any error logs in console.

iamrajee, 2020-04-13:
> That's the problem I don't see any error logs in the console.

Please provide more information like ROS distro, your system and whether your moveIt is binary or source built?
Also, provide the steps to reproduce this issue and screenshot of rviz MTC task panel.

Reaper151, 2020-04-13:
Problem solved. 
It looks like to be a physical limitation of my robot arm that can't do the task but I frustrated at first because it can be animate and arm is following the motion. 
PS. ROS Melodic and moveIt was built by source

Now I resume to get pickplace works but I stuck at "grasp_pose_IK" for a while.
[image]

Reaper151, 2020-04-13:
@iamrajee if I want to discuss about stuck at "grasp_pose_IK" in demo->pickplace should I open new issue?

iamrajee, 2020-04-13:
@Reaper151 Common cause for IK failure is when the object is unreachable, but assuming this is not the case with you as you've already experimented with various object_pose. 
Although we have forbidden the collision between gripper and object before picking in the demo, it may so happen that some other part of the robot could be in collision with it while picking thus resulting in IK failure. You can increase the distance between gripper and object while picking by changing `grasp_frame_transform`. e.g for panda arm you can do something like
```
grasp_frame_transform: [0, 0, 0.2, 1.571, 0.785, 1.571] #here 0.2 is that distance
```
Of course for your robot this could be different. If the problem still persists then surely go ahead and open a new issue.

v4hn, 2020-04-14:
From the picture it still looks like the requested pose might not be reachable.
Can you move your arm by hand (or in simulation by slider) to the pose you want to reach?

Are the failing solutions associated with a comment (extend the right listing of individual solutions to see the comments section)?

Reaper151, 2020-04-14:
All it need to do is just move back a little bit less than 0.2m in cartesian demo that i succeed.
[image]

v4hn, 2020-04-15:
I'm not sure how to support you at all right now.

To simplify the task, you can reduce it to a simple 
`CurrentState` - `Connect` - `grasp pose IK (generate grasp pose)` structure.

But your problem remains that IK fails for the requested pose I guess.
This is either because the pose is indeed not reachable, your IK solvers are incomplete, or because of an unexpected collision you did not consider (might be an accidentally allowed collision between unrelated links of the robot).

Reaper151, 2020-04-15:
@v4hn  For what it might worth, I have succeed in mitigrate MoveIt Grasp Demo.
I will recheck allowed collision again is there any turorial that will check and visualize the collision before move?

v4hn, 2020-04-15:
On Wed, Apr 15, 2020 at 05:58:55AM -0700, Reaper151 wrote:
> @v4hn  For what it might worth, I have succeed in mitigrate MoveIt Grasp Demo.
> I will recheck allowed collision again is there any turorial that will check and visualize the collision before move?

No there isn't, but you can try to set the `ignore_collisions` property (setIgnoreCollisions) to false to ensure that's not the problem.

Reaper151, 2020-04-16:
Ah. It's collision problem as I set the `ignore_collisions` property (setIgnoreCollisions) to true, then pipeline is complete.

Does computeIK refer to 'disble_collisions' tags in SRDF because I found one pair of link that collided but already has 'disable_collisions' by 'adjacent' reason in SRDF but problem remained?

rhaschke, 2020-04-16:
The allowed collision matrix (ACM) defined in the SRDF should be considered by ComputeIK. I guess there is another collision as well. If you increase the width of the solution view, you will see a comment explaining the reason of failure:
[image]

Also, you may notice that the visualization of the end effector changes from visual to collision geometries depending on the error type (i.e. IK failure vs. collision).

Reaper151, 2020-04-16:
Thanks problem solved it's other collision cause by my workspace. Appreciate for all of your help.
This is conclusion.

> The allowed collision matrix (ACM) defined in the SRDF should be considered by ComputeIK. I guess there is another collision as well. If you increase the width of the solution view, you will see a [...]

Anyway do you have any plan for wrapper of pickplace in python?

rhaschke, 2020-04-16:
I do have a lot of plans for python, but no time to implement them...
If this issue is solved, please close it, @Reaper151.

v4hn, 2020-04-16:
>  This is conclusion.

Closing.

> Anyway do you have any plan for wrapper of pickplace in python.

Robert has work in progress for usable python bindings that would allow for it, but time is short.

closed 2020-04-16 by v4hn

Reaper151, 2020-04-16:
I'm looking forward to see it!!!
Cheers