moveit/moveit_task_constructor #159: ComputIk: long waiting time between each failed attempt

cpt-yoshi, 2020-04-13:
Presented in the first GIF, is a motion task that takes roughly 50 seconds to find a solution.

The problem is that the the last ComputeIk stage, see second GIF, takes 36 seconds before finding a solution. But it takes only 0.6 seconds of processing time to find a solution for this stage. There are 13 failed attemps because of collision or no valid IK which is normal because of my setup. Between 2 failed attempts, there is ~3 seconds lost (36 seconds lost in total). I have tried tuning the timeout propertie as low as possible without success...

Setup configuration:
- RRTConnect
- Default Cartesian properties
- Stage ComputeIk properties: **timeout** = 0.1, **max_ik_solutions**=8
- All stages were added to the same SerialContainer
- kinematic_solver: kdl_kinematics_plugin
- Disabling constraints on Connect stages didn't help.

Is there something I'm missing? Thank you for your time and have a happy 2020 easter.

[image]

[image]
_You can click on the second GIF for details about the properties._

v4hn, 2020-04-14:
That's quite a nice Task you setup there! :+1: 

> The problem is that the the last ComputeIk stage, see second GIF, takes 36 seconds before finding a solution

This is not really the case, as you said yourself. The `ComputeIK` stage is actually fast.
The main problem here is that the task applies plain round-robin scheduling for the stages and it probably takes roughly 3 seconds to round trip and have the stage process the next pose target.

Three ways to improve things:
- Why do you abuse a GraspPoseGenerator if you already know the exact/rough pose you need for this IK request? A `PoseGenerator` or a custom generator spawning relevant states is clearly better
- Change the ComputeIK stage to process all its `upstream_solutions_` before returning from `compute`
- change the scheduling algorithm altogether

cpt-yoshi, 2020-04-17:
> Why do you abuse a GraspPoseGenerator if you already know the exact/rough pose you need for this IK request? A PoseGenerator or a custom generator spawning relevant states is clearly better

For this example, you are right. This last pose should have a constraint in orientation wrt gravity so that the object at the tip falls into a bin (Not shown in the image).

But I would have the same corner case, if I were to try and grasp an object that is only attainable at a certain angle. The GraspPoseGenerator is then necessary (It can help a lot depending on the setup) to avoid singularities. 
> Change the ComputeIK stage to process all its upstream_solutions_ before returning from compute

Bingo! Down from 50 seconds to ~16 seconds. That fixes the issue. Shoulden't we have a parameter in ComputeIk stage that modifies the `upstream_solution_` behavior:
1. Default.
2. Process all upstream_solutions_ until a valid IK is found.
3. Proces all upstream_solutions_ .

> change the scheduling algorithm altogether

Is the round-robin scheduler in `task.cpp`?

[image]

Big THX to @v4hn !!

v4hn, 2020-04-17:
Glad to help. Feel free to send me a video of your task execution to return the favor. :sunglasses: 

> Shoulden't we have a parameter in ComputeIk stage that modifies the upstream_solution_ behavior

In my personal opinion, this stage is already way too complex and should be rewritten...
Adding more options to internal behavior just makes it worse... Feel free to propose a patch if you really want to have that property available.

> Is the round-robin scheduler in task.cpp?

It's not quite as simple, because the task [forwards the scheduling to its container](https://github.com/ros-planning/moveit_task_constructor/blob/master/core/src/task.cpp#L285) which usually is [a Serial container iterating over its children](https://github.com/ros-planning/moveit_task_constructor/blob/master/core/src/container.cpp#L524).
At the moment scheduling is done locally by the containers which is not adequate but works.
I am going to rework scheduling to a job-based system, but that takes time, requires major reworking of the APIs and agreement with my co-maintainer. ;)

I'll close this for now.

closed 2020-04-17 by v4hn