moveit/moveit_task_constructor #169: MoveTo with cartesian solver returns error

YannickRiou, 2020-06-03:
ROS Version : Kinetic
Robot : PR2 

When trying to use cartesian space solver to move the arms from current state to home position, I get this error : 
`  -  -   --  0   --  -  - / myTask
    -  -   --  0   --  -  - / current state
    -  -   --  0   --  -  - / move to home right

[ WARN] [1591195945.496220566]: Returning dirty link transforms
pr2_cartesian: /opt/ros/kinetic/include/moveit/robot_state/robot_state.h:1414: const Affine3d& moveit::core::RobotState::getGlobalLinkTransform(const moveit::core::LinkModel*) const: Assertion `checkLinkTransforms()' failed.
Aborted (core dumped)
`
It doesn't happen if I use joint_interpolation solver for this move. 
With cartesian solver, I also tried to simply the move (I moved the robot the goal state manually, then moved it down some centimeter, so that the cartesian move just have to move the arm up few centimeter to achieve the goal), but I got the same error. 

Here is the task : 
`void planTest(Task &t) {

	t.loadRobotModel();

	auto cartesian = std::make_shared<solvers::CartesianPath>();
	//cartesian->setProperty("jump_threshold", 0.0);

	auto joint_interpolation = std::make_shared<solvers::JointInterpolationPlanner>();

	Stage* current_state = nullptr;
	auto initial = std::make_unique<stages::CurrentState>("current state");
	current_state = initial.get();
	t.add(std::move(initial));

	{
		auto stage = std::make_unique<stages::MoveTo>("move to home right", cartesian);
		stage->setProperty("group", "right_arm");
		stage->setGoal("RIGHT_ARM_INITIAL_POSE");
		t.add(std::move(stage));
}

  std::cerr << t << std::endl;
	t.plan(10);
}`

It's just an experiment I am doing to see how the arms react to the two solvers. Is cartesianPath solver only usable with MoveRelative stage or approach translation ?

v4hn, 2020-06-03:
Looks like a plain bug. This usage should of course work. That's the whole point. :-)

Did you build moveit's master branch from source for this?

Running ROS melodic and a MoveIt `master` build I cannot reproduce your problem and it works as expected.

YannickRiou, 2020-06-04:
I am using Kinetic on a custom branch. But I think the problem might be about going back from Isometry3d to Affine3d (as Isometry3d leads to error on Kinetic distro). I am trying to see if the master branch is working for me (but I am having some error about "moc_flat_merge_proxy_model.cpp: No such file or directory").

rhaschke, 2020-06-04:
MoveIt master and MTC should also work on Kinetic. Please, try to clean your catkin build space to get rid of the moc file issue.

YannickRiou, 2020-06-08:
Even after cleaning, I didn't manage to use the master branch of MTC, I struggled with Isometry3D error (that I needed to change to Affine3d). I used the master branch of Henning Kayser and had no issue after changing Isometry3d to Affine3d. Now I can use cartesian space planner, but it can't find a solution when there is an obstacle on the way.

rhaschke, 2020-06-08:
Using the master branch with all its dependencies (see moveit.rosinstall) should work also on Kinetc - without any changes to Isometry3d. The cartesian space planner will not plan around obstacles - it's a simple Cartesian trajectory generator.

closed 2020-06-09 by YannickRiou

YannickRiou, 2020-06-16:
Even with all the dependencies, moveit in master branch and MTC in master branch, I still have the "moc error" (I have also cleaned workspace before rebuilding). Error about isometry seems to be resolved (without having to change back to Affine). Here is the full error : 
```
Starting  >>> moveit_task_constructor_visualization                                                                                            
_______________________________________________________________________________________________________________________________________________
Errors     << moveit_task_constructor_visualization:make /home/yannickriou/PR2/ws_mtc_test/logs/moveit_task_constructor_visualization/build.make.000.log
/home/yannickriou/PR2/ws_mtc_test/src/moveit_task_constructor/visualization/motion_planning_tasks/utils/flat_merge_proxy_model.cpp:604:42: fatal error: moc_flat_merge_proxy_model.cpp: No such file or directory
compilation terminated.
make[2]: *** [motion_planning_tasks/utils/CMakeFiles/motion_planning_tasks_utils.dir/flat_merge_proxy_model.cpp.o] Error 1
make[1]: *** [motion_planning_tasks/utils/CMakeFiles/motion_planning_tasks_utils.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2

```

rhaschke, 2020-06-17:
Are you running a standard Ubuntu 16.04, i.e. default cmake, Qt? 
Are you using catkin tools or catkin_make? 
Did you tried a second build attempt? The moc_* files are automatically generated by Qt's preprocessor, which should be triggered in the right order by cmake.

DangMinh24, 2020-06-21:
Have you found the solution yet? I have the same problem for ROS-`kinetic`. I followed [this guide](https://github.com/ros-planning/moveit_task_constructor/issues/86#issue-429271317) but probably the `master-branch` (tried with `melodic-branch` also) has a lot of new dependencies that makes compiling process fail.

YannickRiou, 2020-06-23:
Yes I am running Ubuntu 16.04 and I am using "catkin build". I trie several build attempt, even after running "catkin clean" and the same error happen again...

YannickRiou, 2020-06-26:
Finally found what was the problem (at least from my side !). In the master of MTC, ```set(CMAKE_INCLUDE_CURRENT_DIR ON)``` is missing  in the moveit_task_constructor/visualization folder. It allows to find the moc files generated in the build/moveit_task_constructor_visualization/motion_planning_tasks/utils folder.

rhaschke, 2020-06-26:
Hm. Weird that it works on the Travis CI server: https://travis-ci.com/github/ros-planning/moveit_task_constructor/branches

rhaschke, 2020-06-26:
If you found a solution, please file a PR fixing this issue. Thanks.