moveit/moveit_task_constructor #171: Collision objects missing in Solution.msg start_scene

cpt-yoshi, 2020-06-04:
The `moveit_task_constructor_msgs::Solution` start_scene does not include information of the collision objects.

In the pick and place demo, the information about the `table` and the `object` is missing in the `start_scene.world`:
```cpp
bool PickPlaceTask::execute() {
    ...
    moveit_task_constructor_msgs::ExecuteTaskSolutionGoal execute_goal;
    task_->solutions().front()->fillMessage(execute_goal.solution);
	
    //(gdb) print execute_goal.solution.start_scene.world
    //(gdb) print *task_->solutions().front()->start()->scene()->world_
    ...
}
```
The objects are not stored in the start_scene of the solution:
```shell
(gdb) print execute_goal.solution.start_scene.world
$9 = {collision_objects = std::vector of length 0, capacity 0, octomap = {header = {seq = 0, stamp = {<ros::TimeBase<ros::Time, ros::Duration>> = {sec = 0, nsec = 0}, <No data fields>}, frame_id = ""}, 
    origin = {position = {x = 0, y = 0, z = 0}, orientation = {x = 0, y = 0, z = 0, w = 0}}, octomap = {header = {seq = 0, stamp = {<ros::TimeBase<ros::Time, ros::Duration>> = {sec = 0, 
            nsec = 0}, <No data fields>}, frame_id = ""}, binary = 0 '\000', id = "", resolution = 0, data = std::vector of length 0, capacity 0}}}
```
The objects are stored in the planning scene:
```shell
(gdb) print *task_->solutions().front()->start()->scene()->world_
$11 = {_vptr.World = 0x7ffff4699b00 <vtable for collision_detection::World+16>, objects_ = std::map with 2 elements = {
    ["object"] = std::shared_ptr<collision_detection::World::Object> (use count 198, weak count 0) = {get() = 0x555555f93cf0}, 
    ["table"] = std::shared_ptr<collision_detection::World::Object> (use count 740, weak count 0) = {get() = 0x5555560e54d0}}, observers_ = std::vector of length 2, capacity 2 = {0x555555fe8be0, 
    0x555556039b10}}
```

I would need this information because I do some post processing on some of the sub_trajectories. For example, I would change a relative move so that it is handled by an admittance controller. This requires knowledge of the object frame (ex. a button on a device) that I want to apply a constant force on, It is a parameter needed by my controller. Note that the device will be moved in another location in the same task (Grasping pose), so I cannot know in advance the object frame.

For now, I use a relative distance wrt the end-effector to create the object frame , because there is knowledge of the robot and its attachments in the `start_scene`. A cleaner solution would be able to get the transformation directly from the `start_scene` + diff of the sub_trajectory of interest like this:

```cpp
const auto& frame = planning_scene.getFrameTransform("object");
```

cpt-yoshi, 2020-06-04:
I think that the `msg.start_scene` is never poulated.

The only places where start_scene is filled:

``` cpp
// introspection.cpp
void Introspection::fillSolution(moveit_task_constructor_msgs::Solution& msg, const SolutionBase& s) {
    s.fillMessage(msg, this);
    msg.process_id = impl->process_id_;
    msg.task_id = impl->task_->id();
    s.start()->scene()->getPlanningSceneMsg(msg.start_scene);
}

// display_solution.cpp
void DisplaySolution::fillMessage(moveit_task_constructor_msgs::Solution& msg) const {
    start_scene_->getPlanningSceneMsg(msg.start_scene);
    msg.sub_trajectory.resize(data_.size());
    auto traj_it = msg.sub_trajectory.begin();
    for (const auto& sub : data_) {
        sub.scene_->getPlanningSceneDiffMsg(traj_it->scene_diff);
	sub.trajectory_->getRobotTrajectoryMsg(traj_it->trajectory, sub.joints_);
	++traj_it;
    }
}

```
In the demo, the `Introspection::fillSolution` is never called...

rhaschke, 2020-06-12:
Thanks for tracking this down. I added the missing call in #175.

PR #175 (2020-06-18): SolutionMsg: always fill start_scene
So far, the `start_scene` field of a `SolutionMsg` was only filled by `Introspection::fillSolution()`,
but not yet by `Task::execute()`. This PR fixes #171.

I also considered filling `start_scene` in `SolutionBase::fillMessage()`. However, for sequences, this field would have been filled multiple times then, which is not an option.
closed 2020-06-18 by v4hn