moveit/moveit_task_constructor #174: Robot Support

danem, 2020-06-12:
I'm trying to get moveit_task_constructor to work with a Rethink Sawyer robot in simulation. As a quick test I modify the Cartesian demo to specify the correct joints and planning group for my robot. When running, the UI correctly updates with the specified task, but the robot doesn't move. I notice the following warning in the terminal

    ros.moveit_task_constructor_visualization: unknown task: 21234@ttt/

where `ttt` is my username.

And you can see the task planning state [here](https://i.imgur.com/yuY4NnO.png?1).

Are there any extra steps I need to take to get a robot that works well with moveit in general to work with MTC?

Thanks

iamrajee, 2020-06-12:
Your state `x+0.2` may not be accessible, you can click on the failed solution(denoted in red colour) to better understand why it's failing.
Also, you don't need 'Motion Planning` in the `Displays` tab while using MTC.

rhaschke, 2020-06-12:
@danem, do you start from a straight robot config? 
Possibly you just cannot move along x-axis by 20cm from this state?

danem, 2020-06-12:
@rhaschke That was a good suggestion, but I'm not sure if that was the problem. My robot is being simulated in gazebo, and I noticed the cartesian demo wasn't getting the actual robot state (eg the joints were all zeroed out). I added a `PlanningSceneMonitor` to the example and verified that the state I'm now getting is correct, but I'm still running into the same problem.

@iamrajee For future reference, I might be misunderstanding the UI here, but I don't see any additional information detailing the cause of failure. You mean the panel with the ∞ symbol, correct?

Thanks for the help and all the work you put into this.

rhaschke, 2020-06-12:
> I added a PlanningSceneMonitor to the example.

I'm not sure what exactly you mean by this. A PSM is not necessary. It suffices to start with the `CurrentState` stage like in the modular.cpp demo:
https://github.com/ros-planning/moveit_task_constructor/blob/f262d3058f78a451d122c9316efd032553f195d0/demo/src/modular.cpp#L101

> ... and verified that the state I'm now getting is correct, but I'm still running into the same problem.

So, the initial robot state shown when you select the failed "solution" of stage `x+0.2` is not straight anymore?

iamrajee, 2020-06-12:
> You mean the panel with the ∞ symbol, correct?

Yes.
Also, It's good practice to provide the code as a gist than as an attachment.

danem, 2020-06-12:
@rhaschke Ah ok, I didn't realize I had to click in the cost panel to get the robot to display properly. To answer your question, no the robot isn't straight and it does complete a motion, but still reports as being failed. 

[image]

rhaschke, 2020-06-12:
Does it really complete the full motion? Have a look at the failure comment (which is hidden in your cost panel).

danem, 2020-06-12:
[Here](https://streamable.com/hs689i) is a video showing the behavior more clearly.

Thanks again

rhaschke, 2020-06-12:
I still suspect that the 20cm distance isn't completely covered. It's a bug that there is no comment shown at all. But, please, try to reduce the distance, say to 10cm.

danem, 2020-06-12:
Yes, I'm finding that it is just a matter of the robot not being able to reach those positions. 

As for the comment bug, I built from source, but I might have an old commit. I'll pull the latest and let you know if that resolves my issue.

Thanks

rhaschke, 2020-06-12:
Thanks. To be clear, please provide the exact commit ID.

danem, 2020-06-12:
So with the latest commit (f262d3058f78a451d122c9316efd032553f195d0) I still don't get any comments on failure. 

[image]

This error in particular was solved again by adjusting the values to suit my robot, but I see no hint of this. This is true whether I'm using my launch file or [demo.launch](https://github.com/ros-planning/moveit_task_constructor/blob/master/demo/launch/demo.launch).

rhaschke, 2020-06-12:
I'm sorry. Indeed the failure comment was not set yet... Fixed in #176.

closed 2020-06-12 by rhaschke