moveit/moveit_task_constructor #177: [ERROR] Initialization failed: Error initializing stage: open hand: Unknown joint pose: open

Privilger, 2020-06-14:
Hi, 
I'm using this package to do pick and place for my fetch robot. 
I change the `/demo/config/panda_config.yaml` a little bit.

It seems that I need to predefine some poses and name them, like `"open"`,  right?

If so, where can I define them?

Thanks

rhaschke, 2020-06-14:
These predefined poses come from your robot's SRDF. You can edit them, for example, with the moveit setup assistant.

closed 2020-06-14 by rhaschke

Privilger, 2020-06-14:
Hi Robert,

The URDF file of the fetch robot is provided by the Fetch company. 
What should I do to add these predefined poses, can you give me some hints?

Thank you very much!

rhaschke, 2020-06-14:
I talked about SRDF not URDF. The SRDF file is part of your moveit_config package for your robot. Please have a look into the [corresponding tutorial](http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/setup_assistant/setup_assistant_tutorial.html) for details.

Privilger, 2020-06-14:
Thanks a lot!