moveit/moveit_task_constructor #180: MTC gives solution with collision

YannickRiou, 2020-06-24:
ROS distro : kinetic
Robot : PR2 
MTC version : https://github.com/henningkayser/moveit_task_constructor , master branch 

Code to reproduce : 
`#include <moveit/task_constructor/task.h>

#include <moveit/task_constructor/stages/current_state.h>
#include <moveit/task_constructor/stages/generate_grasp_pose.h>
#include <moveit/task_constructor/stages/generate_place_pose.h>
#include <moveit/task_constructor/stages/simple_grasp.h>
#include <moveit/task_constructor/stages/pick.h>
#include <moveit/task_constructor/stages/connect.h>
#include <moveit/task_constructor/solvers/pipeline_planner.h>
#include <moveit/task_constructor/solvers/cartesian_path.h>
#include <moveit/task_constructor/stages/move_to.h>
#include <moveit/task_constructor/stages/move_relative.h>
#include <moveit/task_constructor/stages/modify_planning_scene.h>
#include <moveit/task_constructor/stages/compute_ik.h>

#include <moveit_visual_tools/moveit_visual_tools.h>

#include <moveit_task_constructor_msgs/ExecuteTaskSolutionAction.h>

#include <actionlib/client/simple_action_client.h>
#include <actionlib/server/simple_action_server.h>

#include <ros/ros.h>
#include <moveit_msgs/CollisionObject.h>
#include <shape_msgs/SolidPrimitive.h>
#include <moveit_visual_tools/moveit_visual_tools.h>
#include <geometry_msgs/Pose.h>

#include <moveit/planning_scene_interface/planning_scene_interface.h>

using namespace moveit::task_constructor;

void spawnObject(moveit::planning_interface::PlanningSceneInterface& psi) {

	moveit_msgs::CollisionObject o;
	o.id= "object";
	o.header.frame_id= "base_footprint";
	o.primitive_poses.resize(1);
	o.primitive_poses[0].position.x = 0.6;
	o.primitive_poses[0].position.y = 0.4;
	o.primitive_poses[0].position.z = 0.75+(0.20/2)+0.01;
	o.primitive_poses[0].orientation.x =0.0;
	o.primitive_poses[0].orientation.y =0.0;
	o.primitive_poses[0].orientation.z =0.0;
	o.primitive_poses[0].orientation.w =1.0;
	o.primitives.resize(1);
	o.primitives[0].type= shape_msgs::SolidPrimitive::BOX;
	o.primitives[0].dimensions.resize(3);
	o.primitives[0].dimensions[0]= 0.05;
	o.primitives[0].dimensions[1]= 0.05;
	o.primitives[0].dimensions[2]= 0.20;
	psi.applyCollisionObject(o);

	o.id= "obstacle";
	o.header.frame_id= "base_footprint";
	o.primitive_poses.resize(1);
	o.primitive_poses[0].position.x = 0.6;
	o.primitive_poses[0].position.y = -0.1;
	o.primitive_poses[0].position.z = 0.75+(0.40/2);
	o.primitive_poses[0].orientation.x =0.0;
	o.primitive_poses[0].orientation.y =0.0;
	o.primitive_poses[0].orientation.z =0.0;
	o.primitive_poses[0].orientation.w =1.0;
	o.primitives.resize(1);
	o.primitives[0].type= shape_msgs::SolidPrimitive::BOX;
[... 237 lines cut ...]
When trying to execute the first solution (so the one with less cost), Moveit show this error message : 
`[ INFO] [1593011223.967760376]: Executing TaskSolution
[ INFO] [1593011224.057087036]: Fake execution of trajectory
[ INFO] [1593011224.058072849]: Fake execution of trajectory
[ INFO] [1593011227.259599078]: Fake execution of trajectory
[ INFO] [1593011228.260709522]: Fake execution of trajectory
[ INFO] [1593011230.363981178]: Fake execution of trajectory
[ INFO] [1593011231.465997327]: Fake execution of trajectory
[ INFO] [1593011231.569082610]: Trajectory component '10/14 - subsolution 0 of stage 0' is invalid
[ INFO] [1593011231.569485097]: Found a contact between 'obstacle' (type 'Object') and 'object' (type 'Robot attached'), which constitutes a collision. Contact information is not stored.
[ INFO] [1593011231.569602489]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1593011231.569997709]: Fake execution of trajectory
[ INFO] [1593011231.572810366]: Trajectory component '10/14 - subsolution 0 of stage 0' is invalid
[ INFO] [1593011231.573311518]: Found a contact between 'obstacle' (type 'Object') and 'object' (type 'Robot attached'), which constitutes a collision. Contact information is not stored.
[ INFO] [1593011231.573352046]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1593011231.573384469]: Stopping execution because the path to execute became invalid(probably the environment changed)
[ INFO] [1593011231.665432107]: Fake trajectory execution cancelled
[ INFO] [1593011231.665800215]: Stopped trajectory execution.
[ INFO] [1593011231.665912296]: Completed trajectory execution with status PREEMPTED ...`

As expected the best solution is the shortest one that smash into the obstacle, but moveit doesn't like it and stop the execution. 

Why MTC take this as a solution if there is a collision ? 

If this kind of issue was corrected on recent version, I am sorry and I will try get the latest version of MTC.

v4hn, 2020-06-24:
> MTC version : https://github.com/henningkayser/moveit_task_constructor , master branch 

Why do you built from henning's version instead of upstream?
Anything in there you need in master?
Please only report issues if you can reproduce them from this master branch (possibly with unrelated changes on top).

> If this kind of issue was corrected on recent version, I am sorry and I will try get the latest version of MTC.

It might be related to this MoveIt issue which we fixed only recently:
https://github.com/ros-planning/moveit/pull/2113

> Why MTC take this as a solution if there is a collision ?

Obviously it shouldn't and in my experience doesn't.

YannickRiou, 2020-06-29:
I didn't build the master branch because I had issues with "MOC" files generation (which seems to be corrected by https://github.com/ros-planning/moveit_task_constructor/issues/169#issuecomment-650063904).
Here is what I made to potentially find what is the problem : 
- First, if I take the master branch of MTC and moveit, and the moveit_pr2 repo at **kinetic-devel** branch and build it on a kinetic machine (16.04), I get solutions with collisions. 

- Second, if I take the master branch of MTC and moveit, and moveit_pr2 repo at **kinetic-devel** branch and build it on a melodic machine (18.04), I get an error : ```
moveit_pr2/pr2_moveit_plugins/pr2_arm_kinematics/src/pr2_arm_kinematics_plugin.cpp:93:138: error: invalid new-expression of abstract class type ‘pr2_arm_kinematics::PR2ArmIKSolver’
   pr2_arm_ik_solver_.reset(new pr2_arm_kinematics::PR2ArmIKSolver(robot_model, base_frame_,tip_frame_, search_discretization_,free_angle_)); ```.

- At this point, I took the master branch of MTC and moveit, and moveit_pr2 repo at **melodic-devel** branch and build it on a melodic machine (18.04), the build succeeded and if I try to plan the same example as above, it gives solutions without collisions ! 

- To be sure that this was the problem, I got back to my machine on kinetic (16.04), and built the master branch of MTC and moveit, and moveit_pr2 repo at **melodic-devel** branch. I had the same result : solutions without collisions ! 

So the problems seems to be about the PR2ArmsKinematicPlugins which have some changes from kinetic to melodic.

YannickRiou, 2020-06-29:
I didn't build the master branch because I had issues with "MOC" files generation (which seems to be corrected by https://github.com/ros-planning/moveit_task_constructor/issues/169#issuecomment-650063904). Now I made some test with the same scenario as above (pr2 pick and place) with the two setup : 
- MTC and moveit master branch, and moveit_pr2 (plugins) on melodic-devel branch on a kinetic (16.04) machine. Sometimes MTC finds solutions without collisions, and sometimes it finds solutions with collisions, it is completely random with the same scenario run multiple times. 

- MTC and moveit master branch, and moveit_pr2 (plugins) on melodic-devel branch on a melodic (18.04) machine. The result is the same as the first setup.

YannickRiou, 2020-06-29:
I didn't build the master branch because I had issues with "MOC" files generation (which seems to be corrected by #181).

> It might be related to this MoveIt issue which we fixed only recently: [ros-planning/moveit#2113](https://github.com/ros-planning/moveit/pull/2113)

Even with the master branch of MTC and moveit, it stills gives me randomly solutions with collisions.

v4hn, 2020-08-18:
> Can't it be related to this Moveit issue ros-planning/moveit#1217 as the problems is between two collision objects and not the robot itself ?

Indeed. Last time I investigated this issue though, I could not reproduce the reported problem itself but only found a number of issues with user setups. Currently I'm not convinced this issue refers to a single problem in MoveIt's sources, so I'm not sure how it relates to your issue.

Try to come up with a minimal example in a single MTC stage (feeding a fixed state and a fixed target).
If you can reproduce the same thing with a simple `MoveGroupAction` outside MTC then, it's related to MoveIt, otherwise you hit an MTC issue.

YannickRiou, 2020-08-19:
While trying to come up with a minimal example, a stepped on something. How planning is "shared" between MTC and Moveit ? 

Let's say I add collision objects like this : 
```
void spawnObject() {
	moveit::planning_interface::PlanningSceneInterface psi;
	moveit_msgs::CollisionObject o;
	o.id= "object";
	o.header.frame_id= "base_footprint";
	o.primitive_poses.resize(1);
	o.primitive_poses[0].position.x = 0.62;
	o.primitive_poses[0].position.y = 0.18;
	o.primitive_poses[0].position.z = 0.86;
	o.primitive_poses[0].orientation.x =0.0;
	o.primitive_poses[0].orientation.y =0.0;
	o.primitive_poses[0].orientation.z =0.0;
	o.primitive_poses[0].orientation.w =1.0;
	o.primitives.resize(1);
	o.primitives[0].type= shape_msgs::SolidPrimitive::BOX;
	o.primitives[0].dimensions.resize(3);
	o.primitives[0].dimensions[0]= 0.05;
	o.primitives[0].dimensions[1]= 0.05;
	o.primitives[0].dimensions[2]= 0.20;
	psi.applyCollisionObject(o);

	o.id= "obstacle";
	o.header.frame_id= "base_footprint";
	o.primitive_poses.resize(1);
	o.primitive_poses[0].position.x = 0.6;
	o.primitive_poses[0].position.y = -0.05;
	o.primitive_poses[0].position.z = 0.75+(0.40/2);
	o.primitive_poses[0].orientation.x =0.0;
	o.primitive_poses[0].orientation.y =0.0;
	o.primitive_poses[0].orientation.z =0.0;
	o.primitive_poses[0].orientation.w =1.0;
	o.primitives.resize(1);
	o.primitives[0].type= shape_msgs::SolidPrimitive::BOX;
	o.primitives[0].dimensions.resize(3);
	o.primitives[0].dimensions[0]= 0.05;
	o.primitives[0].dimensions[1]= 0.05;
	o.primitives[0].dimensions[2]= 0.40;
	psi.applyCollisionObject(o);

	o.id= "obstacle2";
	o.header.frame_id= "base_footprint";
	o.primitive_poses.resize(1);
	o.primitive_poses[0].position.x = 0.6;
	o.primitive_poses[0].position.y = -0.35;
	o.primitive_poses[0].position.z = 0.75+(0.40/2);
	o.primitive_poses[0].orientation.x =0.0;
	o.primitive_poses[0].orientation.y =0.0;
	o.primitive_poses[0].orientation.z =0.0;
	o.primitive_poses[0].orientation.w =1.0;
	o.primitives.resize(1);
	o.primitives[0].type= shape_msgs::SolidPrimitive::BOX;
	o.primitives[0].dimensions.resize(3);
	o.primitives[0].dimensions[0]= 0.05;
	o.primitives[0].dimensions[1]= 0.05;
	o.primitives[0].dimensions[2]= 0.40;
	psi.applyCollisionObject(o);

        o.id= "obstacle3";
	o.header.frame_id= "base_footprint";
	o.primitive_poses.resize(1);
	o.primitive_poses[0].position.x = 0.6;
	o.primitive_poses[0].position.y = -0.2;
	o.primitive_poses[0].position.z = 0.75+0.40;
	o.primitive_poses[0].orientation.x =0.0;
	o.primitive_poses[0].orientation.y =0.0;
	o.primitive_poses[0].orientation.z =0.0;
	o.primitive_poses[0].orientation.w =1.0;
	o.primitives.resize(1);
	o.primitives[0].type= shape_msgs::SolidPrimitive::BOX;
	o.primitives[0].dimensions.resize(3);
	o.primitives[0].dimensions[0]= 0.05;
	o.primitives[0].dimensions[1]= 0.05+0.3;
	o.primitives[0].dimensions[2]= 0.05;
	psi.applyCollisionObject(o);

	moveit_msgs::CollisionObject table;
	table.id= "table";
	table.header.frame_id= "base_footprint";
	table.primitive_poses.resize(1);
	table.primitive_poses[0].position.x = 0.95;
	table.primitive_poses[0].position.y = 0.0;
	table.primitive_poses[0].position.z = 0.75/2;
	table.primitive_poses[0].orientation.x =0.0;
	table.primitive_poses[0].orientation.y =0.0;
	table.primitive_poses[0].orientation.z =0.0;
	table.primitive_poses[0].orientation.w =1.0;
	table.primitives.resize(1);
	table.primitives[0].type= shape_msgs::SolidPrimitive::BOX;
	table.primitives[0].dimensions.resize(3);
	table.primitives[0].dimensions[0]= 0.85;
	table.primitives[0].dimensions[1]= 1.35;
	table.primitives[0].dimensions[2]= 0.75;
	psi.applyCollisionObject(table);
} 
```
On Rviz, I will see them on the motion planning panel under the scene objects tab. But if I toggle the motion planning display to only see the motion planning tasks display, the scene is only showing the robot but not the collision object. Can it be the reason why MTC doesn't take into account the collision object ?

v4hn, 2020-08-19:
Following this description, the error is clearly in your setup.

If you use `PlanningSceneInterface::applyCollisionObject` (btw there is a more efficient `Object**s**` version too, the objects will be added to the planning scene of the `move_group` node.

If you *afterwards* initialize a `moveit::task_constructor::Task` and use a `CurrentState` stage, it will fetch that planning scene from the `move_group` node again and use it in the task. If you use a `FixedState` on the other hand, this will not happen automatically.

If you can't see the objects in the solution of the start stage of your `Task`, the scene was not correctly initialized/populated.

YannickRiou, 2020-08-19:
Ok I see, I just never used FixedState so I used it badly with planning scene, now it seems clearer to me.

Instead I finally managed to construct a minimal example (I didn't find any way to make it more minimalist...) that cause MTC to find a solution but with a collision.

https://github.com/YannickRiou/moveit_collision_pb

Just clone it in a catkin_workspace and launch pr2.launch located in the pr2_test package, it should launch rviz and the node with Moveit and MTC. 

I also put on the repo the other package I used to be sure to have the same version. 

In the meantime I will try to have a second example with Moveit "alone" to see the result.

rhaschke, 2020-08-19:
Thanks for providing this example. However, it looks like the .gitmodules file is missing. It's not possible to checkout the submodules.

YannickRiou, 2020-08-20:
Oups, sorry, forgot it, now it should be ok.

rhaschke, 2020-08-20:
Sorry, this still isn't working:
```
Error initializing stage:
Left start: Unknown joint pose: start
```
A `start` pose isn't defined in the srdf.

YannickRiou, 2020-08-20:
Haven't updated the moveit_pr2 submodules that I had localy... Now it should be ok, sorry for the mistakes

rhaschke, 2020-09-08:
Duplicate of https://github.com/ros-planning/moveit/issues/2262. Closing here, because probably not MTC-related.

closed 2020-09-08 by rhaschke