moveit/moveit_task_constructor #182: Scheduler: Planning fails because of connect stage

cpt-yoshi, 2020-07-09:
When I have a task that has 2 `Connect` stages, the first one only computes 3 solutions. The second `Connect` as well as the other stages keeps searching for new solutions as time goes by. 

For my particular setup (See Pictures below), the setup fails to find a solution because the first `Connect` stage only connects to 1 out of 408 possible grasping poses using the `Alternatives` container. It turns out that the 3 poses will fail further down the task because the robot cannot move with the grasped object to the desired pose.

If the first Connect would try other grasping poses, it would eventually find some solutions. It is hard to add a cost function for the grasping poses because the environment is not always the same. The current Cost for all grasping poses is 0.

> At the moment scheduling is done locally by the containers which is not adequate but works.
> I am going to rework scheduling to a job-based system, but that takes time, requires major reworking of the APIs and agreement with my co-maintainer. ;)

Is there a reason why the first `connect` stops after 3 iterations? I know that the scheduling will be change in the future. I don't know if this is a bug or a current limitation of the scheduler.

<details><summary>Task Tree</summary>

[image]

</details>

<details><summary>First Connect end pose (Will fail for the other stages)</summary>

[image]

</details>

<details><summary>First Connect end pose (Will succeed for the other stages)</summary>

[image]

</details>

<details><summary>Second connect end pose</summary>

[image]

</details>

v4hn, 2020-07-09:
> Is there a reason why the first connect stops after 3 iterations? I know that the scheduling will be change in the future.

It should not stop. It will consider all possible combinations between start and end states.
However, the state pairs will only trigger computations [if the states are *compatible*](https://github.com/ros-planning/moveit_task_constructor/blob/master/core/src/stages/connect.cpp#L98).
Compatibility requires all collision objects to be in the exact same locations and all joints that are not explicitly configured in the connect must be in the same positions.

At the moment there is no good way to introspect these problems, which is quite frustrating.
You can [enable ROS_DEBUG message for the `Connecting` namespace](https://github.com/ros-planning/moveit_task_constructor/blob/master/core/src/stage.cpp#L677) to get some insights.
If you have good ideas for how to display issues without adding overly verbose (and thus useless) output for *each* incompatible state-pair, I'd be happy to hear them.

rhaschke, 2020-07-09:
I agree. Your problem is probably the incompatibility of the states. How do you init the stages inside the Alternatives container?
Please show, what's inside there.

cpt-yoshi, 2020-07-09:
So the `current_state_` variable is retrieved from the `CurrentState` stage. After that, all the `GenerateGraspPose` stages in the Alternatives container are feed with the same `current_state_` variable.

```c++

task_.reset();
task_.reset(new moveit::task_constructor::Task("test"));
Task& t = *task_;
t.loadRobotModel();

sampling_planner_ = std::make_shared<solvers::PipelinePlanner>();
sampling_planner_->setProperty("max_velocity_scaling_factor", .5);
sampling_planner_->setProperty("max_acceleration_scaling_factor", 0.5);
sampling_planner_->setProperty("goal_joint_tolerance", 1e-4);  // 1e-5

cartesian_planner_ = std::make_shared<solvers::CartesianPath>();
cartesian_planner_->setMaxVelocityScaling(0.5);
cartesian_planner_->setMaxAccelerationScaling(0.5);
cartesian_planner_->setStepSize(.01);
cartesian_planner_->setJumpThreshold(1.5);

# Task creation
sc = std::make_unique<SerialContainer>("serial container");
moveit::task_constructor::Stage* current_state_ = nullptr;

{  // Current State
    auto _current_state = std::make_unique<stages::CurrentState>("current state");
    current_state_ = _current_state.get();
    sc->insert(std::move(_current_state));
}
{  // Move Gripper
    auto stage = std::make_unique<stages::MoveTo>(name, sampling_planner_);
    stage->setGoal(robot_state);
    stage->setGroup(group_name);
    sc->insert(std::move(stage));
}
{  // Connect
    auto stage = std::make_unique<stages::Connect>(name, stages::Connect::GroupPlannerVector{ { group_name, sampling_planner_ } });
    stage->setTimeout(timeout_connect_);
    stage->properties().configureInitFrom(Stage::PARENT);
    sc->insert(std::move(stage));
}
{  // Alternatives + GenerateGraspPose
    auto parallel_container = std::make_unique<Alternatives>("Alternatives grasp container");

    for (int i = 0; i < size; ++i) {
        auto stage = std::make_unique<stages::GenerateGraspPose>(name);
	stage->properties().configureInitFrom(Stage::PARENT);
	stage->setEndEffector(eef_name);
	stage->setObject(target_frame_name);
	stage->setObjectPose(target_frame[i]);
	stage->setAngleDelta(target_angle_delta[i]);
	stage->setAngleAxis(target_axis[i]);
	stage->setEndEffector(eef_name);				
	stage->setPreGraspPose(pre_grasp_state[i]);
        stage->setTimeout(timeout_);
        stage->setMonitoredStage(current_state_);

	auto wrapper = std::make_unique<stages::ComputeIK>("insertion pose IK", std::move(stage));
	wrapper->setTimeout(timeout_);
	wrapper->setMaxIKSolutions(max_ik_solutions);
	wrapper->setIKFrame(tcp_frame, tcp_frame_name);
	wrapper->properties().configureInitFrom(Stage::PARENT, { "eef", "group" });
	wrapper->properties().configureInitFrom(Stage::INTERFACE, { "target_pose" });
	wrapper->setProperty("eef", eef_name);
	wrapper->setProperty("group", arm_group_name);
	parallel_container->insert(std::move(wrapper));
    }
    sc->insert(std::move(parallel_container));
}

...

```

rhaschke, 2020-07-10:
In principle, this looks good. To further debug this, do you mind providing your full code in a gist?

cpt-yoshi, 2020-07-10:
Does the `current_state_` only needs to be retrieved from the `CurrentState` and the `FixedState` stages?

> You can enable ROS_DEBUG message for the Connecting namespace to get some insights.

Is this done through rosconsole configuration? I didin't know that I could target a namespace!

> In principle, this looks good. To further debug this, do you mind providing your full code in a gist?

I will add a gist soon. 

In the mean time when I remove the second Connect and the other stages, the first Connect stage works as expected. It calculated 37 solutions. The code below shows the second `Connect` stage and another stage that requires the `current_state_`.

```c++

// Continuation of the code from my last post (above)
...

{  // Second Connect
    auto stage = std::make_unique<stages::Connect>(name, stages::Connect::GroupPlannerVector{ { group_name, sampling_planner_ } });
    stage->setTimeout(timeout_connect_);
    stage->properties().configureInitFrom(Stage::PARENT);
    sc->insert(std::move(stage));
}

{  // GenerateInsertionPose this stage is identical to the generateGraspPose except that it dosent need a hand_state
	auto stage = std::make_unique<stages::GenerateInsertPose>(name);
	stage->properties().configureInitFrom(Stage::PARENT);
	stage->setEndEffector(eef_name);
	stage->setObject(target_frame_name);
	stage->setObjectPoseTransform(target_frame);
	stage->setAngleDelta(target_angle_delta);
	stage->setAxis(target_axis);
	stage->setMonitoredStage(current_state_);
	stage->setEndEffector(eef_name);
	stage->setTimeout(timeout_);
	
	auto wrapper = std::make_unique<stages::ComputeIK>("insertion pose IK", std::move(stage));
	wrapper->setTimeout(timeout_);
	wrapper->setMaxIKSolutions(8);
	wrapper->setIKFrame(tcp_frame, tcp_frame_name);
	wrapper->properties().configureInitFrom(Stage::PARENT, { "eef", "group" });
	wrapper->properties().configureInitFrom(Stage::INTERFACE, { "target_pose" });
	wrapper->setProperty("eef", eef_name);
	wrapper->setProperty("group", arm_group_name);
	sc->insert(std::move(wrapper));
}

...

```

cpt-yoshi, 2020-07-10:
FYI When adding the logger verbose to DEBUG I get this message: 

``` terminal
[DEBUG] [1594415705.955347993]: Connect: different object pose: stylus
[DEBUG] [1594415705.955827014]: Connect: different object pose: stylus
[DEBUG] [1594415705.956281683]: Connect: different object pose: stylus
[DEBUG] [1594415705.956689054]: Connect: different object pose: stylus
[DEBUG] [1594415705.957146585]: Connect: different object pose: stylus
[DEBUG] [1594415705.957544892]: Connect: different object pose: stylus
[DEBUG] [1594415705.957938313]: Connect: different object pose: stylus
[DEBUG] [1594415705.958332815]: Connect: different object pose: stylus
[DEBUG] [1594415705.958760936]: Connect: different object pose: stylus
[DEBUG] [1594415705.959161241]: Connect: different object pose: stylus
[DEBUG] [1594415705.959532704]: Connect: different object pose: stylus
[DEBUG] [1594415705.959953292]: Connect: different object pose: stylus
[DEBUG] [1594415705.960404977]: Connect: different object pose: stylus
...
```

> However, the state pairs will only trigger computations if the states are compatible.

The `stylus` is the object that I grasped with the gripper. I will do some more testing this weekend. I don't know which states are comparing themselves, but after the first Connect (when I attach the object) the `stylus` is not at the same pose compared to its initial state.

rhaschke, 2020-07-11:
I think the culprit is that you use the `current_state_` to initialize your `GenerateInsertPose` stage:
`stage->setMonitoredStage(current_state_);`
Obviously, this second stage needs to be initialized from the result of the grasp pipeline, i.e. the `Attach object` stage.

cpt-yoshi, 2020-07-13:
> Obviously, this second stage needs to be initialized from the result of the grasp pipeline, i.e. the Attach object stage.

This is already done. Without it, the planning fails to compute. See this [gist](https://gist.github.com/cpt-yoshi/ee7dda5c719314246e7d2e23aaf45376). My apologies for not including all the stages before submitting this issue.

cpt-yoshi, 2020-07-13:
So you were right @v4hn and @rhaschke, this has nothing to do with the scheduler. The `Connect1` has added 405 possible states. The problem is that when the  `Connect2` is triggered, the pending size of the `Connect1` stage is set to 0.

The `Connect1` pending map is reset because of this [line](https://github.com/ros-planning/moveit_task_constructor/blob/master/core/src/stage.cpp#L634-L636).

``` terminal
...
[DEBUG] [1594651833.237652816]: Connect1: newState() -> pending.insert pending size = 397
[DEBUG] [1594651833.239946140]: Connect1: newState() -> pending.insert pending size = 398
[DEBUG] [1594651833.241434267]: Connect1: newState() -> pending.insert pending size = 399
[DEBUG] [1594651833.242859018]: Connect1: newState() -> pending.insert pending size = 400
[DEBUG] [1594651833.244407321]: Connect1: newState() -> pending.insert pending size = 401
[DEBUG] [1594651833.245865521]: Connect1: newState() -> pending.insert pending size = 402
[DEBUG] [1594651833.247574637]: Connect1: newState() -> pending.insert pending size = 403
[DEBUG] [1594651833.250064691]: Connect1: newState() -> pending.insert pending size = 404

[DEBUG] [1594651835.153845470]: Connect2: canCompute() -> pending size = 0

[DEBUG] [1594651835.155496210]: Connect1: canCompute() -> pending size = 405
[DEBUG] [1594651835.155529270]: Connect1: canCompute() -> pending size = 405
[DEBUG] [1594651835.155617191]: Connect1: canCompute() -> pending size = 405
[DEBUG] [1594651835.155636471]: Computing stage 'Connect1'
[DEBUG] [1594651835.155659829]: Connect1: compute() -> pending.pop() -> pending size = 404
[DEBUG] [1594654385.408716724]: Connect1: newState() -> before pending.remove_if() -> pending size = 404
[DEBUG] [1594654385.408861133]: Connect1: newState() -> after pending.remove_if() -> pending size = 0
...
```

v4hn, 2020-07-13:
The referenced code is supposed to drop scheduled interfaces if the other end of the partial trajectory turned out to be infeasible (and thus solving more of this particular partial solution does not make sense anymore).

Do you expect this to be a bug in your own code or the framework?
I'm not fully convinced the cost-update logic works as it should in all contexts yet and it has been on my task list to make this pruning optional..

cpt-yoshi, 2020-07-13:
> Do you expect this to be a bug in your own code or the framework?

I think this is a bug because when I comment the [reference code](https://github.com/ros-planning/moveit_task_constructor/blob/master/core/src/stage.cpp#L634-L636), the framework finds a planning solution! As you said, maybe this needs to be optional depending on the task. I will do more tests with other tasks.

[image]

> The referenced code is supposed to drop scheduled interfaces if the other end of the partial trajectory turned out to be infeasible (and thus solving more of this particular partial solution does not make sense anymore).

I agree, but only if the first partial solution has explored all of it's possible states. For my task, because the first pipeline (grasping the object) did not explore more states (grasp poses from Alternatives container), the end of the partial trajectory (the object that pushes a button) did not have a feasible solution. So by dropping the other grasping pose states, the planning was stuck and could not find a valid solution.

v4hn, 2020-07-14:
> I agree, but only if the first partial solution has explored all of it's possible states.

Yes, if a path can be extended in multiple ways, the minimal cost should be considered among all possible paths.
I never confirmed @rhaschke actually implemented it that way and it sounds like this is not what happens.
Patches are welcome of course. :-)

Discarding the invalid partial trajectories might be optional, but not discarding them should not solve your overall problem as the paths are still marked as failures (with infinite cost). It could facilitate debugging though because the statistics would show *all* feasible solutions, not just those that weren't discarded because of some other stage.

rhaschke, 2020-12-04:
@CaptainYoshi, sorry that this issue wasn't tackled for so long. I tried to reproduce this in a [minimal unittest](https://github.com/ros-planning/moveit_task_constructor/pull/220/commits/52a3b29b6467328e06e92aa756a4c23e781dc933#diff-f037b4a383685f7fc0953f5ab7bf53d569ebf02367fddb0230a551c99d3e4776R114-R118), but I wasn't able to do so. Could you please provide your failing example - ideally in a minimal form, to better understand when the issue actually occurs.

In any case, I also [removed the offending lines](https://github.com/ros-planning/moveit_task_constructor/pull/220/commits/0880316941665a3bfda6620d1b3a2fe4b53af617#diff-b284e09d92fea0042ecfec277496e7e6fdccc48459e7a2d008c8be34bb33ad57L696-L700).

rhaschke, 2020-12-08:
Ping @CaptainYoshi. Can you still reproduce this problem on the original code? If so, I'm really interested in the specific situation, because I wasn't able to reproduce. Right now, I'm deep into this code and it would be great to finish the rework on pruning soon.

closed 2020-12-08 by rhaschke

CaptainYoshi, 2020-12-08:
> Can you still reproduce this problem on the original code?

I will try to reproduce it his afternoon on the original code.

CaptainYoshi, 2020-12-08:
I tested against master branch but have an error when generating the grasp pose saying that there is a collsion for 2 static objects in the scene. Was there changes lately in MTC that would report those collision (It dosen't with a previous commit). Maybe padding or something else. Il try again tonight.

rhaschke, 2020-12-09:
Hi @CaptainYoshi, I'm not aware of all the changes we made in MoveIt and MTC since July. If you can reproduce your particular issue on an older version of MTC (and MoveIt?) that's just fine. Just tell me the git versions (SHA).

CaptainYoshi, 2020-12-15:
Great work! I cannot reproduce the error with MTC master branch `9f18219993fd7a4a9a9e42b7e7fe4cc6be402e17` with my original code (I could reproduce the bug with my original code and the old version of MTC).

rhaschke, 2020-12-15:
Hi @CaptainYoshi, thanks for confirming that the latest master branch resolves your issue. However, I would like to understand your original issue. So, please provide your original code as well as the MTC commit SHA to reproduce your issue. Thanks.

CaptainYoshi, 2020-12-16:
I've added a minimalist setup @rhaschke :
- https://github.com/CaptainYoshi/robot_setup
- https://github.com/CaptainYoshi/moveit_task_constructor/tree/replicate_mtc_issue_182/demo
- https://github.com/CaptainYoshi/moveit/tree/orientation_constraint

I think you could use the latest official MoveIt instead of my repo (Did not test that).

Bringup Rviz with robot
```bash
$ roslaunch bob_moveit_config demo.launch robot_name:=ur3_dual_eef
```
Run MTC demo (original minimalist code)
```bash
$ rosrun moveit_task_constructor_demo pick_place_demo

# Wait 10 seconds because it computes all upstream solutions in compute_ik
```
The MTC compute_ik stage has a tweak that will compute every upstream solutions. To disable this behavior, see these [lines](https://github.com/CaptainYoshi/moveit_task_constructor/blob/replicate_mtc_issue_182/core/src/stages/compute_ik.cpp#L449-L455).

The task planning should fail 99% of the time because the Connect 1 stage has valid solution that are not valid with the second connect. This closely ressembles [WIP: Rework Pruning #221](https://github.com/ros-planning/moveit_task_constructor/pull/221) because the Connect 1 stage stops computing for other solutions as if it was removed because in the Connect 2 stage, those solutions fails.

rhaschke, 2020-12-29:
Thanks for assembling this minimalist setup. But, I wasn't able to get it running:
- The package `ur_description` is missing.
  Using one that I had lying around, the `kinematics_file` parameter is not accepted by the `ur3_robot` macro. Removing this parameter worked.
- Using the proposed `rosrun` command, all parameters required by the `pick_place_demo` are undefined.
- Using `roslaunch moveit_task_constructor_demo pickplace.launch` instead, loading `panda_config.yaml`, fails as well.

captain-yoshi, 2020-12-29:
@rhaschke You can grab my [universal_robot](https://github.com/captain-yoshi/universal_robot) package.

> Using the proposed rosrun command, all parameters required by the pick_place_demo are undefined.

I think that I modified some things from the core pkg, so you will need to grab the whole MTC package, not just demo. (My bad I pointed you to the demo package in the previous message)
https://github.com/captain-yoshi/moveit_task_constructor/tree/replicate_mtc_issue_182

If you already did that, are you sure you are using the right branch? I have removed or commented sections that were loading parameters from the config file ... I think. I don't use the parameters in the demo.

With these changes you should be able to run it.

rhaschke, 2020-12-29:
Yes, I'm using the right branch. As you see [here](https://github.com/captain-yoshi/moveit_task_constructor/blob/replicate_mtc_issue_182/demo/src/pick_place_demo.cpp#L77-L81), all the parameters are still loaded.

captain-yoshi, 2020-12-29:
But the `createTable` is commented and never called.

captain-yoshi, 2021-01-18:
@rhaschke Just curious, have you been able to reproduce the error?

rhaschke, 2021-01-18:
I had to downgrade to Melodic to get your version running as is. Looks like, I can reproduce the issue now, but not deterministically.

PR #220 (2020-12-08): Fix SerialContainer
Fixes #218. Fixes #182.
The `SerialContainer` was accepting solution sequences comprising a failure-only `Connect` stage, because the `Connect` stage didn't correctly mark its sequentialized subsolutions as a failure (but only the overall sequence). @JafarAbdi, please validate that this solves your issue.