moveit/moveit_task_constructor #184: rviz plugin doesn't update statistics (anymore)

rhaschke, 2020-07-14:
Hi @v4hn, I noticed that the rviz plugin doesn't update the solution statistics anymore.
I'm not sure where this comes from, because it definitely worked in the past. Do you see this issue as well?

v4hn, 2020-07-15:
> Hi @v4hn, I noticed that the rviz plugin doesn't update the solution statistics anymore.
> I'm not sure where this comes from, because it definitely worked in the past. Do you see this issue as well?

This morning I observed that the statistics in the gui only update when RViz becomes focused and/or I click on an entry in the tasks_view.
It is probably related to the GUI rework, even if I did not notice anything before.
I'll investigate later today (unless you look into it before then :)).

rhaschke, 2020-07-15:
I don't think this issue is related to your GUI rework. I'm suspecting Qt itself actually. Trying with Ubuntu 20.04 now...
Don't invest time (yet) into this.

v4hn, 2020-07-15:
> I'm suspecting Qt itself actually.

I observe the behavior I described with Qt 5.14.2 .

> Don't invest time (yet) into this.

Ok.

rhaschke, 2020-07-15:
Initially, I suspected Qt not handling the `dataChanged()` signal correctly from the ROS callback handler:
https://bugreports.qt.io/browse/QTBUG-46517
However, adding the presumably missing call to `qRegisterMetaType()` didn't solve the issue. Hence, I bisected the issue and found the culprit: 75bc6566f6474ab7fcb158f6789ac6e31e71aa73 (part of #146). Looking into the code, the reason is obvious:
you added a virtual method `dataChanged()` hiding the original signal: https://github.com/ros-planning/moveit_task_constructor/commit/75bc6566f6474ab7fcb158f6789ac6e31e71aa73#diff-7eddd75c4f54f2f980287bc7e3ec5f17R192-R193
I'm sorry not having reviewed this PR in time... I'm not sure I will have time to fix this within the remaining week. Next two week I will be absent due to holidays.

PR #186 (2020-07-16): Fixup: forward dataChanged to QTreeView
... to trigger visualization updates.

That was a rather obvious mistake on my side, sorry :(

Fixes #184
closed 2020-07-16 by rhaschke