moveit/moveit_task_constructor #185: how to actuate gazebo robot

fmros, 2020-07-15:
Hy there,

thanks for your great work!

I am trying to get a gazebo-simulated robot to move with the moveit_task_contructor.
Working:
moveit-motion planning: robot is moving in gazebo
cartesian demo example has been adapted to new robot, the task planner plugin has been loaded in rviz and when I start
to plan a task as in the cartesian.cpp the task is loaded into the plugin and a visualization of the robot is moving. But(!) not the "real" robot in gazebo. What do I have to do to move the robot? I tried to include in the cartesian.cpp task.execute() but the programm did not return from that call without giving a notice whats happening.

Thank you very much in advance!

rhaschke, 2020-07-15:
MTC uses the same approach to execute motion plans as MoveIt itself. Hence, if you can move your robot using standard MoveIt, it should work in MTC as well. Maybe you are missing the [`ExecuteTaskSolutionCapability`](https://github.com/ros-planning/moveit_task_constructor/blob/master/capabilities/capabilities_plugin_description.xml) in your robot's moveit_config?

fmros, 2020-07-15:
I added this to my moveit launch file.
```
 <!-- Load  ExecuteTaskSolutionCapability so we can execute found solutions in simulation -->
 <param name="move_group/capabilities" value="move_group/ExecuteTaskSolutionCapability" />
```
So you thik that following main() shoud work?
```
int main( int argc, char **argv )
{
    ros::init( argc, argv, "mtc_tutorial" );
    // run an asynchronous spinner to communicate with the move_group node and rviz
    ros::AsyncSpinner spinner( 1 );
    spinner.start();

    auto task = createTask();
    try {
        if( task.plan() ) {
            task.introspection().publishSolution( *task.solutions().front() );
        }
    } catch( const InitStageException &ex ) {
        std::cerr << "planning failed with exception" << std::endl << ex << task;
    }

    ROS_INFO( "Execute task" );
    task.execute( *task.solutions().front() );
    ROS_INFO( "Execution complete" );

    ros::waitForShutdown();  // keep alive for interactive inspection in rviz
    return 0;
}
```

rhaschke, 2020-07-15:
Do you see the ExecuteTaskSolutionCapability listed by the move_group node? Do you receive log output from the move_group node regarding task execution?

fmros, 2020-07-16:
I see that 
[image] 
is not connected to move_group (Running the demo, it is should connected).

No log output from move_group for task execution. At this point, without running any task, am I missing some configuration in the launch file? 
my launch file:

```
<?xml version="1.0"?>
<launch>
    <include file="$(find XXX_moveit)/launch/planning_context.launch">
    <arg name="load_robot_description" value="false"/>
    </include>

    <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher">
    <param name="/use_gui" value="false"/>
    <rosparam param="/source_list">[/controllers/joint_states]</rosparam>
    </node>

    <!-- Load  ExecuteTaskSolutionCapability so we can execute found solutions in simulation -->
    <param name="move_group/capabilities" value="move_group/ExecuteTaskSolutionCapability" />

    <include file="$(find XXX_moveit)/launch/move_group.launch">
    <arg name="publish_monitored_planning_scene" value="true"/>
    <arg name="allow_trajectory_execution" value="true"/>
    </include>

    <!-- Run Rviz and load the default config to see the state of the move_group node -->
    <arg name="use_rviz" default="false" />
    <include file="$(find XXX_moveit)/launch/moveit_rviz.launch" if="$(arg use_rviz)">
    <arg name="rviz_config" value="$(find XXX_moveit)/launch/moveit_task_constructor.rviz"/>
    </include>
    <include file="$(find XXX_common_meta)/visualization/rviz_task_constructor.launch">
    </include>

    <!--
    Convert joint states to TF transforms for rviz, etc
    -->
    <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" respawn="true" output="screen">
    </node>
</launch>
```

fmros, 2020-07-16:
I see the 
[image]
which is not connected to move_group. As seen when running the demo it should be connected.

rhaschke, 2020-07-16:
You need to have this line in your launch file (but you already have):
```xml
<!-- Load  ExecuteTaskSolutionCapability so we can execute found solutions in simulation -->
<param name="move_group/capabilities" value="move_group/ExecuteTaskSolutionCapability" />
```

fmros, 2020-07-16:
May I need to have the same package structure as in this repository? I guess having this repo in the workspace is enough that the plugin ExecuteTaskSolutionCapability gets detected?
Strange thing is that I actually have the MTC GUI in rviz and loaded correctly. The visualization of the tasks works also correctly.

rhaschke, 2020-07-16:
You should see the ExecuteTaskSolutionCapability listed by move_group node on startup.
You don't need the same package structure. A normal moveit_config package is fine.

fmros, 2020-07-17:
Is there an example of a working setup and connection of the action server arround that somebody knows?

v4hn, 2020-07-17:
If you have a rather recent moveit configuration, it will also set the capabilities parameter in `move_group.launch`.
You need to either switch the order of the `<param>` and the `<include move_group>` (the last occurance of a param takes precedence), or change the `<param>` to an arg:

```
    <include file="$(find XXX_moveit)/launch/move_group.launch">
    <arg name="publish_monitored_planning_scene" value="true"/>
    <arg name="allow_trajectory_execution" value="true"/>
    <arg name="capabilities" value="move_group/ExecuteTaskSolutionCapability" />
    </include>
```

fmros, 2020-07-17:
thank you, `<arg name="capabilities" value="move_group/ExecuteTaskSolutionCapability" />` changed the story, the action server is loading now. Now i get a Segmentation Fault:

```
[1594985952.748140201, 260.280000000]: Planning successful, execute task...
Segmentation fault (core dumped)
```
----
Ok if we continue here or shall i open a new issue?

rhaschke, 2020-07-17:
I guess the shown output comes from the MTC planning node? Could you, please, also show the output of the move_group node running the ExecuteTaskSolutionCapability?
Which versions of MoveIt and MTC are you using?
How do you trigger task execution? From the rviz plugin or from code?
To reproduce your issue, it would be helpful if you provide a minimal code example.

fmros, 2020-07-17:
There is no output coming from move_group node. 
ROS Melodic & Ubuntu 18.04
```
$ moveit_version 
1.0.4-Alpha
```
MTC: Current master branch
Task execution is triggered by a small adapted cpp node. main function is as [above](https://github.com/ros-planning/moveit_task_constructor/issues/185#issuecomment-658709613).

Here minimal code example:
```

using namespace moveit::task_constructor;

Task createTask()
{
    Task t;
    t.stages()->setName( "Palletizing Action" );

    const std::string group = "robot";

    // create Cartesian interpolation "planner" to be used in various stages
    auto cartesian_interpolation = std::make_shared<solvers::CartesianPath>();
    // create a joint-space interpolation "planner" to be used in various stages
    auto joint_interpolation = std::make_shared<solvers::JointInterpolationPlanner>();

    // start from a fixed robot state
    t.loadRobotModel();
    auto scene = std::make_shared<planning_scene::PlanningScene>( t.getRobotModel() );
    {
        auto &state = scene->getCurrentStateNonConst();
        state.setToDefaultValues( state.getJointModelGroup( group ), "pallet" );

        auto fixed = std::make_unique<stages::FixedState>( "initial state" );
        fixed->setState( scene );
        t.add( std::move( fixed ) );
    }

    {
        auto stage = std::make_unique<stages::MoveRelative>( "y -0.8", cartesian_interpolation );
        stage->setGroup( group );
        geometry_msgs::Vector3Stamped direction;
        direction.header.frame_id = "world";
        direction.vector.y = -0.8;
        stage->setDirection( direction );
        t.add( std::move( stage ) );
    }

    {
        auto stage = std::make_unique<stages::MoveRelative>( "x -0.55", cartesian_interpolation );
        stage->setGroup( group );
        geometry_msgs::Vector3Stamped direction;
        direction.header.frame_id = "world";
        direction.vector.x = -0.55;
        stage->setDirection( direction );
        t.add( std::move( stage ) );
    }

    {
        auto stage = std::make_unique<stages::MoveRelative>( "y +0.8", cartesian_interpolation );
        stage->setGroup( group );
        geometry_msgs::Vector3Stamped direction;
        direction.header.frame_id = "world";
        direction.vector.y = +0.8;
        stage->setDirection( direction );
        t.add( std::move( stage ) );
    }
    return t;
}

int main( int argc, char **argv )
{
    ros::init( argc, argv, "palletizer" );
    // run an asynchronous spinner to communicate with the move_group node and rviz
    ros::AsyncSpinner spinner( 1 );
    spinner.start();

    auto task = createTask();

    try {
        if( task.plan() ) {

            task.introspection().publishSolution( *task.solutions().front() );
        }
        ROS_INFO( "Planning successful, execute task..." );

        moveit_msgs::MoveItErrorCodes execute_result = task.execute( *task.solutions().front() );
        std::cout << execute_result << std::endl;
        if( execute_result.val != moveit_msgs::MoveItErrorCodes::SUCCESS ) {
            ROS_INFO( "Task execution Failed!" );
        } else {
            ROS_INFO( "Task execution completed!" );
        }
    } catch( const InitStageException &ex ) {
        std::cerr << "planning failed with exception" << std::endl << ex << task;
    }

    ros::waitForShutdown();  // keep alive for interactive inspection in rviz
    return 0;
}

```

fmros, 2020-07-17:
I detected that this has to do with a maybe? a not reachable position of the robot. I added another stage for z translation and saw that for certain values it crashes. But no output of move_group node about why. The task.execute() command just runs into the Segmentatino Fault. Have no idea how to debug...

v4hn, 2020-07-17:
> The task.execute() command just runs into the Segmentatino Fault. Have no idea how to debug...

The usual: Compile with `-g -Og` and look at the trace in gdb.

rhaschke, 2020-07-17:
Providing the backtrace of the segfault would be helpful indeed.

v4hn, 2020-08-04:
The last reported issue has nothing to do with the original title of the thread. I'll close this thread.

If you can provide more information (and possibly a minimal example) feel free to create a new thread.

closed 2020-08-04 by v4hn