moveit/moveit_task_constructor #187: ModifyPlanningScene stage does not contain attachedCollisionObjects if used with "AllowCollision"

karolyartur, 2020-07-17:
In the `pick_place_task.cpp` example in `moveit_task_constructor_demo`, there are two `ModifyPlanningScene` stage instances that behave in a confusing way. [Here](https://github.com/ros-planning/moveit_task_constructor/blob/master/demo/src/pick_place_task.cpp#L237), when `ModifyPlanningScene` is used with the `allowCollision` command, the resulting sub_trajectory does not seem to contain the `attachedCollisionObject` that it is setting the collisions for. 

You can test this locally by adding these lines after [line 440](https://github.com/ros-planning/moveit_task_constructor/blob/master/demo/src/pick_place_task.cpp#L440):

```cpp
        for (auto subtraj : execute_goal.solution.sub_trajectory)
	{
		std::cout << "Printing new stage, attached coll. objects: ";
		for (auto aco : subtraj.scene_diff.robot_state.attached_collision_objects)
			std::cout << aco.object.id;
		
		std::cout << std::endl;
	}
```

The output that I am getting is this:

```
Printing new stage, attached coll. objects: 
Printing new stage, attached coll. objects: 
Printing new stage, attached coll. objects: 
Printing new stage, attached coll. objects: 
Printing new stage, attached coll. objects: 
Printing new stage, attached coll. objects: 
Printing new stage, attached coll. objects: 
Printing new stage, attached coll. objects: object
Printing new stage, attached coll. objects: 
Printing new stage, attached coll. objects: object
Printing new stage, attached coll. objects: 
Printing new stage, attached coll. objects: object
Printing new stage, attached coll. objects: object
Printing new stage, attached coll. objects: object
Printing new stage, attached coll. objects: object
Printing new stage, attached coll. objects: 
Printing new stage, attached coll. objects: 
Printing new stage, attached coll. objects: 
Printing new stage, attached coll. objects: 
```

I would expect that the object remains an `attachedCollisionObject` in each stage between the `attach` and `detach` stages. Is this behavior intentional?

Thanks in advance.

PS: Why are the `allowCollision` stage and the `attachObject` stage in the `pick_place_task` separate instances in the first place? I am not sure why they are not combined. Is there a reason that I am missing?

rhaschke, 2020-07-17:
Note that you print PlanningScene **diffs**: Only changes to the scene are reported!
You are right, both stages that modify the planning scene in sequence could be combined.

closed 2020-07-17 by rhaschke

karolyartur, 2020-07-27:
Thank you!
What confused me was that the solutions of the stages for robot motion also contained the attached collision object, because the robot state was updated.