moveit/moveit_task_constructor #188: New Grasp Generator Stage

bostoncleek, 2020-07-17:
This is an internship project proposed by @mamoll and @JStech. I would appreciate feedback or guidance.

**Is your feature request related to a problem? Please describe.**
I want to add more robust grasping support using [GPD](https://github.com/atenpas/gpd) and eventually [Dex-Net](https://github.com/BerkeleyAutomation/dex-net) as well. I would also like to add support that allows for sensor input into stages. 

**Describe the solution you'd like**
An alternative grasp generator stage using GPD. The GPD lib outputs many candidates that can be evaluated based on cost and feasibility using the ComputeIK stage to select the best grasp.

v4hn, 2020-07-17:
> This is an internship project proposed by @mamoll and @JStech.

Go ahead. :-)

> I would also like to add support that allows for sensor input into stages.

At the moment there would be no difference between generating the grasps outside of MTC and forwarding them to a generic grasp stage and forwarding the point clouds/images to the stage directly. Both approaches are clearly possible. Do you see some merit in forwarding the sensor data directly?

bostoncleek, 2020-07-17:
Writing a generic generator that can also take sensor input would remove cumbersome dependencies and make the MTC frame work more flexible and easy to use.  I can start here.   

The approach and lift directions depend on the grasp candidate which is related to IK feasibility. Is there a way to propagate the pre-grasp and post-grasp poses to other stages after the IK stage?

rhaschke, 2020-07-17:
In principle, you can assign arbitrary properties, e.g. pre- and post-grasp hand postures, to any solution. And you can ask MTC to forward these properties in adjacent stages (`setForwardedProperties()`). I never finished to write an example pick and place task relying on these concepts, instead of hard-coding them into the task program.
Maybe, it is a good starting point to rewrite the existing [Pick](https://github.com/ros-planning/moveit_task_constructor/blob/4bee03476855dcbddf468173f0c02ee3487b6d98/core/include/moveit/task_constructor/stages/pick.h#L92-L106) and [Place](https://github.com/ros-planning/moveit_task_constructor/blob/4bee03476855dcbddf468173f0c02ee3487b6d98/core/include/moveit/task_constructor/stages/pick.h#L109-L123) containers to use this more generic scheme. This will be needed anyway if you want to pass such properties from the grasp generator stage.

bostoncleek, 2020-07-18:
I started looking into the Pick and Place containers. It also looks like it would be useful for these containers to be able to modify the planning scene to control whether collisions are allowed when grasping, lifting, or placing objects.

bostoncleek, 2020-07-18:
I started looking into these containers. It looks like in order for them to properly pick/place both containers would need  to modify the planning scene to control which collisions are allowed during grasping, lifting, and placing.