moveit/moveit_task_constructor #191: Set constraint link name wrt an attached object

cpt-yoshi, 2020-07-26:
I want to add a `constraint` in one of my `Connect` stage. The constraint fails when the link_name is an attached object attached via the `ModifyPlanningScene` stage:
``` c++
...
moveit_msgs::Constraints constraint;
constraint.name = "test";

Eigen::Quaterniond quat = Eigen::AngleAxisd(0, Eigen::Vector3d::UnitZ()) *
                          Eigen::AngleAxisd(M_PI_2, Eigen::Vector3d::UnitY()) *
                          Eigen::AngleAxisd(0, Eigen::Vector3d::UnitX());

moveit_msgs::OrientationConstraint oc;
oc.header.frame_id = "optical_table";
oc.link_name = "object_name";  // Object that is attached to the eef
oc.orientation.w = quat.w();
oc.orientation.x = quat.x();
oc.orientation.y = quat.y();
oc.orientation.z = quat.z();

oc.absolute_x_axis_tolerance = 2 * M_PI;
oc.absolute_y_axis_tolerance = 0.4;
oc.absolute_z_axis_tolerance = 0.4;
oc.weight = 100;
constraint.orientation_constraints.push_back(oc);

stage->setPathConstraints(std::move(constraint));
...
```

From [moveit tutorials](https://ros-planning.github.io/moveit_tutorials/doc/subframes/subframes_tutorial.html) it is possible to set the end-effector link to be an object an even a subframe.

``` c++
group.clearPoseTargets();
group.setEndEffectorLink(end_effector_link);
/*
group.setEndEffectorLink("cylinder/tip");    // Example 1
group.setEndEffectorLink("panda_hand");      // Example 2
*/
group.setStartStateToCurrentState();
group.setPoseTarget(pose);
```

When parsing the MTC library, I dont see the method `setEndEffectorLink`. Is this feature missing from MTC?

v4hn, 2020-08-04:
Sorry for the late response.

Your code should "just work" but apparently it doesn't. This has nothing to do with any notion of "EndEffector" in the MoveGroupInterface code you quote.
It might actually be a bug in your planner configuration. Do you have have [the `ResolveConstraintFrames` adapter loaded](https://github.com/ros-planning/moveit/blob/master/moveit_setup_assistant/templates/moveit_config_pkg_template/launch/ompl_planning_pipeline.launch.xml#L10) in the planning pipeline?

cpt-yoshi, 2020-08-04:
My apologies. I made a mistake, the orientation constraint was not wrt the mesh. After the fix the planning solves correctly. But the console reports this errors on a successful solution for the connect stage:

``` terminal
[DEBUG] [1596556607.717987725]: Computing stage 'filtered'
[ INFO] [1596556607.719912078]: Planner configuration 'left_eef' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1596556607.720311533]: left_eef: Allocating specialized state sampler for state space
[ INFO] [1596556607.720422833]: left_eef/left_eef: Starting planning with 1 states already in datastructure
[ INFO] [1596556608.246790618]: left_eef/left_eef: Created 4 states (2 start + 2 goal)

[ INFO] [1596556608.246864900]: Solution found in 0.526790 seconds
[ INFO] [1596556608.450591070]: SimpleSetup: Path simplification took 0.203637 seconds and changed from 3 to 2 states
[ERROR] [1596556608.464468861]: Link 'cuvette' not found in model 'ur3_dual_eef'
[ WARN] [1596556608.464553573]: Could not find link model for link name cuvette
```
I guess that this is not really an error because a solution was found. But there are some solutions that does not respect the constraints. This happens when there appears to have multiple segments in one solution (See gif). I can open up another issue if you prefer.

The object `cuvette` mesh has the same orientation frame as the frame id `optical_table` (See pictures). Fixed code:
``` c++
moveit_msgs::Constraints constraint;
constraint.name = "test";

Eigen::Quaterniond quat = Eigen::AngleAxisd(0, Eigen::Vector3d::UnitZ()) *
                          Eigen::AngleAxisd(0, Eigen::Vector3d::UnitY()) *
                          Eigen::AngleAxisd(0, Eigen::Vector3d::UnitX());

moveit_msgs::OrientationConstraint oc;
oc.header.frame_id = "optical_table";
oc.link_name = "cuvette";  // Object that is attached to the eef
oc.orientation.w = quat.w();
oc.orientation.x = quat.x();
oc.orientation.y = quat.y();
oc.orientation.z = quat.z();

oc.absolute_x_axis_tolerance = 0.4; // ~23 degrees
oc.absolute_y_axis_tolerance = 0.4; // ~23 degrees
oc.absolute_z_axis_tolerance = 2 * M_PI; // Represents the gravity axis (Pointing UP)
oc.weight = 100;
constraint.orientation_constraints.push_back(oc);

```

[image]
[image]
[image]

v4hn, 2020-08-04:
> [ERROR] [1596556608.464468861]: Link 'cuvette' not found in model 'ur3_dual_eef'

If this actually talks about `oc.link_name` you are probably still missing the `ResolveConstraintFrames` adapter.
To verify whether this is an MTC issue, try to reproduce this planning request in MoveIt (using the MoveGroupInterface).
I would expect it will fail as well (which would mean this is not an issue with MTC, but with your setup or MoveIt general).

cpt-yoshi, 2020-08-04:
> Do you have have the ResolveConstraintFrames adapter loaded in the planning pipeline?

Yes I do!

> If this actually talks about oc.link_name 

The only place I see `not found in model ` when grepping the moveit repo is in the `robot_model.cpp` file. There is a good chance that this is related to the` oc.link_name`.

> I would expect it will fail as well (which would mean this is not an issue with MTC, but with your setup or MoveIt general).

It does not fail with MTC. It gives good solutions. I will investigate this further and report back.

v4hn, 2020-08-04:
> There is a good chance that this is related to the` oc.link_name`.

You can easily check by changing the string and observing the warning. :)

> It does not fail with MTC. It gives good solutions.
> I will investigate this further and report back.

If it *also* generates bad solutions it rather fails in my opinion.
Of course it could also be some independent warning and the constraint planning works as it should...

cpt-yoshi, 2020-08-05:
> If it *also* generates bad solutions it rather fails in my opinion.

I agree!

> I would expect it will fail as well (which would mean this is not an issue with MTC, but with your setup or MoveIt general).

You are right! This has nothing to do with MTC. When adding path constraints in the `Connect` stage with a link_name from an attached object, MoveIt silence the [ConstraintEvaluationResult](https://github.com/ros-planning/moveit/blob/master/moveit_core/kinematic_constraints/src/kinematic_constraint.cpp#L597-L598) to success. I will add a pull request by the end of the week. My initial fix fixes this issue, but I will do more testing.

On the other hand the `ComputeIK` stage can retrieve the attached object because it uses the [getWorld](https://github.com/ros-planning/moveit/blob/master/moveit_core/planning_scene/src/planning_scene.cpp#L1874) function to calculate the transformation.

<details><summary>Backtrace for Connect stage</summary>
<p>

``` terminal
(gdb) backtrace
#0  moveit::core::RobotModel::getLinkModel (this=0x555556f0b730, name="cuvette", has_link=0x0) at /home/bob/ws/ros/catkin_ws/src/moveit/moveit_core/robot_model/src/robot_model.cpp:1162
#1  0x00007ffff3891c41 in moveit::core::RobotModel::getLinkModel (this=0x555556f0b730, name="cuvette", has_link=0x0)
    at /home/bob/ws/ros/catkin_ws/src/moveit/moveit_core/robot_model/src/robot_model.cpp:1137
#2  0x00007ffff2b3e686 in kinematic_constraints::OrientationConstraint::configure (this=0x55555d4f56d0, oc=..., tf=...)
    at /home/bob/ws/ros/catkin_ws/src/moveit/moveit_core/kinematic_constraints/src/kinematic_constraint.cpp:501
#3  0x00007ffff2b44f52 in kinematic_constraints::KinematicConstraintSet::add (this=0x7fffffff8880, oc=std::vector of length 1, capacity 1 = {...}, tf=...)
    at /home/bob/ws/ros/catkin_ws/src/moveit/moveit_core/kinematic_constraints/src/kinematic_constraint.cpp:1111
#4  0x00007ffff2b45241 in kinematic_constraints::KinematicConstraintSet::add (this=0x7fffffff8880, c=..., tf=...)
    at /home/bob/ws/ros/catkin_ws/src/moveit/moveit_core/kinematic_constraints/src/kinematic_constraint.cpp:1140
#5  0x00007ffff72d8da1 in planning_scene::PlanningScene::isPathValid (this=0x55555d005d30, trajectory=..., path_constraints=..., goal_constraints=std::vector of length 0, capacity 0, group="left_eef", 
    verbose=false, invalid_index=0x7fffffff8a50) at /home/bob/ws/ros/catkin_ws/src/moveit/moveit_core/planning_scene/src/planning_scene.cpp:2170
#6  0x00007ffff72d941a in planning_scene::PlanningScene::isPathValid (this=0x55555d005d30, trajectory=..., path_constraints=..., group="left_eef", verbose=false, invalid_index=0x7fffffff8a50)
    at /home/bob/ws/ros/catkin_ws/src/moveit/moveit_core/planning_scene/src/planning_scene.cpp:2232
#7  0x00007fffe9e0465a in planning_pipeline::PlanningPipeline::generatePlan(std::shared_ptr<planning_scene::PlanningScene const> const&, moveit_msgs::MotionPlanRequest_<std::allocator<void> > const&, planning_interface::MotionPlanResponse&, std::vector<unsigned long, std::allocator<unsigned long> >&) const () from /home/bob/ws/ros/catkin_ws/devel/lib/libmoveit_planning_pipeline.so.1.1.0
#8  0x00007fffe9e057a1 in planning_pipeline::PlanningPipeline::generatePlan(std::shared_ptr<planning_scene::PlanningScene const> const&, moveit_msgs::MotionPlanRequest_<std::allocator<void> > const&, planning_interface::MotionPlanResponse&) const () from /home/bob/ws/ros/catkin_ws/devel/lib/libmoveit_planning_pipeline.so.1.1.0
#9  0x00007fffedea4dba in moveit::task_constructor::solvers::PipelinePlanner::plan (this=0x5555570fa6e0, from=std::shared_ptr<const planning_scene::PlanningScene> (use count 8, weak count 1) = {...}, 
    to=std::shared_ptr<const planning_scene::PlanningScene> (use count 3, weak count 1) = {...}, jmg=0x555556f0d310, timeout=5, result=std::shared_ptr<robot_trajectory::RobotTrajectory> (empty) = {...}, 
    path_constraints=...) at /home/bob/ws/ros/catkin_ws/src/moveit_task_constructor/core/src/solvers/pipeline_planner.cpp:113
#10 0x00007fffed85e550 in moveit::task_constructor::stages::Connect::compute (this=0x55555718b0b0, from=..., to=...)
    at /home/bob/ws/ros/catkin_ws/src/moveit_task_constructor/core/src/stages/connect.cpp:157
#11 0x00007fffede31a8a in moveit::task_constructor::ConnectingPrivate::compute (this=0x5555570ee510) at /home/bob/ws/ros/catkin_ws/src/moveit_task_constructor/core/src/stage.cpp:672
#12 0x00007fffeddc4215 in moveit::task_constructor::StagePrivate::runCompute (this=0x5555570ee510)
    at /home/bob/ws/ros/catkin_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/stage_p.h:152
#13 0x00007fffeddbe529 in moveit::task_constructor::WrapperBase::compute (this=0x555557100b30) at /home/bob/ws/ros/catkin_ws/src/moveit_task_constructor/core/src/container.cpp:696
#14 0x00007fffeddb9ac1 in moveit::task_constructor::ContainerBasePrivate::compute (this=0x5555570da480) at /home/bob/ws/ros/catkin_ws/src/moveit_task_constructor/core/src/container.cpp:106
#15 0x00007fffeddc4215 in moveit::task_constructor::StagePrivate::runCompute (this=0x5555570da480)
    at /home/bob/ws/ros/catkin_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/stage_p.h:152
#16 0x00007fffeddbd113 in moveit::task_constructor::SerialContainer::compute (this=0x55555717f830) at /home/bob/ws/ros/catkin_ws/src/moveit_task_constructor/core/src/container.cpp:531
[... cut ...]
#22 0x00007ffff5a92f49 in ProtocolPlanner::plan (this=0x7fffffffb640, action_block=std::vector of length 5, capacity 5 = {...}, max_solutions=30)
    at /home/bob/ws/ros/catkin_ws/src/bob/bob_planner/src/protocol_planner/protocol_planner.cpp:280
#23 0x00005555556657be in test_dual_eef_protocol2 () at /home/bob/ws/ros/catkin_ws/src/bob/bob_planner/src/pp.cpp:560
#24 0x000055555567fb82 in main (argc=1, argv=0x7fffffffc168) at /home/bob/ws/ros/catkin_ws/src/bob/bob_planner/src/pp.cpp:1970

```

</p>
</details>

v4hn, 2020-08-05:
> I will add a pull request by the end of the week.

Thanks! Looking forward to review it.

> On the other hand the ComputeIK stage can retrieve the attached object

yes, it's a very different code path.

cpt-yoshi, 2020-08-08:
> I will add a pull request by the end of the week.

Basically what I wrote was something that resembles the `ResolveConstraintFrames` function before I fond out that the ResolveConstraintFrames is not always triggered! So for the **same** Connect stage the planning has completly different backtraces

<details><summary>Here is a backtrace that does not triggers the ResolveConstraintFrames</summary>
<p>

``` terminal
(gdb) backtrace
#0  moveit::core::RobotModel::getLinkModel (this=0x555556f0b730, name="cuvette", has_link=0x0) at /home/bob/ws/ros/catkin_ws/src/moveit/moveit_core/robot_model/src/robot_model.cpp:1162
#1  0x00007ffff3891c41 in moveit::core::RobotModel::getLinkModel (this=0x555556f0b730, name="cuvette", has_link=0x0)
    at /home/bob/ws/ros/catkin_ws/src/moveit/moveit_core/robot_model/src/robot_model.cpp:1137
#2  0x00007ffff2b3e686 in kinematic_constraints::OrientationConstraint::configure (this=0x55555d4f56d0, oc=..., tf=...)
    at /home/bob/ws/ros/catkin_ws/src/moveit/moveit_core/kinematic_constraints/src/kinematic_constraint.cpp:501
#3  0x00007ffff2b44f52 in kinematic_constraints::KinematicConstraintSet::add (this=0x7fffffff8880, oc=std::vector of length 1, capacity 1 = {...}, tf=...)
    at /home/bob/ws/ros/catkin_ws/src/moveit/moveit_core/kinematic_constraints/src/kinematic_constraint.cpp:1111
#4  0x00007ffff2b45241 in kinematic_constraints::KinematicConstraintSet::add (this=0x7fffffff8880, c=..., tf=...)
    at /home/bob/ws/ros/catkin_ws/src/moveit/moveit_core/kinematic_constraints/src/kinematic_constraint.cpp:1140
#5  0x00007ffff72d8da1 in planning_scene::PlanningScene::isPathValid (this=0x55555d005d30, trajectory=..., path_constraints=..., goal_constraints=std::vector of length 0, capacity 0, group="left_eef", 
    verbose=false, invalid_index=0x7fffffff8a50) at /home/bob/ws/ros/catkin_ws/src/moveit/moveit_core/planning_scene/src/planning_scene.cpp:2170
#6  0x00007ffff72d941a in planning_scene::PlanningScene::isPathValid (this=0x55555d005d30, trajectory=..., path_constraints=..., group="left_eef", verbose=false, invalid_index=0x7fffffff8a50)
    at /home/bob/ws/ros/catkin_ws/src/moveit/moveit_core/planning_scene/src/planning_scene.cpp:2232
#7  0x00007fffe9e0465a in planning_pipeline::PlanningPipeline::generatePlan(std::shared_ptr<planning_scene::PlanningScene const> const&, moveit_msgs::MotionPlanRequest_<std::allocator<void> > const&, planning_interface::MotionPlanResponse&, std::vector<unsigned long, std::allocator<unsigned long> >&) const () from /home/bob/ws/ros/catkin_ws/devel/lib/libmoveit_planning_pipeline.so.1.1.0
#8  0x00007fffe9e057a1 in planning_pipeline::PlanningPipeline::generatePlan(std::shared_ptr<planning_scene::PlanningScene const> const&, moveit_msgs::MotionPlanRequest_<std::allocator<void> > const&, planning_interface::MotionPlanResponse&) const () from /home/bob/ws/ros/catkin_ws/devel/lib/libmoveit_planning_pipeline.so.1.1.0
#9  0x00007fffedea4dba in moveit::task_constructor::solvers::PipelinePlanner::plan (this=0x5555570fa6e0, from=std::shared_ptr<const planning_scene::PlanningScene> (use count 8, weak count 1) = {...}, 
    to=std::shared_ptr<const planning_scene::PlanningScene> (use count 3, weak count 1) = {...}, jmg=0x555556f0d310, timeout=5, result=std::shared_ptr<robot_trajectory::RobotTrajectory> (empty) = {...}, 
    path_constraints=...) at /home/bob/ws/ros/catkin_ws/src/moveit_task_constructor/core/src/solvers/pipeline_planner.cpp:113
#10 0x00007fffed85e550 in moveit::task_constructor::stages::Connect::compute (this=0x55555718b0b0, from=..., to=...)
    at /home/bob/ws/ros/catkin_ws/src/moveit_task_constructor/core/src/stages/connect.cpp:157
#11 0x00007fffede31a8a in moveit::task_constructor::ConnectingPrivate::compute (this=0x5555570ee510) at /home/bob/ws/ros/catkin_ws/src/moveit_task_constructor/core/src/stage.cpp:672
#12 0x00007fffeddc4215 in moveit::task_constructor::StagePrivate::runCompute (this=0x5555570ee510)
    at /home/bob/ws/ros/catkin_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/stage_p.h:152
#13 0x00007fffeddbe529 in moveit::task_constructor::WrapperBase::compute (this=0x555557100b30) at /home/bob/ws/ros/catkin_ws/src/moveit_task_constructor/core/src/container.cpp:696
#14 0x00007fffeddb9ac1 in moveit::task_constructor::ContainerBasePrivate::compute (this=0x5555570da480) at /home/bob/ws/ros/catkin_ws/src/moveit_task_constructor/core/src/container.cpp:106
#15 0x00007fffeddc4215 in moveit::task_constructor::StagePrivate::runCompute (this=0x5555570da480)
    at /home/bob/ws/ros/catkin_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/stage_p.h:152
#16 0x00007fffeddbd113 in moveit::task_constructor::SerialContainer::compute (this=0x55555717f830) at /home/bob/ws/ros/catkin_ws/src/moveit_task_constructor/core/src/container.cpp:531
#17 0x00007fffeddb9ac1 in moveit::task_constructor::ContainerBasePrivate::compute (this=0x55555701a780) at /home/bob/ws/ros/catkin_ws/src/moveit_task_constructor/core/src/container.cpp:106
#18 0x00007fffeddc4215 in moveit::task_constructor::StagePrivate::runCompute (this=0x55555701a780)
    at /home/bob/ws/ros/catkin_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/stage_p.h:152
#19 0x00007fffeddbd113 in moveit::task_constructor::SerialContainer::compute (this=0x555555f01c60) at /home/bob/ws/ros/catkin_ws/src/moveit_task_constructor/core/src/container.cpp:531
[... cut ...]
#58 0x000055555567fb82 in main (argc=1, argv=0x7fffffffc168) at /home/bob/ws/ros/catkin_ws/src/bob/bob_planner/src/pp.cpp:1970

```

</p>
</details>

It seems that the [adaptAndPlan](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning/planning_pipeline/src/planning_pipeline.cpp#L229-L231) is not always triggered depending on the `adapter_chain_` variable... 

Do you know why there are times that the `ResolveConstraintFrames` is not always triggered for the same Connect stage?

v4hn, 2020-08-10:
> Do you know why there are times that the ResolveConstraintFrames is not always triggered for the same Connect stage?

No. The `adapter_chain_` is loaded from ROS parameters and should not change at any point in the lifetime of the pipeline.
I suspect an issue in your setup (possibly you have multiple pipelines and one does not define a chain?) but there might be some underlying issue in MoveIt.

cpt-yoshi, 2020-08-10:
> Do you know why there are times that the ResolveConstraintFrames is not always triggered for the same Connect stage?

The `adapter_chain_` is loaded correctly. My mistake, the adaptAndPlan is always triggered.

There are 2 paths when planning using [generatePlan](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning/planning_pipeline/src/planning_pipeline.cpp#L210-L213):

1. [isStateValid](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning/planning_pipeline/src/planning_pipeline.cpp#L231) -> The `OrientationConstraint` msg link_name and orientation will be changed if the link_name is a sub-frame.
1. [isPathValid](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning/planning_pipeline/src/planning_pipeline.cpp#L261)  -> The `OrientationConstraint` msg is not changed

After validating the State, it tries to validate the Path but if the `OrientationConstraint` link_name is a sub-frame, it fails to retrieve it.

For `State` validation the resolvedConstraints is applied, but for the `Path` validation it is not.
Is the above the intended behavior?

v4hn, 2020-08-10:
O.o No! you found a shortcoming in the `ResolveConstraintFrames` adapter that should be fixed in the upstream package.
The original authors likely forgot to apply and verify path constraints when they added the plugin.

cpt-yoshi, 2020-08-10:
> Looking at the sources though, at first glance it seems to transform path constraints as expected.

Yes that is correct. But the [planning_scene->isPathValid](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning/planning_pipeline/src/planning_pipeline.cpp#L261) does not pass through the adapter.

v4hn, 2020-08-10:
> But planning_scene->isPathValid does not pass through the adapter.

That's true. Implementing the frame resolver here will also render the ResolveConstraintsFrame request adapter obsolete, because if you have to transform the request in the scope of that method anyway, you might as well do it *before* passing them into the pipeline...

Could you please create an issue and/or pull-request in the MoveIt repository?

cpt-yoshi, 2020-08-12:
> Could you please create an issue and/or pull-request in the MoveIt repository?

Yes first thing in the morning!