moveit/moveit_task_constructor #195: Connect Stage Not Solving

danem, 2020-08-03:
I'm trying to get the pick and place demo working with my robot (Rethink Sawyer), and despite finding thousands of solutions for the other stages, it cannot solve for [move to pick](https://github.com/ros-planning/moveit_task_constructor/blob/master/demo/src/pick_place_task.cpp#L157). I've included a screen recording of the observed behavior (sorry for the poor quality): 

https://gfycat.com/snoopycloudygoat

As you can see, every sub task is solved for except move-to-pick. In fact, there doesn't appear to be any attempts made to solve the move-to-pick stage.

[Here](https://gist.github.com/danem/eac13ba375b9d7d813854616178591ad) are the modifications I made to get the demo working with my robot. I found that I had to explicitly specify the group property for the open/close stages.

Here is my [workspace](https://drive.google.com/file/d/1Wm8wmJmqgT0Wz5zzWxxsQCV4PmVxQJBK/view?usp=sharing) in case you want to dig deeper.

Thanks

danem, 2020-08-04:
Thanks for the insight, I wasn't aware of `rqt_logger_level`.

Looking at the output I'm seeing that it is failing at [this](https://github.com/ros-planning/moveit_task_constructor/blob/master/core/src/stage.cpp#L698) check. I'm unsure why that would be the case though given that the code is mostly untouched.

v4hn, 2020-08-04:
Well, which object changed its location and what's the difference. Once you know that it should be easier to debug...
Did you verify that this problem does not exist with the original demo in your workspace?

danem, 2020-08-05:
The original code running with the panda bot works without a hitch. I can't see any discernible difference between the two that would effect the positioning of the object. I've verified that the matrices reported [here](https://github.com/ros-planning/moveit_task_constructor/blob/master/core/src/stage.cpp#L737) are in fact different, but I'm unsure why there would be an attached body to begin with. Furthermore, skipping the check entirely hasn't fixed the issue.

v4hn, 2020-08-05:
> I can't see any discernible difference between the two that would effect the positioning of the object.

Neither can I.

> I'm unsure why there would be an attached body to begin with.

Very true! You probably looked at the log messages for the "move to place" Connect stage and they have nothing to do with your actual problem.

danem, 2020-08-05:
So I took a bit more time to dig through the logs and I discovered that the stage is failing at [this](https://github.com/ros-planning/moveit_task_constructor/blob/master/core/src/stages/connect.cpp#L121) check. This is more of a problem with my robot than with MTC. 

Thanks for the tips.

closed 2020-08-05 by danem