moveit/moveit_task_constructor #202: Limit number of stage solutions

rhaschke, 2020-08-27:
In tasks with many connect stages, the amount of solutions combinatorially explodes. To mitigate this problem, it might be meaningful to limit the overall number of solutions per stage/container.

v4hn, 2021-03-22:
I very much think this is necessary, especially when used with sampling stages that can run infinitely often.
We talked about this before and it is not entirely straight-forward to do it correctly.
If we simply allow to limit arbitrary stages we will loose planner completeness in the computation graph and the system can easily get stuck in local minima.

Instead, we should also disable/drop solutions if their partial solution fails entirely.
Lastly, to avoid getting stuck in local minima we could add an optimizing planner mode that disables the current best solution and all paths with higher costs.

This is quite an elaborated project though.

dazednconfusing, 2022-02-21:
Is there a way to enforce connection stage planning to be performed last? That way all planning can exit as soon as a complete solution is found for each sub-task and as long as generators sandwich the connect, the external interface should be unaffected.

rhaschke, 2022-02-22:
> Is there a way to enforce connection stage planning to be performed last?

No.

> all planning can exit as soon as a complete solution is found

If you just want to exit as soon as one solution is found, you can do so but providing `max_solutions=1` to `Task::plan()`.