moveit/moveit_task_constructor #203: Generator Stage (← →) "does not match" interface with Wrapper Stage (→ →)

DangMinh24, 2020-08-27:
Hi,

I tried to use the `stages::GenerateGraspPose` for a simple "pick" demo. If I understand correctly,  `stages::GenerateGraspPose` comes along with a Wrapper Stage. But I can not work with either `stages::PredicateFilter` or `stages::ComputeIK`. 

Minimal code:
```cpp
  Task t;
  t.stages()->setName("Grasping an object");
  // t.add(std::make_unique<stages::CurrentState>("current stage"));

  const std::string group = "robot_body";
  // t.add(createModule(group));
  // create Cartesian interpolation "planner" to be used in stages

  auto cartesian = std::make_shared<solvers::CartesianPath>();

  // create joint interpolation "planner"
  auto joint_interpolation =
      std::make_shared<solvers::JointInterpolationPlanner>();

  // create pipeline sampler for OMPL, STOMP, ... planners
  auto sampling_planner = std::make_shared<solvers::PipelinePlanner>();
  sampling_planner->setProperty("goal_position_tolerance", 5e-5);
  sampling_planner->setProperty("goal_joint_tolerance", 5e-5);
  sampling_planner->setProperty("num_planning_attempts", 3u);

  t.setProperty("group",
                "robot_body");  // group controll the whole body except gripper
  t.setProperty("eef", "gripper_ee");  // EE defined in MoveIt
  t.setProperty("hand", "gripper");    // * gripper group
  t.setProperty("hand_grasping_frame",
                "right_l6");  // Frame that "group" can control
  t.setProperty("ik_frame", "right_l6");
  /****************************************************
   *                                                  *
   *               Current State                      *
   *                                                  *
   ***************************************************/
  Stage* current_state_ptr =
      nullptr;  // Forward current_state on to grasp pose generator
  {
    auto current_state =
        std::make_unique<stages::CurrentState>("current state");

    current_state_ptr = current_state.get();
    t.add(std::move(current_state));
  }
  Stage* attach_object_stage = nullptr;
  {
    auto grasp = std::make_unique<SerialContainer>("pick object");
    t.properties().exposeTo(grasp->properties(),
                            {"eef", "hand", "group", "ik_frame"});
    grasp->properties().configureInitFrom(Stage::PARENT,
                                          {"eef", "hand", "group", "ik_frame"});
    {
      auto stage = std::make_unique<stages::MoveTo>("default grasping pose",
                                                    sampling_planner);
      stage->properties().configureInitFrom(Stage::PARENT, {"group", "eef"});
      stage->setTimeout(5.0);
      stage->setGoal("grasping_pose");
      grasp->insert(std::move(stage));
    }
    {
      // Sample grasp pose
      auto stage =
          std::make_unique<stages::GenerateGraspPose>("generate grasp pose");

      // stage->properties().configureInitFrom(Stage::PARENT);
      // stage->properties().set("marker_ns", "grasp_pose");
      stage->setEndEffector("gripper_ee");
      stage->setPreGraspPose("open_gripper");
      stage->setObject("object_A");
      stage->setAngleDelta(M_PI / 12);
      stage->setMonitoredStage(current_state_ptr);  // Hook into current

      // Compute IK
      auto wrapper = std::make_unique<stages::ComputeIK>("grasp pose IK",
                                                         std::move(stage));
      wrapper->properties().configureInitFrom(Stage::PARENT, {"eef", "group"});
      wrapper->setMaxIKSolutions(8);
      wrapper->setMinSolutionDistance(1.0);
      wrapper->setIKFrame("right_l6");
      wrapper->properties().configureInitFrom(Stage::INTERFACE,
                                              {"target_pose"});
      grasp->insert(std::move(wrapper));
    }
    t.add(std::move(grasp));
  }
```

Further details:
I defined `group` and `hand` (group for hand?), `eef` in MoveIt. Not quite understand about `ik_frame` and `hand_grasping_frame` (but I assumed it to be a frame that `group` can control). `grasping_pose`, 'open_gripper', 'close_gripper' are poses also defined in MoveIt. And I not quite understand about `setPreGraspPose()` -> assume to be `open_gripper`.
`object_A` is also defined in planning scene.

Please let me know if there is any configuration that I have missed.

v4hn, 2020-08-27:
You are missing a `Connect` stage between the `current state` and your `pick object` container to generate a motion between the current state of the system and the computed `default grasping pose`.
Please have a look at [the demo](https://github.com/ros-planning/moveit_task_constructor/blob/master/demo/src/pick_place_task.cpp#L156) we provide for pick and place.

closed 2020-08-27 by v4hn

DangMinh24, 2020-08-27:
Thanks @v4hn! It works beautifully! I did not catch idea of `stages::Connect`, now it makes lots of senses.