moveit/moveit_task_constructor #204: Property 'group': undefined

MichaelMano3, 2020-09-07:
Hi,
I'm using moveit task constructor to do pick and place with real robot which is "px100".
When i roslaunch moveit_task_constructor_demo pickplace.launch,i get this error.ｉdidn't change any code in pick_place_task.cpp.
I'm new to ROS so i don't know how to solve this.

For now it can creat table and object in the rviz but can't excute.

[image]

Here is the log.

... logging to /home/yang/.ros/log/a726d08a-f0db-11ea-9b2f-58fb84ea9fe8/roslaunch-yang-T50-30293.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://yang-T50:43341/

SUMMARY
========

PARAMETERS
 * /px100/mtc_tutorial/approach_object_max_dist: 0.15
 * /px100/mtc_tutorial/approach_object_min_dist: 0.1
 * /px100/mtc_tutorial/arm_group_name: interbotix_arm
 * /px100/mtc_tutorial/arm_home_pose: Sleep
 * /px100/mtc_tutorial/eef_name: interbotix_gripper
 * /px100/mtc_tutorial/execute: True
 * /px100/mtc_tutorial/grasp_frame_transform: [0, 0, 0.1, 1.571...
 * /px100/mtc_tutorial/hand_close_pose: Closed
 * /px100/mtc_tutorial/hand_frame: px100/ee_arm_link
 * /px100/mtc_tutorial/hand_group_name: interbotix_gripper
 * /px100/mtc_tutorial/hand_open_pose: Open
 * /px100/mtc_tutorial/lift_object_max_dist: 0.1
 * /px100/mtc_tutorial/lift_object_min_dist: 0.01
 * /px100/mtc_tutorial/max_solutions: 10
 * /px100/mtc_tutorial/object_dimensions: [0.02, 0.01]
 * /px100/mtc_tutorial/object_name: object
 * /px100/mtc_tutorial/object_pose: [0.2, 0.1, 0, 0, ...
 * /px100/mtc_tutorial/object_reference_frame: world
 * /px100/mtc_tutorial/place_pose: [0.2, -0.1, 0, 0,...
 * /px100/mtc_tutorial/place_surface_offset: 0.0001
 * /px100/mtc_tutorial/spawn_table: True
 * /px100/mtc_tutorial/surface_link: table
 * /px100/mtc_tutorial/table_dimensions: [0.3, 0.3, 0.02]
 * /px100/mtc_tutorial/table_name: table
 * /px100/mtc_tutorial/table_pose: [0.3, 0, 0, 0, 0, 0]
 * /px100/mtc_tutorial/table_reference_frame: world
 * /px100/mtc_tutorial/world_frame: world
 * /rosdistro: melodic
 * /rosversion: 1.14.9

NODES
  /px100/
    mtc_tutorial (moveit_task_constructor_demo/pick_place_demo)

ROS_MASTER_URI=http://localhost:11311

process[px100/mtc_tutorial-1]: started with pid [30317]
[ INFO] [1599463869.936462101]: Init moveit_task_constructor_demo
[ INFO] [1599463871.002693318]: Loading task parameters
[ INFO] [1599463871.029270207]: Initializing task pipeline
[ INFO] [1599463871.039186961]: Loading robot model 'px100'...
[ INFO] [1599463871.039261454]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ WARN] [1599463871.104039924]: IK plugin for group 'interbotix_arm' relies on deprecated API. Please implement initialize(RobotModel, ...).
[ INFO] [1599463871.106319218]: IK Using joint px100/shoulder_link -3.14159 3.14159
[ INFO] [1599463871.106361157]: IK Using joint px100/upper_arm_link -1.93732 1.8675
[ INFO] [1599463871.106391692]: IK Using joint px100/forearm_link -1.6057 2.11185
[ INFO] [1599463871.106403507]: IK Using joint px100/gripper_link -2.14675 1.74533
[ INFO] [1599463871.106446482]: Looking in common namespaces for param name: interbotix_arm/position_only_ik
[ INFO] [1599463871.108074994]: Looking in common namespaces for param name: interbotix_arm/solve_type
[ INFO] [1599463871.111232116]: Using solve type Speed
[ INFO] [1599463871.111645235]: Start searching for task solutions
[ INFO] [1599463871.155865833]: Using planning interface 'OMPL'
[ INFO] [1599463871.158791724]: Param 'default_workspace_bounds' was not set. Using default value: 10
[ INFO] [1599463871.159191961]: Param 'start_state_max_bounds_error' was set to 0.1
[ INFO] [1599463871.159553100]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1599463871.160024353]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1599463871.160313669]: Param 'jiggle_fraction' was set to 0.05
[ INFO] [1599463871.160703439]: Param 'max_sampling_attempts' was not set. Using default value: 100
[ INFO] [1599463871.160790662]: Using planning request adapter 'Add Time Parameterization'
[ INFO] [1599463871.160828917]: Using planning request adapter 'Resolve constraint frames to robot links'
[ INFO] [1599463871.160879953]: Using planning request adapter 'Fix Workspace Bounds'
[ INFO] [1599463871.160900203]: Using planning request adapter 'Fix Start State Bounds'
[ INFO] [1599463871.160918406]: Using planning request adapter 'Fix Start State In Collision'
[ INFO] [1599463871.160935809]: Using planning request adapter 'Fix Start State Path Constraints'
[ INFO] [1599463871.187195350]: Planner configuration 'interbotix_gripper' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1599463871.187533342]: interbotix_gripper/interbotix_gripper: Starting planning with 1 states already in datastructure
[ INFO] [1599463871.197846757]: interbotix_gripper/interbotix_gripper: Created 4 states (2 start + 2 goal)
[ INFO] [1599463871.197877661]: Solution found in 0.010418 seconds
[ INFO] [1599463871.198739333]: SimpleSetup: Path simplification took 0.000802 seconds and changed from 3 to 2 states
terminate called after throwing an instance of 'std::runtime_error'
  what():  Property 'group': undefined
in stage 'close hand': declared, but undefined, inherits from parent
in stage 'pick object': defined here
[px100/mtc_tutorial-1] process has died [pid 30317, exit code -6, cmd /home/yang/ws_moveit/devel/lib/moveit_task_constructor_demo/pick_place_demo __name:=mtc_tutorial __log:=/home/yang/.ros/log/a726d08a-f0db-11ea-9b2f-58fb84ea9fe8/px100-mtc_tutorial-1.log].
log file: /home/yang/.ros/log/a726d08a-f0db-11ea-9b2f-58fb84ea9fe8/px100-mtc_tutorial-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

And the rqt_graph:

[image]

Is there someplace I need to configure for my robot in the pick_place_task.cpp?

rhaschke, 2020-09-07:
That's a bug in the demo. See https://github.com/ros-planning/moveit_task_constructor/pull/205 for a fix.

closed 2020-09-07 by v4hn

MichaelMano3, 2020-09-07:
> That's a bug in the demo. See #205 for a fix.

Thank you for your replay.I follow the Fix pick-place you send me,
delete:  stage->properties().property("group").configureInitFrom(Stage::PARENT, hand_group_name_);
add: stage->setGroup(hand_group_name_);
But it seems have same problem:Property 'group': undefined. I already delete "group" here,so maybe this  error is from somewhere else.
[image]

rhaschke, 2020-09-07:
Do you get the exact same error (for the exact same stage)? #205 applies the same change _twice_: for _open_ and _close_ hand. Did you apply it twice as well?

MichaelMano3, 2020-09-07:
> Do you get the exact same error (for the exact same stage)? #205 applies the same change _twice_: for _open_ and _close_ hand. Did you apply it twice as well?

Yes,i get same error on the same stage.I changed both Close Hand and Open Hand.

rhaschke, 2020-09-07:
Then, probably you didn't recompile, or it's still using the old code for some reason. For sure, there is no inheritance configured anymore.

PR #205 (2020-09-07): Fix pick-place-demo: Correctly set hand group
In the panda example, the `hand` group was "hand" as well, so it didn't make a difference... However, if the group has a different name, it's important to either inherent from the correct parent property (`hand`) or to directly set the correct name.

Fixes #204.