moveit/moveit_task_constructor #209: ComputeIk sometimes fails to check collision

JonasTietz, 2020-09-23:
Hello,
I have noticed that sometimes the solutions of the ComputeIk node are definitely in collision but get reported as success.

[image]

For me it is not a big problem right now, because the collision gets detected once a trajectory is planned to that pose.

rhaschke, 2020-09-24:
This is weird. Haven't noticed this before. Can you provide a minimal failing example for further debugging, please?

v4hn, 2020-09-24:
This might be related to the robot models. I have seen this issue in the past as well with the same robot model (our custom PR2), but did not get a minimal example to work.
It seems that collision checking fails under some circumstances for the elbow meshes and primitive cubes.

As @JonasTietz also states, this is not safety-critical because the surrounding stages will reject this state as start/end of a trajectory, so I did not investigate too much yet.

rhaschke, 2020-09-24:
If the collision is detected by the surrounding stage in the exactly same start/end state, we do have a bug. 
If the collision is only detected after some additional robot motion, the issue is probably due to your meshes as Michael suggested.
So, I would leave it as a task for @JonasTietz to validate which of those two situations we have at hand.

v4hn, 2020-09-24:
Of course he is free to inspect further ;) but he should also get on with his actual research topic. :)

Either way there is an issue.

Schteve-Bastian, 2021-08-19:
I'm encountering the same issue. IK plans through our container, but generally the following stages are unable to plan to/from the collision state.
[image]

PR #286 (2021-08-23): Fix some issues with ComputeIK
Fixes #209
closed 2021-08-23 by rhaschke