moveit/moveit_task_constructor #211: MoveTo stage and attached collision object disappearing

YannickRiou, 2020-10-21:
I found a strange behavior while using MoveTo stage to move the robot arm to a pre-defined Moveit pose.

Before planning the task, my robot have an object attached to his gripper.
When reviewing the planned path using Rviz and Motion Planning Task panel, the planned path show the movement of the arm with the grasped object, but at the end of the trajectory it just disappear.
After executing the task, the object just disappear from the planning scene.

A minimal example can be found here https://github.com/YannickRiou/moveit_collision_pb and running pr2.launch. There error "Attached body 'obj_230' not found" appear even if solution is found. 

Is it just a misuse of this stage or is it a bug on planning scene handling with this stage ?

rhaschke, 2020-10-26:
Attached objects are part of the `RobotState`, not part of the scene. Thus, your `robot_trajectory` doesn't include the attached object. To understand the issue, I tried to reproduce it locally.
However, that wasn't easy as your minimal example - again - was missing named robot states, i.e. `start` and `end`. 
I just introduced some dummy states for them and got the example running. Please ensure that your examples are working and produce the observed issue!
But: I cannot reproduce the issue with latest branch [`forward-master`](https://github.com/ros-planning/moveit_task_constructor/tree/forward-master). Are you sure, you are up-to-date?

YannickRiou, 2020-10-28:
I updated the moveit_collision_pb repo with a moveTo branch using the forward-master branch for moveit_task_constructor repo (tested locally so now eveyrthing should be ok according to start and end poses). After commenting the assert because of this error : 
```
moveit_collision_pb/moveit_task_constructor/core/src/cost_terms.cpp:87:21: error: cannot convert ‘const SubTrajectorySignature {aka const std::function<double(const moveit::task_constructor::SubTrajectory&, std::__cxx11::basic_string<char>&)>}’ to ‘bool’ in initialization
  assert(bool{ term_ });
```

The result stay the same. This error ```[ERROR] [1603871135.336511739]: Attached body 'obj_230' not found``` is still there and the object disappear after the moveTo "end". 
Here is what's happening : 
[image]

And the log of the launch : https://gist.github.com/YannickRiou/951baf4dc6f2598058c804d8eeedc251

JafarAbdi, 2020-10-28:
I was able to reproduce it locally, the reason is this line is overriding the whole robot state including the attached objects

> https://github.com/ros-planning/moveit_task_constructor/blob/be270cb57466d456ede2131d6cfaa8257b8428d2/core/src/stages/move_to.cpp#L256

The solution is to copy back the attached objects after setting the robot state
```
diff --git a/core/src/stages/move_to.cpp b/core/src/stages/move_to.cpp
index f9f915c..72bc2c9 100644
--- a/core/src/stages/move_to.cpp
+++ b/core/src/stages/move_to.cpp
@@ -254,6 +254,13 @@ bool MoveTo::compute(const InterfaceState& state, planning_scene::PlanningSceneP
 	// store result
 	if (robot_trajectory) {
 		scene->setCurrentState(robot_trajectory->getLastWayPoint());
+		std::vector<const moveit::core::AttachedBody*> attached_bodies;
+		state.scene()->getCurrentState().getAttachedBodies(attached_bodies);
+		for (const auto& attached_body : attached_bodies)
+			scene->getCurrentStateNonConst().attachBody(
+			    attached_body->getName(), attached_body->getShapes(), attached_body->getFixedTransforms(),
+			    attached_body->getTouchLinks(), attached_body->getAttachedLinkName(), attached_body->getDetachPosture(),
+			    attached_body->getSubframeTransforms());
 		if (dir == BACKWARD)
 			robot_trajectory->reverse();
 		solution.setTrajectory(robot_trajectory);
```

rhaschke, 2020-10-29:
@JafarAbdi, as pointed out in https://github.com/ros-planning/moveit_task_constructor/issues/211#issuecomment-716876914, the crucial question is: Why does the `RobotState` in `robot_trajectory` loose the attached objects?
As the planning starts from the correct RobotState (including all attached objects), the resulting trajectory should comprise them as well. If that's not the case, there is a bug somewhere. Your proposed solution is just a workaround for the bug.

rhaschke, 2020-10-29:
Further reducing the example to the even simpler sequence `CurrentState - AttachObject - MoveTo` works as expected.

JafarAbdi, 2020-10-29:
> Why does the `RobotState` in `robot_trajectory` loose the attached objects?

After looking into the MoveIt source code, `planner_->generatePlan` will eventually call [ompl_interface::ModelBasedPlanningContext::solve](https://github.com/ros-planning/moveit/blob/master/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L677) and once it find a solution the result waypoint will be added to the trajectory in [ompl_interface::ModelBasedPlanningContext::convertPath](https://github.com/ros-planning/moveit/blob/master/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L434) which copy `complete_initial_robot_state_` first, could it be becuase we don't specify the start state in [pipeline_planner](https://github.com/ros-planning/moveit_task_constructor/blob/master/core/src/solvers/pipeline_planner.cpp#L86) .? or the robot state in the planning scene `from` don't have the attached objects .?

I'll look into this issue again this week and will share my findings

rhaschke, 2020-10-29:
I found the culprit: `FixStartStateBounds` was overriding all attached objects - but only if it actually adapted the state.
This caused the pipeline planner to work with no attached objects and [`convertPath()`](https://github.com/ros-planning/moveit/blob/a461efe7dbef01328e3d72c1c83e66dce050eb2c/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L437) to initialize from a `complete_initial_robot_state_`, which wasn't actually complete.
I filed a fix in MoveIt: https://github.com/ros-planning/moveit/pull/2398

YannickRiou, 2020-11-02:
The fix that you provided solves the problem on my side ! That is great ! I will have to do more test to see if it also solves #2262 but the explanation that you gave in the fix seems to be right !

closed 2020-11-02 by YannickRiou