moveit/moveit_task_constructor #212: Is it possible for reverse trajectory in MTC?

DangMinh24, 2020-10-22:
Hi all,

I wonder is it possible to move "reverse" in MTC or necessary to generate new stage with opposite parameters. For instance, is it possible to move "reverse" for this particular stage:

```cpp
  /**********************
   * MOVE Y -0.15 RELATIVE *
   **********************/
  {
    auto stage =
        std::make_unique<moveit::task_constructor::stages::MoveRelative>(
            "move y-0.15", sampling_planner);
    stage->properties().configureInitFrom(
        moveit::task_constructor::Stage::PARENT);
    stage->setGroup("robot_body");

    geometry_msgs::Vector3Stamped direction;
    direction.header.frame_id = "base";
    direction.vector.y = -0.15;
    stage->setDirection(direction);

    t.add(std::move(stage));
  }
```

or have to generate opposite stage such as:

```cpp
  /**********************
   * MOVE Y +0.15 RELATIVE *
   **********************/
  {
    auto stage =
        std::make_unique<moveit::task_constructor::stages::MoveRelative>(
            "move y+0.15", sampling_planner);
    stage->properties().configureInitFrom(
        moveit::task_constructor::Stage::PARENT);
    stage->setGroup("robot_body");

    geometry_msgs::Vector3Stamped direction;
    direction.header.frame_id = "base";
    direction.vector.y = +0.15;
    stage->setDirection(direction);

    t.add(std::move(stage));
  }
```

Thanks

rhaschke, 2020-10-22:
It's perfectly possible to "plan reversely". Let's say you have a known target stage X, e.g. grasping an object.
Then you can just _prepend_ this state with a `MoveRelative` stage. The corresponding motion is then planned in reverse direction and executed _towards_ the target stage X.

closed 2020-10-22 by rhaschke

JafarAbdi, 2020-10-30:
@rhaschke Is there a way to reverse a solution with a custom min_distance/max_distance

GenerateGraspPose -> MoveRelative (with custom min_distance/max_distance) -> stage that exactly reverse the previous solution found by MoveRelative

rhaschke, 2020-10-30:
I don't see a need to explicitly reverse a trajectory. You can just plan in reverse direction.
How would you apply the reversed trajectory? Within the task sequence it's not applicable as the end start probably doesn't match the current state anymore. The only application I could imagine is to store the grasping trajectory in reverse order for later releasing.

DangMinh24, 2020-10-30:
Thanks @rhaschke. I was thinking about scenario: once robot move on specific trajectory from A to B, is it possible to re-use the same trajectory to "retreat" back from B to A (With the hope to optimize performance of some repetitive and complete actions such as rotating a robot hand k-rounds).