moveit/moveit_task_constructor #218: Task report success despite one of the stages always failing

JafarAbdi, 2020-11-10:
I have a bug where the task report success despite one of the stages never generate a successful solution (in particular Connect), as you can see in the video below the task did succeed but the number of successful solutions in `move to pick` is zero, when I click on the task to visualize it somehow a solution get generated, also, executing the task start from the new solution of `move to pick`

It should be reproducible using this [branch] (https://github.com/JafarAbdi/moveit_task_constructor/tree/task_bug)(I just added a constraint to make the stage always fail)

- Tested on Ubuntu 18.04 - Melodic - the master branch of MoveIt - both master branch of MTC and https://github.com/ros-planning/moveit_task_constructor/pull/216
- [Video link](https://youtu.be/yy-oe3NU-U8)

JafarAbdi, 2020-11-12:
I think I found the bug 

When plan inside [Connect::compute](https://github.com/ros-planning/moveit_task_constructor/blob/master/core/src/stages/connect.cpp#L133) [fails](https://github.com/ros-planning/moveit_task_constructor/blob/master/core/src/stages/connect.cpp#L173) it will call [makeSequential](https://github.com/ros-planning/moveit_task_constructor/blob/master/core/src/stages/connect.cpp#L180) which will insert a subsolution not marked as a failure with setting the start/end states (they will call addOutgoing/addIncoming) the returned SolutionSequence will be marked as failure [here](https://github.com/ros-planning/moveit_task_constructor/blob/master/core/src/stages/connect.cpp#L175) but this doesn't mark the subsolutions as failure

@rhaschke any idea how this could be fixed .? should we pass the success flag and mark the subsolutions as a failure, too .?

rhaschke, 2020-11-12:
Sorry for being unresponsive. I'm quite busy these days. I will try to look into this issue over the weekend. There is some other pending bug in Connect...

mamoll, 2020-12-02:
@rhaschke what is the other pending bug you're referring to? Perhaps we can help. We have a workaround for the bug Jafar reported, but it'd be nice to use a "clean" solution.

rhaschke, 2020-12-03:
The other bug is #182. But, I didn't have time yet to look into both of these issues.

PR #220 (2020-12-08): Fix SerialContainer
Fixes #218. Fixes #182.
The `SerialContainer` was accepting solution sequences comprising a failure-only `Connect` stage, because the `Connect` stage didn't correctly mark its sequentialized subsolutions as a failure (but only the overall sequence). @JafarAbdi, please validate that this solves your issue.
closed 2020-12-08 by rhaschke