moveit/moveit_task_constructor #222: Demo broken

gautz, 2020-12-15:
When running ``rosrun moveit_task_constructor_demo scanning``
My computer name has some ``-``sign, I get following error:
```
terminate called after throwing an instance of 'ros::InvalidNameException'
  what():  Character [-] at element [16] is not valid in Graph Resource Name [get_solution_ComputeNameWith-].  Valid characters are a-z, A-Z, 0-9, / and _.
Aborted (core dumped)
```
I can traceback the regression to some time after commit be270cb57466d456ede2131d6cfaa8257b8428d2

JafarAbdi, 2020-12-15:
@gautz This's should be fixed in https://github.com/ros-planning/moveit_task_constructor/pull/223

closed 2020-12-15 by rhaschke

gautz, 2020-12-16:
thanks for the quick fix, execution works! I am still getting an error that may be unrelated.
``Unregistered type: St3mapINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEdSt4lessIS5_ESaISt4pairIKS5_dEEE``

JafarAbdi, 2020-12-16:
> thanks for the quick fix, execution works! I am still getting an error that may be unrelated.
> `Unregistered type: St3mapINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEdSt4lessIS5_ESaISt4pairIKS5_dEEE`

This corresponds to the following type `std::map<std::string, double>` are you trying to set a property's value to `std::map` .?

gautz, 2020-12-16:
yes, in cartesian.cpp on line 106:
``std::map<std::string, double> offsets = { { "panda_joint1", M_PI / 6. }, { "panda_joint3", -M_PI / 6 } };``

rhaschke, 2020-12-16:
You can safely ignore this warning. It just means that you cannot inspect the value of this property in rviz.