moveit/moveit_task_constructor #224: Motion Planning Tasks Rviz plugin is keeping old tasks

MarqRazz, 2020-12-23:
I am seeing an issue where the Rviz plugin is holding onto all plans since startup. This can make for a very large "Task Tree" and after a new task is received the old ones are no longer valid so you can't see any of the solutions.
[image]

rhaschke, 2020-12-23:
This is on purpose: it allows you to inspect _received_ solutions even though the corresponding mtc node already finished.
However, you are right, this is limited to actually published solutions (typically top-level ones).
In the master branch you are able to customize how the plugin will handle new incoming tasks that override existing old ones:
You can choose between keep (all), replace existing, and always remove. Just go into the settings tab:

[image]

MarqRazz, 2020-12-23:
I figured there would be an option for this. Thanks for pointing me to the settings!!

closed 2020-12-23 by MarqRazz