moveit/moveit_task_constructor #226: pipeline lanner pilz

gautz, 2021-01-15:
Using planning interface 'Pilz Industrial Motion Planner'
[ INFO] [1610719761.358863247]: Generating CIRC trajectory...
[ERROR] [1610719761.359009442]: No path constraint named "interim" or "center" found (found unknown constraint: "req.path_constraints.name" instead)

gautz, 2021-01-15:
in the Properties tab of the task pannel in RViz I get:
``path_constraints -> YAML error``
``name -> could not find expected ':'``

jschleicher, 2021-01-18:
No, RViz isn't implemented, but your C++ example should work. Could you post the final MotionPlanRequest sent to the planner? Something like `rostopic echo /move_group/goal` or `sequence_move_group/goal` should print the message, if you're using the action inferface.

rhaschke, 2021-01-18:
Planning with CIRC works as expected: https://github.com/ros-planning/moveit_task_constructor/tree/pilz-circ-example:
https://github.com/ros-planning/moveit_task_constructor/blob/5934e4dc6bf77bd0a46c8765db7e0bce25addebc/demo/src/cartesian.cpp#L74-L97

@jschleicher: As MTC isn't employing the service or action interface, it's not possible to observe the MotionPlanRequest from cmdline.

In rviz, the properties are not correctly displayed, indeed. That's a known (but won't-fix) issue of C++ message serialization to yaml.

gautz, 2021-01-18:
Thanks for helping with debugging. Indeed CIRC also works now.
I could now identify that the issue is raised when a moveto CIRC stage is run after a moveto LIN stage.

rhaschke, 2021-01-18:
To debug this isse, it would be helpful if you augment the above example to reproduce the issue.

gautz, 2021-01-18:
So instantiating the solver and setting the plannerID in the block defining the MoveTo stage solved the problem. As instead I used to define it at the task definition level.
Before:
```
 Task createTask()
{
	Task t;
	t.stages()->setName("Cartesian path");
	const std::string group = "manipulator";
	auto pilz_planner = std::make_shared<solvers::PipelinePlanner>();
 	pilz_planner->setPlannerId("CIRC"); 
 	pilz_planner->setProperty("max_acceleration_scaling_factor", 0.1); 
 	pilz_planner->setProperty("max_velocity_scaling_factor", 0.1); 

    { 
    auto stage = std::make_unique<stages::MoveTo>("pilz", pilz_planner); 
    stage->setGroup(group); 
    geometry_msgs::PoseStamped goal; 
    goal.header.frame_id = "world"; 
    goal.pose.orientation.x = 1; 
    goal.pose.position.x = 0.45; 
    goal.pose.position.z = 0.45; 
    stage->setGoal(goal); 
    moveit_msgs::Constraints path_constraints; 
    path_constraints.name = "interim"; 
    moveit_msgs::PositionConstraint pos_constraint; 
    goal.pose.position.x = 0.3; 
    pos_constraint.constraint_region.primitive_poses.push_back(goal.pose); 
    pos_constraint.link_name = "panda_link8"; 
    pos_constraint.header.frame_id = "world"; 
    path_constraints.position_constraints.push_back(pos_constraint); 
    stage->setPathConstraints(path_constraints); 

    pilz_planner->setPlannerId("CIRC");

    t.add(std::move(stage)); 
    } 
}
```
After:
```
 Task createTask()
{
    Task t;
    t.stages()->setName("Cartesian path");
    const std::string group = "manipulator";
    { 
    auto pilz_planner = std::make_shared<solvers::PipelinePlanner>();
    pilz_planner->setPlannerId("CIRC"); 
    pilz_planner->setProperty("max_acceleration_scaling_factor", 0.1); 
    pilz_planner->setProperty("max_velocity_scaling_factor", 0.1); 
    pilz_planner->setPlannerId("CIRC");

    auto stage = std::make_unique<stages::MoveTo>("pilz", pilz_planner); 
    stage->setGroup(group); 
    geometry_msgs::PoseStamped goal; 
    goal.header.frame_id = "world"; 
    goal.pose.orientation.x = 1; 
    goal.pose.position.x = 0.45; 
    goal.pose.position.z = 0.45; 
    stage->setGoal(goal); 
    moveit_msgs::Constraints path_constraints; 
    path_constraints.name = "interim"; 
    moveit_msgs::PositionConstraint pos_constraint; 
    goal.pose.position.x = 0.3; 
    pos_constraint.constraint_region.primitive_poses.push_back(goal.pose); 
    pos_constraint.link_name = "panda_link8"; 
    pos_constraint.header.frame_id = "world"; 
    path_constraints.position_constraints.push_back(pos_constraint); 
    stage->setPathConstraints(path_constraints); 
    t.add(std::move(stage)); 
    } 
}
```

rhaschke, 2021-01-18:
The two code snippets shouldn't make a difference. However, as you mentioned, you use the pilz planner for two stages: If you change the PlannerId in between, only the last config will be used: The planner is a shared pointer instance, which is reused!

gautz, 2021-02-26:
For my use-case instantiating the solver in each stage with corresponding PlannerId and velocity did the job. So I would close this.
Thanks for all your inputs!

closed 2021-02-26 by gautz