moveit/moveit_task_constructor #229: pick_place_demo: "Planning failed"?

EricCousineau-TRI, 2021-01-22:
Using same instructions from #227, if I try to run the `pick_place_demo`, it fails:

https://ros-planning.github.io/moveit_tutorials/doc/moveit_task_constructor/moveit_task_constructor_tutorial.html#running-the-demo

See video:
https://user-images.githubusercontent.com/26719449/105537844-85036300-5cc0-11eb-8c12-da4e4770be7c.mp4

```
# Terminal 1
$ roslaunch moveit_task_constructor_demo demo.launch

# Terminal 2
$ roslaunch moveit_task_constructor_demo pickplace.launch
...
[ INFO] [1611344549.564996032]: Init moveit_task_constructor_demo
[ INFO] [1611344550.606822895]: Loading task parameters
[ INFO] [1611344550.621163968]: Initializing task pipeline
[ INFO] [1611344550.627964341]: Loading robot model 'panda'...
[ WARN] [1611344550.628191481]: Link panda_leftfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[ WARN] [1611344550.628246251]: Link panda_rightfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[ INFO] [1611344550.713280666]: Start searching for task solutions
[ INFO] [1611344550.788253292]: Using planning interface 'OMPL'
[ INFO] [1611344550.793196561]: Param 'default_workspace_bounds' was not set. Using default value: 10
[ INFO] [1611344550.793654999]: Param 'start_state_max_bounds_error' was set to 0.1
[ INFO] [1611344550.794475457]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1611344550.794984097]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1611344550.795436249]: Param 'jiggle_fraction' was set to 0.05
[ INFO] [1611344550.795857186]: Param 'max_sampling_attempts' was not set. Using default value: 100
[ INFO] [1611344550.795929977]: Using planning request adapter 'Add Time Parameterization'
[ INFO] [1611344550.795963161]: Using planning request adapter 'Resolve constraint frames to robot links'
[ INFO] [1611344550.795996345]: Using planning request adapter 'Fix Workspace Bounds'
[ INFO] [1611344550.796025351]: Using planning request adapter 'Fix Start State Bounds'
[ INFO] [1611344550.796054107]: Using planning request adapter 'Fix Start State In Collision'
[ INFO] [1611344550.796081219]: Using planning request adapter 'Fix Start State Path Constraints'
[ INFO] [1611344550.808541497]: Planner configuration 'hand' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1611344550.809007327]: hand/hand: Starting planning with 1 states already in datastructure
[ INFO] [1611344550.821122754]: hand/hand: Created 5 states (2 start + 3 goal)
[ INFO] [1611344550.821173753]: Solution found in 0.012305 seconds
[ INFO] [1611344550.831344122]: SimpleSetup: Path simplification took 0.010085 seconds and changed from 4 to 2 states
  0  - ←   0 →   -  0 / pick_place_task
    1  - ←   1 →   -  0 / applicability test
      1  - ←   1 →   -  1 / current state
    -  0 →   1 →   -  1 / open hand
    -  1 →   0 ←   0  - / move to pick
    0  - ←   0 →   -  0 / pick object
      0  - ←   0 ←   0  - / approach object
      0  - ←   0 →   -  0 / grasp pose IK
       25  - ←  25 →   - 25 / generate grasp pose
      -  0 →   0 →   -  0 / allow collision (hand,object)
      -  0 →   0 →   -  0 / close hand
      -  0 →   0 →   -  0 / attach object
      -  0 →   0 →   -  0 / allow collision (object,support)
      -  0 →   0 →   -  0 / lift object
      -  0 →   0 →   -  0 / forbid collision (object,surface)
    -  0 →   0 ←   0  - / move to place
    0  - ←   0 →   -  0 / place object
      0  - ←   0 ←   0  - / lower object
      0  - ←   0 →   -  0 / place pose IK
        0  - ←   0 →   -  0 / generate place pose
      -  0 →   0 →   -  0 / open hand
      -  0 →   0 →   -  0 / forbid collision (hand,object)
      -  0 →   0 →   -  0 / detach object
      -  0 →   0 →   -  0 / retreat after place
    -  0 →   0 →   -  0 / move home
[ERROR] [1611344550.835211666]: Planning failed
[ INFO] [1611344550.835244636]: Planning failed
```

EricCousineau-TRI, 2021-01-22:
FWIW Maybe the collisions for the panda description changed?

From vid @ 20s:
> [image]

rhaschke, 2021-01-22:
Yes, unfortunately, Franka Emika guys released a new Panda `robot_description` that is _always_ in collision.
I'm still using the old robot descriptions available via https://github.com/ros-planning/moveit_resources.

HauNan, 2021-02-22:
> Yes, unfortunately, Franka Emika guys released a new Panda `robot_description` that is _always_ in collision.
> I'm still using the old robot descriptions available via https://github.com/ros-planning/moveit_resources.

@rhaschke Hi rhaschke,I'm suffering from the same problem and I try the old robot descriptions you mentioned in the [site.](https://github.com/ros-planning/moveit_resources.) ,but both 0.7.0 and 0.6.5 are not work.

I try it in Ubuntu 18.04 / ROS Melodic. It is some thing wrong will happen even though I used the old version of robot descriptions in ROS Melodic but not ROS Noetic?

If not, would you please provide the historial version tag of the Panda robot_description that work without in collision? Thank you very much!

[image]

rhaschke, 2021-02-22:
Note that the `franka_description` package from `moveit_resources` is named `moveit_resources_panda_description`.
I just noticed that Franka also updated the description package in Melodic to 0.7.1 recently.

v4hn, 2021-02-22:
@rhaschke Using `moveit_resources_panda_*` should be the default option with [your recent patch](https://github.com/ros-planning/moveit_task_constructor/pull/231). This includes the description, so I'm not sure this is still relevant here?

rhaschke, 2021-02-22:
I agree that this should be closed as #231 was merged.

HauNan, 2021-02-23:
> I agree that this should be closed as #231 was merged.

@rhaschke thank you very much！
I try #231 in the gpd demo and now it work smoothly！
[image]

EricCousineau-TRI, 2021-02-23:
Sweet! I'm going to close this given #231 as y'all mentioned - thanks!

closed 2021-02-23 by EricCousineau-TRI