moveit/moveit_task_constructor #233: Invalid Trajectories Created

danem, 2021-02-16:
Using MTC f99d03386ad8751fb0ade71c09a9810019a950ad.

I am successfully planning a pick operating using MTC. MTC reports that the solution has been successfully found and the plan is correctly visualized in Rviz. However, when trying to execute this solution using the Execute capability, the [generated trajectory ](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning/plan_execution/src/plan_execution.cpp#L371 )contains bad data which is rejected by my trajectory controller with the error "Trajectory message contains waypoints that are not strictly increasing in time.". 

What could be causing the discrepancy between the visualized trajectory and the one being sent to the trajectory controller?

An example of what is sent to the `follow_trajectory` endpoint:

```
 points: 
      - 
        positions: [0.7397, -0.0030679615234375, -0.020325245092773437, 2.0635876197021483, -3.1392916288574217, 0.16873788378906251, 3.1396751240478515]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        effort: []
        time_from_start: 
          secs: 0
          nsecs:         0
      - 
        positions: [0.7397001240013277, -0.03246895310226966, -0.02441177174213565, 2.07099413161763, -3.102367452038901, 0.2030662516651845, 3.1193793785168675]
        velocities: [8.835592602350391e-07, -0.20949387274113646, -0.029118143568025733, 0.052774371929507624, 0.2630995889628328, 0.2446034077445519, -0.14461533788934003]
        accelerations: [3.3711408699462948e-06, -0.7993050236676782, -0.11109765707829938, 0.20135586808443628, 1.0038328111044719, 0.9332622957333059, -0.5517668109423922]
        effort: []
        time_from_start: 
          secs: 0
          nsecs: 198462088 points: 

/ ...
// Bad data starts
///
---
header: 
  seq: 23
  stamp: 
    secs: 1613168986
    nsecs: 518691260
  frame_id: ''
goal_id: 
  stamp: 
    secs: 1613168986
    nsecs: 518692193
  id: "/move_group-38-1613168986.518692193"
goal: 
  trajectory: 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: "world"
    joint_names: [Joint_1, Joint_2, Joint_3, Joint_4, Joint_5, Joint_6]
    points: 
      - 
        positions: [-1.7745037288831231, -0.21050136321802213, 2.405817313134492, -1.6909986578677245, 1.405934550892721, 2.2052587505959518]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        effort: []
        time_from_start: 
          secs: 0
          nsecs:         0
      - 
        positions: [-1.7745037288831231, -0.21050136321802213, 2.405817313134492, -1.6909986578677245, 1.405934550892721, 2.2052587505959518]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        effort: []
        time_from_start: 
          secs: 0
          nsecs:         0
      - 
        positions: [-1.7745037288831231, -0.21050136321802213, 2.405817313134492, -1.6909986578677245, 1.405934550892721, 2.2052587505959518]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        effort: []
        time_from_start: 
          secs: 0
          nsecs:         0
```

[Here](https://giant.gfycat.com/MarvelousFamousCod.webm) is a video of the described behavior.

And [here](https://gist.github.com/danem/515d1d9ae7c116eadd9e04b6cf7000dc) is the code I use to plan and execute my task.

rhaschke, 2021-02-17:
Looks like, there is no timing planned for the trajectory. Did you add a planning request adapter for time parameterization to all planners?
As rviz doesn't care about the timing by default (you would need to switch to realtime display to see this), you don't notice it there.

danem, 2021-02-17:
Thanks for the prompt response.

I believe so. Here is moveit on start up:

```
ros.moveit_ros_planning: Using planning interface 'OMPL'
ros.moveit_ros_planning: Param 'default_workspace_bounds' was not set. Using default value: 10
ros.moveit_ros_planning: Param 'start_state_max_bounds_error' was set to 0.1
ros.moveit_ros_planning: Param 'start_state_max_dt' was not set. Using default value: 0.5
ros.moveit_ros_planning: Param 'start_state_max_dt' was not set. Using default value: 0.5
ros.moveit_ros_planning: Param 'jiggle_fraction' was set to 0.05
ros.moveit_ros_planning: Param 'max_sampling_attempts' was not set. Using default value: 100
ros.moveit_ros_planning: Using planning request adapter 'Add Time Parameterization'
ros.moveit_ros_planning: Using planning request adapter 'Resolve constraint frames to robot links'
ros.moveit_ros_planning: Using planning request adapter 'Fix Workspace Bounds'
ros.moveit_ros_planning: Using planning request adapter 'Fix Start State Bounds'
ros.moveit_ros_planning: Using planning request adapter 'Fix Start State In Collision'
ros.moveit_ros_planning: Using planning request adapter 'Fix Start State Path Constraints'
```

rhaschke, 2021-02-17:
The key question is whether you have the time parameterization configured for (all) your MTC planners.
Figure out which subtrajectory doesn't have a time parameterization and add it in the correpsonding stage.

danem, 2021-02-17:
Sorry, I guess I'm unclear about what you mean here. I don't see any API for adding time parameterization to MTC stages beyond passing a custom `PlanningPipelinePtr` to `solvers::PipelinePlanner`, nor do I see such a thing done in any of the demos. I have however verified that the pipeline created by the default `Task::createPlanner` includes the `AddTimeParameterization` adapter.  

In my case, it appears to consistently fail the lift portion of the planned pick. The lift is done using a `MoveRelative` stage, which is using `solvers::CartesianPath`. I see in [cartesian_path](https://github.com/ros-planning/moveit_task_constructor/blob/master/core/src/solvers/cartesian_path.cpp#L109) that it should be performing the time parameterization. Also note that all the points in the faulty trajectory are identical.

danem, 2021-02-17:
So I noticed that I had a `MoveRelative` stage in my plan that sometimes would be "empty". That is, min and max distance was set to 0 and the components of the direction vector were set to 0. When I remove this stage, my problems disappear. I assumed this wouldn't be an issue, but this explains why the trajectory is has repeated points and no timing information. Is this the expected behavior?

closed 2021-02-17 by danem

rhaschke, 2021-02-17:
Probably, the resulting trajectory of this empty stage would comprise two waypoints: the start and end, both being identical.
Obviously, this will result in repeated points.